meshtastic/firmware #3724: [Feature Request]: Possibility to add Continous Wave Operation where SX126x fited

slash-bit, 2024-04-25:
### Platform

NRF52, ESP32

### Description

It would be good to have an option for enabling Continuous Wave (CW) Operation at the reduced TX power.
This can be used to check stability of frequency. Check signal strength , align directional antennas (Yagi) where continuous signal emission TX is required.
The CW option could be enabled by toggle button in the mobile app , web app.
This mode is implemented here:
https://github.com/clodpi/LoRaMAC-SX1262/tree/master/src/apps/tx-cw
https://github.com/clodpi/LoRaMAC-SX1262/blob/master/src/radio/radio.h

and Described here :https://www.rfsolutions.co.uk/downloads/1537522406DS_SX1261-2_V1.1_SEMTECH.pdf
Page 61. 
The timeout time can ensure the Duty cycle is not exceeded in certain countries
[image]
.

GUVWAF, 2024-04-26:
I'm not sure whether this is a good fit for the firmware, as the rest of the firmware would be completely unused (you can't receive and transmit when you're transmitting yourself).
I think it's better to just flash the firmware that does this to test and then flash Meshtastic again.

closed 2024-04-26 by thebentern

slash-bit, 2024-04-26:
Thanks for consideration. Its a shame this feature won't be worked on. I appreciate this particular feature doesn't enhance Meshtastic's concept of decentralized meshed network. But without tools and modules allowing people to experiment in order to build successful mesh with  optimally placed nodes and tuned antennas it could be challenging. 
On the discord channel I hear people responded to CW as welcomed feature. Ofcoarse a separate firmware can be flashed for this purpose , but I don't see that people who are not very familiar with the process would do that , especially where the node may be mounted on the mast and may be only accessible via admin channel.
The success of the Meshtastic is not only because people can message each other without internet, it gave them an arena to experiment with the radio, antennas , radio wave  propagation etc. So with the right tools to measure signal it would only enhance the success of the project.

KyleMaas, 2024-05-10:
I'd be interested to see any developments on this as well.  The firmware supports specifying a frequency offset, but how would you figure that out without the tools to do so?  Even if this is developed as a separate project, it would be useful.