meshtastic/firmware #3736: [Bug]: T-Beam Supreme (S3 core) loses USB connectivity with latest FW 2.3.6.7a3570a Beta

GregEigsti, 2024-04-27:
### Category

Serial

### Hardware

T-Beam S3

### Firmware Version

2.3.6.7a3570a Beta

### Description

I have a python app that connects to a T-Beam Supreme via serial/USB and noticed that I lose the ability to connect via serial/USB with the latest FW `2.3.6.7a3570a Beta` running. Rolling back to FW `2.3.4.ea61808 Beta` fixes the issue thankfully. The python CLI app `meshtastic` is also unable to connect via serial/USB when this T-Beam Supreme is running FW `2.3.6.7a3570a Beta`. I have rolled the FW from previous to latest several times and this is 100% reproducible; for me anyway ;)

The T-Beam is connected to a Ras Pi 5 via serial/USB where a python script, which uses the Meshtastic python API, interacts with it. The T-Beam is also configured to use WiFi and uses that network connection to forward MQTT. I do notice the same issue when connecting this T-Beam to a MacOS work station.

### Relevant log output

```Shell
greg@meshaprs:~/src/python/meshtastic/meshaprs$ meshtastic --support

If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/python/issues
When adding an issue, be sure to include the following info:
 System: Linux
   Platform: Linux-6.5.0-1015-raspi-aarch64-with-glibc2.38
   Release: 6.5.0-1015-raspi
   Machine: aarch64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v2.3.7
 Executable: /home/greg/.local/bin/meshtastic
 Python: 3.11.6 CPython GCC 13.2.0

Please add the output from the command: meshtastic --info

Will add "meshtastic --info" upon request and privately.
```

GregEigsti, 2024-04-27:
I wonder if the T-Beam, with FW `2.3.6.7a3570a Beta` is stuck in `DFU` (or whatever) mode? I did notice during one of the rollbacks to 2.3.4 that I did not have to do the funky dual button press to make it accept new FW from the web flasher - it was ready to accept new FW immediately after booting with no button presses. Maybe this will help...

GUVWAF, 2024-04-27:
Seems that this #3597 might be the cause.

GUVWAF, 2024-05-01:
I don't have a T-Beam Supreme, but maybe @oseiler2 can have a look.

oseiler2, 2024-05-01:
Let me have a look

oseiler2, 2024-05-02:
It seems that we do need `ARDUINO_USB_CDC_ON_BOOT=1` after all in order for the CLI to work with the ESP32-S3. Otherwise I cannot get the S3 to read from USB Serial.

Issue #3650 that can cause the S3 to block on boot when no Serial is connected (and is triggered by  `ARDUINO_USB_CDC_ON_BOOT=1` in combination with `ARDUINO_USB_MODE=1`) has recently been fixed in [arduino-esp32 @ 2.0.15](https://github.com/espressif/arduino-esp32/releases/tag/2.0.15) . I've confirmed that using `platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.15`

GregEigsti, 2024-05-02:
Thanks! Will give this another try on next beta branch FW release.

closed 2024-05-05 by thebentern

bmarshallbri, 2024-05-16:
I just flashed a TBeam S3 with 2.3.9.f06c56a Alpha and I am still unable to get a serial connection over USB.

Manasovo, 2024-08-06:
I have the same issue with firmware 2.4.0.46d7b82 and 2.3.15.deb7c27. I have not yet had the time to try 2.4.1 or older like 2.3.14 and lower.

In short the USB serial port is not working, it is not possible to read data from it and after some time it disappears even with /dev/tty.

Are you not noticing on tbeam-s3-core again the same behavior you described @GregEigsti ? THX

GregEigsti, 2024-08-06:
@Manasovo - Cannot say; do not currently have access to location where the tbeam-s3-core is running and it is still on `2.3.4.ea61808 Beta`. Sorry about that.

Manasovo, 2024-08-07:
Version 2.3.13 seems to be stable, the serial port works and shows no errors after about 24 hours. I will test further.