meshtastic/firmware #3741: Add BR region

barbabarros, 2024-04-27:
Would it be possible to add the BR region, for Brazil? This way we can have a channel on the official mqtt server, named "msh/BR" by default in the mqtt configurations.

Page 10 of the PDF file for Brazilian parameters

https://lora-alliance.org/wp-content/uploads/2020/11/RP_2-1.0.2.pdf

https://lora-alliance.org/resource_hub/rp2-102-lorawan-regional-parameters/

garthvh, 2024-04-27:
@andrekir Thoughts?

andrekir, 2024-04-28:
the LoRa region is added to the root topic just as a suggestion to separate traffic but users can use any root topic they want. creating duplicate LoRa regions for every country is unnecessary.

barbabarros, 2025-07-20:
@andrekir please review this issue.

Brazil uses a different frequency plan than Australia and the USA.
You can check the documents included in the message above.

902 - 907.5 MHz
915 - 928 MHz (AU915-928)
433 - 435 MHz (EU433)

As I said, it's not a duplicate frequency plan because it's different from all the others.

fifieldt, 2025-07-21:
@barbabarros , can you please obtain original source references from Anatel ? We've often found the lora alliance docs have slight issues and don't capture all of the nuances. Portuguese is fine.

barbabarros, 2025-07-21:
@fifieldt

902-907.5/915-928 Mhz
(page 15)
https://sistemas.anatel.gov.br/anexar-api/publico/anexos/download/a028ab5cc4e3f97442830bba0c8bd1dd

(Annex IV)
https://informacoes.anatel.gov.br/legislacao/resolucoes/2025/2001-resolucao-772

about 433-435Mhz (not for LoRa if I understand correctly)
"6.11. Frequency hopping and spread spectrum techniques are not permitted in bands below 440 MHz."
https://informacoes.anatel.gov.br/legislacao/atos-de-requisitos-tecnicos-de-gestao-do-espectro/2024/1919-ato-926

if you need any further information, please let me know.
thank you.

fifieldt, 2025-07-21:
Thanks @barbabarros, very thorough.

So it looks like for the 915 - 928 MHz band ANZ is identical, 433-435 is not allowed(?) ... which leaves 902-907.5 as 'missing'.

Since 915 - 928 MHz has more bandwidth, can I ask you to check with the Brazilian community about the desire to use 902-907.5?

We could add a band for 902. However, since 915 - 928 MHz is identical to an existing band we wouldn't add one for that.

Please check if there's any concern about having the choice of BR902 and ANZ. The worry might be that people would automatically pick BR without knowing that ANZ is an option and get less bandwidth or be on a different frequency to the majority of the community as a result :) If there isn't the community desire for the 902 band, it might be safer not to add it to avoid confusion ... but we definitely can if the community wants it :)

barbabarros, 2025-07-21:
@fifieldt could you rename ANZ to ANZ_BR or something similar?
On the Android app, it currently looks like Australia / New Zealand. Could you rename it to Australia / Brazil / New Zealand, so we can add BR902 without any concerns?

What do you think about 2.4GHz? Will it be added as well?
thank you.

fifieldt, 2025-07-21:
That would be up to the app developers, I'm afraid - not an area I am skilled in :( I'm just trying to manage the limited resources we have on the hardware devices.

It looks like the main difference in 2.4GHz is the power limit (10dBm in the current one vs 30dBm allowed in Brazil) ... to be honest, we haven't done much Meshtastic at 2.4GHz, so we probably need to make a plan for that more generally before adding in other frequencies...

barbabarros, 2025-07-21:
I can’t speak for the entire Brazilian community, but from what I see on the Meshtastic map, all nodes in Brazil currently use ANZ, so users already know that ANZ is a valid option for use in Brazil.

Therefore, I don’t see any issue with adding BR902.
Renaming the dropdown entry in the Android app would be necessary to make things clear.

closed 2025-07-23 by caveman99