meshtastic/firmware #3743: [Bug]: ESP32 boards shutdown after on_battery_shutdown_after_secs even when connected to USB

fifieldt, 2024-04-28:
### Category

Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.3.4.ea61808

### Description

As [reported](https://meshtastic.discourse.group/t/question-re-configuration-power-shutdown-on-battery-delay/12241) by chrisj, ESP32 boards shut down after "Shutdown on battery delay" seconds, even when connected to USB. This is confirmed to affect  Heltec V3 and Heltec Wireless Tracker. This is because `PowerStatus->getHasUSB` returns the incorrect value for these boards.

`PowerStatus->getHasUSB` is used in various places to determine whether a device is connected to external power. For example, in `src/PowerFSMThread.h`, we use getHasUSB to determine whether we should check to shut down a device after `config.power.on_battery_shutdown_after_secs`.

getHasUSB works fine for NRF-based devices, but currently returns 0 (i.e. USB not connected) for ESP32-based devices, even when USB is connected.

### Relevant log output

```Shell
# Heltec Wireless Tracker - USB connected
DEBUG | 01:28:08 87 [Power] Battery: usbPower=0, isCharging=0, batMv=3701, batPct=47

# LilyGo T-Beam - USB connected
DEBUG | 01:29:22 45549 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
```

geeksville, 2024-04-28:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/question-re-configuration-power-shutdown-on-battery-delay/12241/6

garthvh, 2024-04-28:
For the heltec you will need to adjust your adc value until is is showing powered properly, the tbeam which has a better power management chip is detecting isb power.

fifieldt, 2024-04-28:
Thanks Garth. I confirm changing the adc value on heltec-wireless-tracker fixes the hasUSB issue.

Perhaps this is just a documentation fix only? Adding a warning to https://meshtastic.org/docs/configuration/radio/power/#shutdown-after-losing-power to check that USB is correctly detected, and alter ADC value if not?

closed 2024-04-28 by garthvh