meshtastic/firmware #3746: [Bug]: Suspected duplicate device identifier

grinco, 2024-04-28:
### Category

Other

### Hardware

Other

### Firmware Version

2.3.6.7a3570a

### Description

My device has been heard on another continent. See screenshots. I suspect that it's an ID overlap.
[image]
[image]
[image]

GUVWAF, 2024-05-01:
According to Meshmap it's not the same Node ID. It shows different user IDs (`!327b23ca` and `!da575630`), which is derived from this. But I assume you don't have it set to US region when you're in Czech Republic? So then something else is wrong.

I see you run the Linux Native application (it says `PORTDUINO`) which uses a random ID if it doesn't has Bluetooth, but you can specify it with `--hwid`.

thebentern, 2024-05-01:
Meshmap is also not an official meshtastic source, though I do believe much of the data comes from our mqtt.

closed 2024-05-01 by thebentern