meshtastic/firmware #3748: [Bug]: Native build failure in VSC

DavidPeterRoberts78, 2024-04-29:
### Category

Other

### Hardware

Other

### Firmware Version

2.3.8.21311bb

### Description

PlatformIO: Build runs but fails

### Relevant log output

```Shell
Compiling .pio/build/native/libae6/LovyanGFX/lgfx/Fonts/IPA/lgfx_font_japan.c.o
Compiling .pio/build/native/libae6/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_cn.c.o
Compiling .pio/build/native/libae6/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_ja.c.o
Compiling .pio/build/native/libae6/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_kr.c.o
Compiling .pio/build/native/libae6/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_tw.c.o
In file included from .pio/libdeps/native/LovyanGFX/src/lgfx/Fonts/efont/lgfx_efont_cn.c:8:
/Users/erman/.platformio/packages/framework-portduino/cores/portduino/Arduino.h:9:10: fatal error: 'argp.h' file not found
#include <argp.h>
         ^~~~~~~~
Compiling .pio/build/native/libae6/LovyanGFX/lgfx/utility/lgfx_pngle.c.o
In file included from .pio/libdeps/native/LovyanGFX/src/lgfx/Fonts/IPA/lgfx_font_japan.c:8:
/Users/erman/.platformio/packages/framework-portduino/cores/portduino/Arduino.h:9:10: fatal error: 'argp.h' file not found
#include <argp.h>
         ^~~~~~~~
Compiling .pio/build/native/libae6/LovyanGFX/lgfx/utility/lgfx_qoi.c.o
In file included from .pio/libdeps/native/LovyanGFX/src/lgfx/Fonts/efont/lgfx_efont_ja.c:8:
/Users/erman/.platformio/packages/framework-portduino/cores/portduino/Arduino.h:9:10: fatal error: 'argp.h' file not found
#include <argp.h>
         ^~~~~~~~
Compiling .pio/build/native/libae6/LovyanGFX/lgfx/utility/lgfx_qrcode.c.o
In file included from .pio/libdeps/native/LovyanGFX/src/lgfx/Fonts/efont/lgfx_efont_kr.c:8:
/Users/erman/.platformio/packages/framework-portduino/cores/portduino/Arduino.h:9:10: fatal error: 'argp.h' file not found
#include <argp.h>
         ^~~~~~~~
In file included from .pio/libdeps/native/LovyanGFX/src/lgfx/utility/lgfx_pngle.c:42:
In file included from .pio/libdeps/native/LovyanGFX/src/lgfx/utility/pgmspace.h:29:
/Users/erman/.platformio/packages/framework-portduino/cores/portduino/Arduino.h:9:10: fatal error: 'argp.h' file not found
#include <argp.h>
         ^~~~~~~~
1 error generated.
*** [.pio/build/native/libae6/LovyanGFX/lgfx/utility/lgfx_pngle.c.o] Error 1
In file included from .pio/libdeps/native/LovyanGFX/src/lgfx/Fonts/efont/lgfx_efont_tw.c:8:
/Users/erman/.platformio/packages/framework-portduino/cores/portduino/Arduino.h:9:10: fatal error: 'argp.h' file not found
#include <argp.h>
         ^~~~~~~~
1 error generated.
*** [.pio/build/native/libae6/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_cn.c.o] Error 1
1 error generated.
*** [.pio/build/native/libae6/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_kr.c.o] Error 1
1 error generated.
*** [.pio/build/native/libae6/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_ja.c.o] Error 1
1 error generated.
*** [.pio/build/native/libae6/LovyanGFX/lgfx/Fonts/IPA/lgfx_font_japan.c.o] Error 1
1 error generated.
*** [.pio/build/native/libae6/LovyanGFX/lgfx/Fonts/efont/lgfx_efont_tw.c.o] Error 1
============================================================================================== [FAILED] Took 10.88 seconds ==============================================================================================

Environment    Status    Duration
-------------  --------  ------------
native         FAILED    00:00:10.876
========================================================================================= 1 failed, 0 succeeded in 00:00:10.876 =========================================================================================
```

jp-bennett, 2024-04-29:
This is building on Windows? I'm not sure that was ever a supported platform, even when simulator only.

mverch67, 2024-05-01:
This can never work, because the portduino layer is linux only. Think of _native_ as "native linux".
If you have a lot of time and experience then you can implement a new windows target and write a windows variant of SimRadio, that could in principle work. But as probably nobody has interest in that and because of the amount of changes required it's unlikely that this ever gets merged into the code base.

closed 2024-05-01 by thebentern