meshtastic/firmware #3751: Enhancement: Air Quality values getting cut off

HarukiToreda, 2024-04-30:
When Telemetry is turned on, the Air quality values are getting cut off at the bottom of the screen. I think reducing the header size will give enough room for this to show on smaller screens like on the Heltec V3.

todd-herbert, 2024-04-30:
In EnvironmentTelemetry.cpp, it might be an idea to modify the `drawFrame` method, so that it checks `display->height()`. If the height is below some value, could then use`FONT_SMALL` everywhere `FONT_MEDIUM` is used?

HarukiToreda, 2024-04-30:
> In EnvironmentTelemetry.cpp, it might be an idea to modify the `drawFrame` method, so that it checks `display->height()`. If the height is below some value, could then use`FONT_SMALL` everywhere `FONT_MEDIUM` is used?

Thank you, I think I have a working rough draft, I will wait for my other pull request to post the fix for this. 

[image]

paoloponzano, 2024-04-30:
Very nice fix, thank you :)

thebentern, 2024-05-05:
Closed by #3785

closed 2024-05-06 by HarukiToreda