meshtastic/firmware #3760: [Bug]: Heltec Wireless Tracker Battery issues - possible fix

madeofstown, 2024-05-01:
### Category

Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.3.7.30fbcab

### Description

THE PROBLEM
The heltec wireless tracker has poor battery performance and battery monitoring doesn't seem to work at all. It always shows that it is powered by USB and will over drain the battery and potentially cause a fire. 

POSSIBLE FIX
There is a recent (within the last 15 days) post on the Meshtastic discourse forums with a potential fix that I'm hoping can be implemented. https://meshtastic.discourse.group/t/heltec-tracker-v1-and-v1-1-power-config/12016/4

madeofstown, 2024-05-11:
Turns out my battery monitoring issues were caused by hardware defects. I'm still wondering if the "possible fix" would help with battery life though.

mkgin, 2024-07-24:
I have been ill the past three weeks, better now but a lot of things have piled up. Nice that it has been committed. Let's keep this open for now. I'll try to set up a raspberry pi in the next week  to measure current and collect logs to see what the actual difference is.

geeksville, 2024-08-22:
sorry - I just noticed this bug (thanks @GUVWAF !).  I think I just pushed identical changes in https://github.com/geeksville/Meshtastic-esp32/commit/d017fc7a5d392eb35b197c104a33ab2c533bc6ea .  I'll also be adding a commit tomorrow to fix the Vext enable problem on this board.

mkgin, 2024-08-22:
> sorry - I just noticed this bug (thanks @GUVWAF !). I think I just pushed identical changes in [geeksville@d017fc7](https://github.com/geeksville/Meshtastic-esp32/commit/d017fc7a5d392eb35b197c104a33ab2c533bc6ea) . I'll also be adding a commit tomorrow to fix the Vext enable problem on this board.

No problem...  I've been overloaded this summer.

fifieldt, 2024-09-16:
Fixed in https://github.com/meshtastic/firmware/commit/d017fc7a5d392eb35b197c104a33ab2c533bc6ea

closed 2024-09-16 by fifieldt