meshtastic/firmware #3762: [Bug]: Program button no longer responds when Canned messages is enabled on Heltec V3

HarukiToreda, 2024-05-02:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.7.30

### Description

When Canned Messages is turned on with a Cardkb on the Heltec V3. The program button stops working, it does not respond no matter how many times it is pressed.

HarukiToreda, 2024-05-04:
Issue Fixed

HarukiToreda, 2024-05-04:
I was able to pinpoint the issue to the BME680 sensor attached to the SDA SCL pins. When present the program button stops working, All nodes without the sensor work fine.

todd-herbert, 2024-05-04:
Does the button still turn the display on? (Not from sleep, just while idle)

HarukiToreda, 2024-05-04:
> Does the button still turn the display on? (Not from sleep, just while idle)

False alert, I factory reset without reloading from backup and it works now, I keep forgetting this sometimes happens when upgrading firmware

closed 2024-05-04 by HarukiToreda