meshtastic/firmware #3766: [Bug]: Messages not displayed on screen when Store and Forward (SaF) module enabled and device lacks PSRAM (Heltec Wireless Paper)

k1n6b0b, 2024-05-02:
### Category

Hardware Compatibility

### Hardware

Heltec Wireless Paper

### Firmware Version

2.2.24

### Description

On the Heltec Wireless Paper with SaF module enabled as a client (and possible other devices lacking PSRAM) when a message is received, the screen flips to page 1 but does not display the message message. I see the following:
[image]

however if I disable the SaF module, I see the expected message
[image]

Per chat on discord: https://discord.com/channels/867578229534359593/919642584480112750/1235451885813633024

todd-herbert, 2024-05-02:
I believe that the code responsible for this behavior is [Screen.cpp L402](https://github.com/meshtastic/firmware/blob/4f54862d6363aed10dedfc2b1f8aa9d166242144/src/graphics/Screen.cpp#L402).

The comment for this method is:
>  Ignore messages originating from phone (from the current node 0x0) unless range test or store and forward module are enabled

Is this method working as expected? I feel like its logic might be a bit off, but I'm not familiar with this part of the code.

caveman99, 2024-05-03:
i think the reason for this is if your request the S&F history, a lot of messages will pop up on the screen in a very short sequence. so it's logical this is disabled. You'd use S&F only from a phone.

caveman99, 2024-05-03:
The more i think about it we best leave it like this ... :-)

closed 2024-05-03 by caveman99

k1n6b0b, 2024-05-04:
I think it's unfortunate to have to choose between using the screen to view messages and the SaF functionality. We use these as off-grid communicators and losing the message capability on the device would be challenging should the phone die or otherwise

garthvh, 2024-05-04:
This device does not support store and forward.

k1n6b0b, 2024-05-04:
You raise a good question Garth!

Can a non-PSRAM be a SaF Client?

If yes: I asset it is still confusing to silently hide the message screen. I feel this could be improved, and would like it to be. 

If no: Being able to 'enable' SaF is confusing on these devices. 

I would challenge that closing this ticket as expected behavior falls short of the ambiguous user experience provided leading me and others through a discussion on discord. 

I would ask that, if you do not intend to change the message screen 'hidden' behavior you at least explore other means of improving the ux through documentation or disabling that function for unsupported devices.