meshtastic/firmware #3767: [Bug]:Wisblock and T-echo

PhilRosa-Leeke, 2024-05-02:
### Category

Other

### Hardware

T-Echo, Rak4631

### Firmware Version

2.3.6.7a3570a

### Description

Well, and this is a tad odd, it all had to do with the latest firmware (firmware-2.3.6.7a3570a). For some reason it would not save any settings whats so ever! AND yes, before you ask, I tried on the Mac App, the IOS App, the Android App and the online client thingy both using Chrome and Arc, but all failed. 

Then, a light-bulb turned on and I did the following, I simpley over wrote the installed firmware with an older version (firmware-2.2.24.e6a2c06) and Hey Presto evrything worked (and is still working). 

Now I will have to say that it was due to my T-echo that I changed to the older version of the firmware because ever since I updated it to the 2.3.6.7a3570a version, it had been playing odd ball with me. EX: Whenever I tried to turn the T-echo off from the IOS App it would not confirm it had shut down (although it had) and the "Power Off" and "Reboot" button would stay active (normally they fade out once the T-echo has shut down.).

So I then gave it a go on my friends Wisblock and, as stated above, Hey Presto, all is working.

thebentern, 2024-05-02:
Can you grab some serial logs using the web flasher or python cli when you attempt to save via the apps? This is very strange behavior that I am not able to reproduce

PhilRosa-Leeke, 2024-05-02:
The only "logs” I can see are all from the moment I managed to backtrack to the older firmware on the Wisblock. I am attaching the log from my T-echo, I hope that is what you need, if not please let me know and tell me how to get it for you.

Kind regards,

Phil

￼

> 
> 
> Can you grab some serial logs using the web flasher or python cli when you attempt to save via the apps? This is very strange behavior that I am not able to reproduce

MESHTASTIC MESH ACTIVITY LOG
📟 Node info received for: 68598467 - 29/04/24, 20:19:34,78 
📟 Node info received for: 3663963888 - 29/04/24, 20:19:34,84 
📟 Node info received for: 230109826 - 29/04/24, 20:19:34,90 
🎛️ Channel 0 received from: 2059869327 - 29/04/24, 20:19:34,96 
🎛️ Channel 1 received from: 2059869327 - 29/04/24, 20:19:35,02 
🎛️ Channel 2 received from: 2059869327 - 29/04/24, 20:19:35,08 
📟 Device config received: 2059869327 - 29/04/24, 20:19:35,44 
🗺️ Positon config received: 2059869327 - 29/04/24, 20:19:35,50 
🗺️ Power config received: 2059869327 - 29/04/24, 20:19:35,56 
🌐 Network config received: 2059869327 - 29/04/24, 20:19:35,62 
🖥️ Display config received: 2059869327 - 29/04/24, 20:19:35,68 
📻 LoRa config received: 2059869327 - 29/04/24, 20:19:35,74 
📶 Bluetooth config received: 2059869327 - 29/04/24, 20:19:35,80 
🌉 MQTT module config received: 2059869327 - 29/04/24, 20:19:35,86 
🤖 Serial module config received: 2059869327 - 29/04/24, 20:19:35,92 
📣 External Notification module config received: 2059869327 - 29/04/24, 20:19:35,98 
📬 Store & Forward module config received: 2059869327 - 29/04/24, 20:19:36,04 
⛰️ Range Test module config received: 2059869327 - 29/04/24, 20:19:36,10 
📈 Telemetry module config received: 2059869327 - 29/04/24, 20:19:36,16 
🥫 Canned Message module config received: 2059869327 - 29/04/24, 20:19:36,22 
🏮 Ambient Lighting module config received: 2059869327 - 29/04/24, 20:19:36,46 
🕵️ Detection Sensor module config received: 2059869327 - 29/04/24, 20:19:36,52 
🧑‍🤝‍🧑 PAX Counter config received: 2059869327 - 29/04/24, 20:19:36,58 
📟 Node info received for: 3663961428 - 29/04/24, 20:20:01,00 
📈 Telemetry received for: 68598467 - 29/04/24, 20:20:21,76 
💬 Message received from the text message app. - 29/04/24, 20:20:24,74 
📍 Position Packet received from node: 68598467 - 29/04/24, 20:20:37,54 
💬 Sent message 3570426574 from 2059869327 to 68598467 - 29/04/24, 20:20:41,96 
🕸️ Routing received for RequestID: 3570426574 Ack Status: Acknowledged - 29/04/24, 20:20:43,84 
🕸️ Routing received for RequestID: 1861906186 Ack Status: Acknowledged - 29/04/24, 20:21:09,76 
🛎️ Issuing Want Config to Phil_1c8f - 02/05/24, 13:26:17,64 
ℹ️ MyInfo received: 2059869327 - 02/05/24, 13:26:17,93 
🏷️ Device Metadata received from: 2059869327 - 02/05/24, 13:26:17,99 
📟 Node info received for: 2059869327 - 02/05/24, 13:26:18,09 
📟 Node info received for: 3663961428 - 02/05/24, 13:26:18,14 
📟 Node info received for: 68598467 - 02/05/24, 13:26:18,20 
📟 Node info received for: 3663963888 - 02/05/24, 13:26:18,27 
📟 Node info received for: 230109826 - 02/05/24, 13:26:18,32 
🎛️ Channel 0 received from: 2059869327 - 02/05/24, 13:26:18,39 
🎛️ Channel 1 received from: 2059869327 - 02/05/24, 13:26:18,44 
🎛️ Channel 2 received from: 2059869327 - 02/05/24, 13:26:18,50 
📟 Device config received: 2059869327 - 02/05/24, 13:26:18,86 
🗺️ Positon config received: 2059869327 - 02/05/24, 13:26:18,92 
🗺️ Power config received: 2059869327 - 02/05/24, 13:26:18,98 
🌐 Network config received: 2059869327 - 02/05/24, 13:26:19,04 
🖥️ Display config received: 2059869327 - 02/05/24, 13:26:19,10 
📻 LoRa config received: 2059869327 - 02/05/24, 13:26:19,16 
📶 Bluetooth config received: 2059869327 - 02/05/24, 13:26:19,22 
🌉 MQTT module config received: 2059869327 - 02/05/24, 13:26:19,28 
🤖 Serial module config received: 2059869327 - 02/05/24, 13:26:19,34 
📣 External Notification module config received: 2059869327 - 02/05/24, 13:26:19,40 
📬 Store & Forward module config received: 2059869327 - 02/05/24, 13:26:19,46 
⛰️ Range Test module config received: 2059869327 - 02/05/24, 13:26:19,52 
📈 Telemetry module config received: 2059869327 - 02/05/24, 13:26:19,58 
🥫 Canned Message module config received: 2059869327 - 02/05/24, 13:26:19,64 
🏮 Ambient Lighting module config received: 2059869327 - 02/05/24, 13:26:19,91 
🕵️ Detection Sensor module config received: 2059869327 - 02/05/24, 13:26:19,97 
[... 418 lines cut ...]
📟 Node info received for: 2059869327 - 02/05/24, 19:04:13,82 
📟 Node info received for: 751657827 - 02/05/24, 19:04:13,88 
📟 Node info received for: 68598467 - 02/05/24, 19:04:13,94 
📟 Node info received for: 3663961428 - 02/05/24, 19:04:14,00 
📟 Node info received for: 230109826 - 02/05/24, 19:04:14,06 
🎛️ Channel 0 received from: 2059869327 - 02/05/24, 19:04:14,12 
🎛️ Channel 1 received from: 2059869327 - 02/05/24, 19:04:14,18 
🎛️ Channel 2 received from: 2059869327 - 02/05/24, 19:04:14,25 
📟 Device config received: 2059869327 - 02/05/24, 19:04:14,60 
🗺️ Positon config received: 2059869327 - 02/05/24, 19:04:14,66 
🗺️ Power config received: 2059869327 - 02/05/24, 19:04:14,73 
🌐 Network config received: 2059869327 - 02/05/24, 19:04:14,81 
🖥️ Display config received: 2059869327 - 02/05/24, 19:04:14,87 
📻 LoRa config received: 2059869327 - 02/05/24, 19:04:14,93 
📶 Bluetooth config received: 2059869327 - 02/05/24, 19:04:14,99 
🌉 MQTT module config received: 2059869327 - 02/05/24, 19:04:15,06 
🤖 Serial module config received: 2059869327 - 02/05/24, 19:04:15,11 
[... cut ...]
📬 Store & Forward module config received: 2059869327 - 02/05/24, 19:04:15,23 
⛰️ Range Test module config received: 2059869327 - 02/05/24, 19:04:15,32 
📈 Telemetry module config received: 2059869327 - 02/05/24, 19:04:15,38 
🥫 Canned Message module config received: 2059869327 - 02/05/24, 19:04:15,44 
🏮 Ambient Lighting module config received: 2059869327 - 02/05/24, 19:04:15,68 
🕵️ Detection Sensor module config received: 2059869327 - 02/05/24, 19:04:15,74 
🧑‍🤝‍🧑 PAX Counter config received: 2059869327 - 02/05/24, 19:04:15,80

thebentern, 2024-05-02:
The logs I'm referring to can be gathered from here: https://flasher.meshtastic.org/
[image]

todd-herbert, 2024-05-02:
Is there any chance that the devices just need a proper factory erase before upgrading?

PhilRosa-Leeke, 2024-05-02:
I do not think this to be an issue. The Wisblock was straight out of the box and I flashed it with said firmware, but from then on all means of connecting with it failed and only after re-installing the older quoted firmware did it come to life.

closed 2024-05-02 by thebentern