meshtastic/firmware #3779: [Bug]: RAK12500 GPS Checksum Failures

BobsBlueNorth, 2024-05-03:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.3.4.ea61808

### Description

DEVICE:
Rak Wisblock
Baseboard: RAK19007
Core: RAK4631
GPS: RAK12500
Temp/Hum Sensor: RAK1901

PHONE:
Samsung S21, Android 14, Mesthtastic App 2.3.4

I appear to be having an issue with the GPS.  It works about 50% of the time but I am getting Checksum failures and the Device doesn't appear to be sending its position to the phone for display in the app often or sometimes at all.  Anyone run into this?

### Relevant log output

```Shell
INFO | 20:44:05 1485 [GPS] Setting GPS power=1

WARN | 20:44:05 1485 [GPS] 1 new GPS checksum failures, for a total of 8.

DEBUG | 20:44:07 1487 [GPS] WANT GPS=0

DEBUG | 20:44:07 1487 [GPS] GPS Lock took 2, average 7

INFO | 20:44:07 1487 [GPS] Setting GPS power=0

DEBUG | 20:44:07 1487 [GPS] Sleep Time: 112763

DEBUG | 20:44:07 1487 [GPS] publishing pos@66354c98:2, hasVal=1, Sats=6, GPSlock=1

DEBUG | 20:44:07 1487 [GPS] New GPS pos@66354c98:3 lat=61.309072, lon=-147.529347, alt=173, pdop=1.86, track=133.38, speed=0.00, sats=6

DEBUG | 20:44:07 1487 [GPS] onGPSChanged() pos@66354c98, time=1714769047, lat=613090720, lon=-1475293471, alt=173

INFO | 20:44:07 1487 [GPS] updatePosition LOCAL pos@66354c98, time=1714769047, latI=613090720, lonI=-1475293471, alt=173

DEBUG | 20:44:07 1487 [GPS] Setting local position: latitude=613090720, longitude=-1475293471, time=1714769047

DEBUG | 20:44:07 1487 [GPS] Node status update: 5 online, 8 total
```

GPSFan, 2024-05-03:
A few things to try, update to a more recent firmware (yeah they always say that), try removing the RAK 1901. See if anything changes, I don't have a RAK 12500 to test with, but I have seen GPS checksum failures on ESP32 platforms with different GPSs.

GPSFan, 2024-05-04:
I have a 12500 and 13002 on order, and will be able to investigate further soon.

polamnus, 2024-05-05:
Rak19700 + Rak4631 + Rak12500 here...no environmental sensor. I'm seeing an incrementing number of GPS checksum errors. It seems like the 12500 is working, however. I believe there's not always an instant update between the GPS internal location data and what it's sent to the app...don't quote me on this exactly, but I believe it can be up to a minute before it updates the client app via Bluetooth.

Regardless, I'm happy to help test anything...I just got my Rak12500 so wasn't sure if the checksum errors were anomalous or not, as it seems like I'm tracking satellites and getting location updates reasonably well enough. I have noticed what feels like an inordinately long time (~3-5 minutes)before I get a GPS lock from a cold-start with an open sky picture...relative to what I'm used to on a cell-phone, for instance...but as I said I just got my 12500 recently so wasn't sure of what to expect on that front.

LMK if I can help provide any diagnostic data.

Regards!

GPSFan, 2024-05-05:
It's just a hunch, but I think it might be a TinyGPS+ issue. It also may be related to the NRF52 running at ~60MHz and not being able to handle the workload. Neither has any basis in fact as yet, but I'm certain that the Zoe is producing correctly checksummed NMEA sentences.

rc-adammikolajczyk, 2024-05-08:
FWIW, I'm starting to see some additional inconsistencies on my two Rak12500's. Namely, about 25% of the time, they will just never get lock. I've left one overnight, outside, with a clear sky picture and it'll just never get a lock. What I've found is that by constantly rebooting it, I can often make it lock...simply by rebooting it again. If it doesn't obtain a lock within 5 minutes, it's almost never going to, so reboot, wait 5, if not locked, try again.

Also, and I don't mean to conflate issues, but feel it's worth mentioning, my accuracy while locked it terrible. I'm seeing up to a 100 meter discrepancy between every successive position update.

I did test by moving my 12500's to slot D on the 19700 but obtained the same results.

Same results on the alpha and previous two beta firmwares. Same results on two independent units.
Finally, and this one seems weird, but I wanted to mention it, it seems like it almost ALWAYS obtains a lock within 3 minutes if it's plugged into USB-Serial. Is it possible, something about being plugged in, maybe something to do with power, is to blame?

It really seems like every time I try to plug myself in to USB to troubleshoot, I get a lock almost immediately after plugging in.

Once again, glad to run any tests, collect any logs or diagnostics if that would be helpful. I've got two 12500's to test with, and two. 19700/4631 combos.

closed 2024-05-15 by thebentern