meshtastic/firmware #3786: [Bug]: Debounce Up/Down/Select in canned messages to prevent skipping entries

lupusworax, 2024-05-05:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.7.3.0

### Description

When going through the list of canned messages up/ down mostly skips a entry when letting the button go. ( is registered as an additional button press) Very rarely it works for a few clicks, it probably needs a debounce set to prevent it from happening.

### Relevant log output

```Shell
All688
Info205
Debug452
WarnError
Disconnect

Build:Mar 27 2021

rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3808,len:0x44c

load:0x403c9700,len:0xbe4

load:0x403cc700,len:0x2a38

entry 0x403c98d4

E (351) esp_core_dump_flash: No core dump partition found!

E (351) esp_core_dump_flash: No core dump partition found!

��@INFO | ??:??:?? 0

//\ E S H T /\ S T / C

INFO | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset

DEBUG | ??:??:?? 0 Filesystem files (495616/1048576 Bytes):

DEBUG | ??:??:?? 0 /prefs/cannedConf.proto (146 Bytes)

DEBUG | ??:??:?? 0 /prefs/channels.proto (59 Bytes)

DEBUG | ??:??:?? 0 /prefs/config.proto (96 Bytes)

DEBUG | ??:??:?? 0 /prefs/db.proto (698 Bytes)

DEBUG | ??:??:?? 0 /prefs/module.proto (144 Bytes)

DEBUG | ??:??:?? 0 /static/.gitkeep (0 Bytes)

DEBUG | ??:??:?? 0 /static/Logo_Black.svg.gz (602 Bytes)

DEBUG | ??:??:?? 0 /static/Logo_White.svg.gz (610 Bytes)

DEBUG | ??:??:?? 0 /static/apple-touch-icon.png.gz (3164 Bytes)
[... 1304 lines cut ...]

DEBUG | ??:??:?? 50 [RadioIf] Lora RX (id=0x61bfd106 fr=0xa4 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=5.75 rxRSSI=-31 hopStart=3)

DEBUG | ??:??:?? 50 [RadioIf] AirTime - Packet received : 575ms

DEBUG | ??:??:?? 50 [Router] Rx someone rebroadcasting for us (id=0x61bfd106 fr=0xa4 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=5.75 rxRSSI=-31 hopStart=3)

DEBUG | ??:??:?? 50 [Router] didn't find pending packet

DEBUG | ??:??:?? 50 [Router] Found existing packet record for fr=0x7c5a5ba4,to=0xffffffff,id=0x61bfd106

DEBUG | ??:??:?? 50 [Router] Found existing packet record for fr=0x7c5a5ba4,to=0xffffffff,id=0x61bfd106

DEBUG | ??:??:?? 50 [Router] Add packet record (id=0x61bfd106 fr=0xa4 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=5.75 rxRSSI=-31 hopStart=3)

DEBUG | ??:??:?? 50 [Router] Ignoring incoming msg, because we've already seen it (id=0x61bfd106 fr=0xa4 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=5.75 rxRSSI=-31 hopStart=3)

DEBUG | ??:??:?? 50 [Router] cancelSending id=0x61bfd106, removed=0

DEBUG | ??:??:?? 50 [Router] Incoming message was filtered 0x7c5a5ba4

Powered by ▲ Vercel | Meshtastic® is a registered trademark of Meshtastic LLC. | Legal Information.

Connected
```

lupusworax, 2024-05-05:
demonstration video: https://youtu.be/Lp6SAsvw-SU

lupusworax, 2024-05-16:
Built another device with an actual rotary encoder and it suffers the exact same issue.

Alexfehr1987, 2024-08-08:
I can confirm the same behaviour on my wireless tracker.

Gabesthomas, 2024-09-02:
Same behavior in heltec wireless tracker and v3. I've tried playing around as much as possible with settings, no relief.  Devs please add up, down, and select BUTTONS as an input source with added debounce.