meshtastic/firmware #3789: [Bug]: T-Beam v1.2, M8N - GPS module initializes but doesn't work

nikvostro, 2024-05-05:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.3.7

### Description

T-Beam v1.2, M8N - GPS module initializes but doesn't work . GPS module **works fine** with https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/firmware/NMEA_Reset firmware.

Troubleshooting steps performed:
1. Leaving for a long time >12 hours on the window. Several attempts - no luck
2. Factory reset, default settings, etc...
3. Reflash entire FW with pre-flashing "0" (blank) binary to overwrite flash contents
4. Lower ver 2.3.4, and 2.3.6 behaves the same way - no GPS positioning
5. [GPS_reset firmware ](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/firmware/NMEA_Reset/README.MD)feeds good NMEA flow with 15 satellites detected and 5 being used. 

**Important notice:** at the first start of freshly flashed meshtastic firmware I've obtained the device's GPS positing appear somewhere near the real. Like it was "cached" by the M8N module in time when it was driven by GPR_reset firmware a just couple of minutes before flashing Meshtastics firmware. Then GPS position was never reported.

From my POV the issue looks like related to a wrong communication with the module because it works fine (feeding NMEA via COM) with GPS reset firmware by Lilygo.

[DEBUG-20240505-144717.log](https://github.com/meshtastic/firmware/files/15212959/DEBUG-20240505-144717.log)

### Relevant log output

```Shell
Pease see attached log
```

GPSFan, 2024-05-05:
I wish I could replicate your issue.
A few questions:
Is this the first time you have used this particular t-beam for Meshtastic, or any other firmware?
Roughly how long has this t-beam been powered up since you have had it? i.e. 1 hour, 12 hours, a day, a few days?
Are you using the tiny little GPS antenna that usually comes with t-beams?
How long did it take to get a fix with the LiliGo firmware?
Was the location of the antenna the same as with the Meshtastic firmware?
Your log shows a reboot at 14:49:47.601, the following reset_reason=reset implies you pushed the reset button, is this correct?

The LiliGo firmware and Meshtastic configure the M8N differently, so comparing operation of one vs the other is not straightforward.
Your Meshtastic log shows the M8N recognized (14:49:50.920 -> 14:49:54.246) and configured. That means that the t-beam is communicating correctly with the M8N.
If Meshtastic doesn't get an initial GPS lock within 15 minutes (900sec) it gives up and turns off the GPS, you can see this in your log where it sets "WANT GPS =0" at 15:04:49:837, followed by "Setting GPS power=0. 
If you have not had the t-beam powered "long enough" the backup battery will not be charged enough to keep the GPS data during a power off, or power save mode. It can take a long time to charge (controlled by the AXP2101). It will then be performing a "cold start" when turned on, if the signal levels are low (as shown by your GPS display) it can take a long time to acquire enough data to get a fix. Low signal levels are the result of a poor antenna (like the one shipped with the t-beams) and poor view of the sky (sitting in a window). If the GPS power is turned off, and the backup battery cannot sustain the BBR layer in the M8N, the configuration will revert to defaults.
The LiliGo software resets the M8N's config to factory defaults and leaves the power on continuously, even with the tiny GPS antenna it will eventually get a fix.
This issue has come and gone in the firmware for a while, it seems to be related to the backup battery not being charged the first time the t-beam is used, as well as some unknowns that resolve themselves after a while.

nikvostro, 2024-05-06:
Hello Ken!

Thank you for involving!

Let me answer by points:
1. _Is this the first time you have used this particular t-beam for Meshtastic, or any other firmware?_ - for T-Beam - first time, and after failing GPS I've just tried GPS_reset factory FW. No other experience and tests with other FWs.
2. _Roughly how long has this t-beam been powered up since you have had it? i.e. 1 hour, 12 hours, a day, a few days?_ - tried different timings up to several days powered on. I've also tried playing with **Position** settings section by making **Position broadcast** and **GPS update** intervals bigger (1800) - didn't help also.
3. _Are you using the tiny little GPS antenna that usually comes with t-beams?_ - yes. I'm waiting for a bigger one now to arrive. I thought it was the root cause but after flashing GPS_reset this became inactual.
4. _How long did it take to get a fix with the LiliGo firmware?_ - hmmm, didn't fixed that :(  Within 1-2 hours definetely.
5. _Was the location of the antenna the same as with the Meshtastic firmware?_ - yes
6. _Your log shows a reboot at 14:49:47.601, the following reset_reason=reset implies you pushed the reset button, is this correct?_ - can't remember exactly, but most probably yes. Or maybe I've changed some of device Settings that caused a reboot (?)

I thought that battery was related to RTC but thank you for clarifying! Just now I've measured it's coltage 2.4V before powering ON the T-Beam and since ~10 minutes it has grew up to 2.74V and continued raising very-very slowly. In specs I see the 3V of nominal voltage for this type of batt... Not sure if everything is fine here but why GPS work with GPS_reset FW...

What you could advise for the further investigation?

Attaching another log for almost 5 hrs online.
[DEBUG-20240504-171359.log](https://github.com/meshtastic/firmware/files/15217603/DEBUG-20240504-171359.log)

[image]

GPSFan, 2024-05-06:
"How long did it take to get a fix with the LiliGo firmware? - hmmm, didn't fixed that :( Within 1-2 hours definetely."

This is the key to your issue, Meshtastic will give up if no fix in 15 minutes. Because it took so long to get a fix with the LiliGo firmware, that indicates that your view of the sky is very poor. 

Try this, with either firmware take the t-beam outside with a good view of the sky, no trees, buildings nearby. Make sure the GPS antenna is pointed towards the sky. If it takes more than a minuet to get a fix, then either your antenna or M8N is faulty.

nikvostro, 2024-05-06:
And how I can extend the time for getting GPS fix? I assume this linked with the some of broadcast intervals settings in **Position** section.
New GPS antenna is to come from day to day.