meshtastic/firmware #3791: [Bug]: T-beam v1.1 cannot setup CN470 directly

SteveHawk, 2024-05-05:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.3.8

### Description

After freshly flashed and reset a T-beam v1.1 board (the 433/470 MHz variant with SX1268 chip), directly setting region to CN470 would fail, with critical fault # 3 showing on screen. Error logs are attached down below.

Some probably important messages:

1. It reports critical error 7 before rebooting;
2. After booting, it fails to init either RF95, SX1262 or SX1268, then reporting error 3;
3. It also reports error 4 in NodeInfoModule and DeviceTelemetryModule.

However, if I set it to EU433 first (which works fine), then set it back to CN470, it instead works. None of the above error messages showed up, and in both cases (setting to EU433 or CN470), log reports RF95 init success.

### Relevant log output

```Shell
INFO | ??:??:?? 626 [Router] Client is setting the config

INFO | ??:??:?? 626 [Router] Setting config: LoRa

INFO | ??:??:?? 626 [Router] Wanted region 4, using CN

INFO | ??:??:?? 626 [Router] Saving changes to disk

DEBUG | ??:??:?? 626 [Router] Expanding short PSK #1

INFO | ??:??:?? 626 [Router] Wanted region 4, using CN

DEBUG | ??:??:?? 626 [Router] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms

DEBUG | ??:??:?? 626 [Router] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms

INFO | ??:??:?? 626 [Router] Radio freq=478.875, config.lora.frequency_offset=0.000

INFO | ??:??:?? 626 [Router] Set radio: region=CN, name=LongFast, config=0, ch=35, power=19

INFO | ??:??:?? 626 [Router] Radio myRegion->freqStart -> myRegion->freqEnd: 470.000000 -> 510.000000 (40.000000 mhz)

INFO | ??:??:?? 626 [Router] Radio myRegion->numChannels: 160 x 250.000kHz

INFO | ??:??:?? 626 [Router] Radio channel_num: 36

INFO | ??:??:?? 626 [Router] Radio frequency: 478.875000

INFO | ??:??:?? 626 [Router] Slot time: 42 msec

ERROR | ??:??:?? 626 [Router] NOTE! Recording critical error 7 at src/mesh/RF95Interface.cpp:154

INFO | ??:??:?? 626 [Router] Saving /prefs/config.proto

INFO | ??:??:?? 627 [Router] Saving /prefs/module.proto

INFO | ??:??:?? 627 [Router] Rebooting in 7 seconds

DEBUG | ??:??:?? 627 [Router] Module 'Admin' considered

DEBUG | ??:??:?? 627 [Router] Module 'routing' wantsPacket=1

INFO | ??:??:?? 627 [Router] Received routing from=0x0, id=0x78875a0b, portnum=6, payloadlen=17
[... 420 lines cut ...]

DEBUG | ??:??:?? 59 [DeviceTelemetryModule] updateTelemetry LOCAL

DEBUG | ??:??:?? 59 [DeviceTelemetryModule] Node status update: 2 online, 2 total

DEBUG | ??:??:?? 59 [DeviceTelemetryModule] showing standard frames

DEBUG | ??:??:?? 59 [DeviceTelemetryModule] Showing 0 module frames

DEBUG | ??:??:?? 59 [DeviceTelemetryModule] Total frame count: 103

DEBUG | ??:??:?? 59 [DeviceTelemetryModule] Added modules. numframes: 0

DEBUG | ??:??:?? 59 [DeviceTelemetryModule] Finished building frames. numframes: 4

INFO | ??:??:?? 59 [DeviceTelemetryModule] Sending packet to mesh

DEBUG | ??:??:?? 59 [DeviceTelemetryModule] Update DB node 0xa3f991ec, rx_time=0

ERROR | ??:??:?? 59 [DeviceTelemetryModule] Error=4, returning NAK and dropping packet.

ERROR | ??:??:?? 59 [DeviceTelemetryModule] Alloc an err=4,to=0xa3f991ec,idFrom=0x662e2e4d,id=0x662e2e4e

DEBUG | ??:??:?? 59 [DeviceTelemetryModule] Enqueued local (id=0x662e2e4e fr=0xec to=0xec, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=662e2e4d priority=120)
```

fifieldt, 2024-09-16:
Hi @SteveHawk , is the latest firmware better?

SteveHawk, 2024-09-22:
@fifieldt Hi, haven't checked in a while, so I try flashed both 2.5.1 and 2.4.2 onto my T-Beam, and the described issue still exists.

fifieldt, 2024-09-23:
It looks like it's trying to initialise a radio that doesn't speak that frequency. Will need to check the logic in the code.

fifieldt, 2025-06-19:
Thanks! Looking up

src/mesh/RF95Interface.cpp:252

fifieldt, 2025-06-19:
Confirmed. It failed setting the power.

@SteveHawk , could you try setting the power much lower and see if it works?

SteveHawk, 2025-06-19:
Yes, it does appear to be the power issue.

The default power value is 30dBm, and I tried to set it as 3dBm CN470, this time it successfully initialized. Then I tried to set it back to 30dBm, and it got back into the critical error # 3 state.

Then I tried the workaround, and apparently somewhere down the chain the power is automatically reduced to 12dBm when using EU433, even when I have it set to 30dBm. So changing it to CN470 is no problem afterwards.

So what's the limitation here? Is this a sx1278 chip limitation?

SteveHawk, 2025-06-19:
Tried a few power values around, the upper limit appears to be 17dBm. Which seems to align with some info about ax1278 on the internet.

fifieldt, 2025-06-19:
OK, so, tracing power settings ....

what's happening is that the power is being limited to the max power in the region(19), rather than the max power of the radio (17). It looks like we assumed that the region always allows less than the radio, which is not true in this case.

Fixable.

Notes:

```
bool RF95Interface::init()
    RadioLibInterface::init(); --> didn't check yet.

    limitPower(); ---> 

void RadioInterface::limitPower()
{
    uint8_t maxPower = 255; // No limit

    if (myRegion->powerLimit)  ---> == 19
        maxPower = myRegion->powerLimit; == 19

    if ((power > maxPower) && !devicestate.owner.is_licensed) { --> trying for 30 here means power is set to 19
        LOG_INFO("Lower transmit power because of regulatory limits");
        power = maxPower;
    }

    if (TX_GAIN_LORA > 0) { ---> 0, does not run.
        LOG_INFO("Requested Tx power: %d dBm; Device LoRa Tx gain: %d dB", power, TX_GAIN_LORA);
        power -= TX_GAIN_LORA;
    }

    LOG_INFO("Final Tx power: %d dBm", power);
}
```

Now back to init: 

    if (power > RF95_MAX_POWER) // This chip has lower power limits than some --> 19 > 
        power = RF95_MAX_POWER;

NomDeTom, 2025-08-12:
@fifieldt how close did you get to a resolution?

@ndoo you've done a lot of work with power level setting. Do the notes above make sense?

ndoo, 2025-08-12:
This should have been fixed since https://github.com/meshtastic/firmware/pull/7065 - specifically this commit and line: https://github.com/meshtastic/firmware/pull/7065/commits/9ba4d8068934c5ad219ab0b3c1577802523b48bb#diff-ad55475cca6c487a8b020a5f8036f160cd1fc92cce2ea887bfe83f284a6fc855

ndoo, 2025-08-12:
Is it an SX1268, SX1278, etc.? I can't find any information about it being limited to 17dBm.

At any rate, it seems like it can't be autodetected well unless `#define RF95_MAX_POWER` is set lower at 17dBm in the variant, but then this will affect the HF RFM95 modules.

SteveHawk, 2025-08-12:
I found this discussion thread mentioning sx1278 being limited to 17dbm and such: https://github.com/jgromes/RadioLib/discussions/308

I'm not sure, but could this relate to current limit? I checked the datasheet on semtech's website here: <https://www.semtech.com/products/wireless-rf/lora-connect/sx1278>, and it says sx1278 can do 20dbm but current is going to be 120mA, and below that is 17dbm@87mA.

[image]

ndoo, 2025-08-12:
> > Does it initialize correctly once a lower txpower is set?
> 
> Yes, manually set it to 17dbm and it initializes correctly. Any value above that (18/19) it's not happy and errors out.
> 
> And it does seem to relate to power limit somehow, I see this line in boot log: `Current limit set to 100.000000` (And also in all previous logs)
> 
> ```
> INFO  | ??:??:?? 1 Start meshradio init
> INFO  | ??:??:?? 1 Radio freq=478.875, config.lora.frequency_offset=0.000
> INFO  | ??:??:?? 1 Set radio: region=CN, name=LongFast, config=0, ch=35, power=17
> INFO  | ??:??:?? 1 myRegion->freqStart -> myRegion->freqEnd: 470.000000 -> 510.000000 (40.000000 MHz)
> INFO  | ??:??:?? 1 numChannels: 160 x 250.000kHz
> INFO  | ??:??:?? 1 channel_num: 36
> INFO  | ??:??:?? 1 frequency: 478.875000
> INFO  | ??:??:?? 1 Slot time: 28 msec
> INFO  | ??:??:?? 1 Final Tx power: 17 dBm
> DEBUG | ??:??:?? 1 Current limit set to 100.000000
> DEBUG | ??:??:?? 1 Current limit set result 0
> INFO  | ??:??:?? 1 RF95 init result 0
> INFO  | ??:??:?? 1 Frequency set to 478.875000
> INFO  | ??:??:?? 1 Bandwidth set to 250.000000
> INFO  | ??:??:?? 1 Power output set to 17
> INFO  | ??:??:?? 1 RF95 init success
> ```
> 
> But I do remember way back in the early days when I bought the unit, everything works. I might try to flash a very old firmware (2.2 maybe?) and check the log see what's the difference.

If you are [set up to build firmware](https://meshtastic.org/docs/development/firmware/build/), try changing this line to something else, e.g. 120:
https://github.com/meshtastic/firmware/blob/e3dd8164a423a12d88232475a529233144d775b0/src/mesh/RadioLibRF95.h#L71

This was changed in 2021 so even a relatively old version of Meshtastic from after will probably show the same symptoms you are experiencing now. https://github.com/meshtastic/firmware/commit/1a671f287753e9140b72dd8bd83676766059fa0b

SteveHawk, 2025-08-12:
2.0.20 works

```log
INFO  | ??:??:?? 1 Starting meshradio init...
INFO  | ??:??:?? 1 Radio freq=478.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 1 Set radio: region=CN, name=LongFast, config=0, ch=35, power=19
INFO  | ??:??:?? 1 Radio myRegion->freqStart -> myRegion->freqEnd: 470.000000 -> 510.000000 (40.000000 mhz)
INFO  | ??:??:?? 1 Radio myRegion->numChannels: 160 x 250.000kHz
INFO  | ??:??:?? 1 Radio channel_num: 35
INFO  | ??:??:?? 1 Radio frequency: 478.875000
INFO  | ??:??:?? 1 Slot time: 42 msec
INFO  | ??:??:?? 1 Set radio: final power level=19
DEBUG | ??:??:?? 1 Current limit set to 100.000000
DEBUG | ??:??:?? 1 Current limit set result 0
INFO  | ??:??:?? 1 RF95 init result -13
INFO  | ??:??:?? 1 Frequency set to 478.875000
INFO  | ??:??:?? 1 Bandwidth set to 250.000000
INFO  | ??:??:?? 1 Power output set to 19
DEBUG | ??:??:?? 1 Current limit set to 100.000000
DEBUG | ??:??:?? 1 Current limit set result 0
INFO  | ??:??:?? 1 RF95 Radio init succeeded, using RF95 radio
```

2.0.21 doesn't work

```log
INFO  | ??:??:?? 1 Starting meshradio init...
INFO  | ??:??:?? 1 Radio freq=478.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 1 Set radio: region=CN, name=LongFast, config=0, ch=35, power=19
INFO  | ??:??:?? 1 Radio myRegion->freqStart -> myRegion->freqEnd: 470.000000 -> 510.000000 (40.000000 mhz)
INFO  | ??:??:?? 1 Radio myRegion->numChannels: 160 x 250.000kHz
INFO  | ??:??:?? 1 Radio channel_num: 35
INFO  | ??:??:?? 1 Radio frequency: 478.875000
INFO  | ??:??:?? 1 Slot time: 42 msec
INFO  | ??:??:?? 1 Set radio: final power level=19
INFO  | ??:??:?? 7 SX126x init result -705
INFO  | ??:??:?? 7 Frequency set to 478.875000
INFO  | ??:??:?? 7 Bandwidth set to 250.000000
INFO  | ??:??:?? 7 Power output set to 19
DEBUG | ??:??:?? 12 Current limit set to 140.000000
DEBUG | ??:??:?? 12 Current limit set result -705
INFO  | ??:??:?? 17 Set Rx Power Saving Gain mode; result: 64831
WARN  | ??:??:?? 17 Failed to find SX1262 radio
WARN  | ??:??:?? 17 SX126xInterface(cs=18, irq=33, rst=23, busy=32)
INFO  | ??:??:?? 17 Starting meshradio init...
INFO  | ??:??:?? 17 Radio freq=478.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 17 Set radio: region=CN, name=LongFast, config=0, ch=35, power=19
INFO  | ??:??:?? 17 Radio myRegion->freqStart -> myRegion->freqEnd: 470.000000 -> 510.000000 (40.000000 mhz)
INFO  | ??:??:?? 17 Radio myRegion->numChannels: 160 x 250.000kHz
INFO  | ??:??:?? 17 Radio channel_num: 35
INFO  | ??:??:?? 17 Radio frequency: 433.174988
INFO  | ??:??:?? 17 Slot time: 42 msec
INFO  | ??:??:?? 17 Set radio: final power level=19
INFO  | ??:??:?? 23 SX126x init result -705
INFO  | ??:??:?? 23 Frequency set to 433.174988
INFO  | ??:??:?? 23 Bandwidth set to 250.000000
INFO  | ??:??:?? 23 Power output set to 19
DEBUG | ??:??:?? 28 Current limit set to 140.000000
DEBUG | ??:??:?? 28 Current limit set result -705
INFO  | ??:??:?? 33 Set Rx Power Saving Gain mode; result: 64831
WARN  | ??:??:?? 33 Failed to find SX1268 radio
INFO  | ??:??:?? 33 Not using WIFI
ERROR | ??:??:?? 33 NOTE! Recording critical erro?
??src/main.cpp:513
INFO  | ??:??:?? 33 PowerFSM init, USB power=1
INFO  | ??:??:?? 33 Enter state: BOOT
[ 33292][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

So something in between probably breaks it. https://github.com/meshtastic/firmware/compare/v2.0.20.7100416...v2.0.21.83e6cea

SteveHawk, 2025-08-13:
Another related report: https://github.com/meshtastic/firmware/issues/6261

TTGO Lora32 v2.1-1.6.1 is also using sx1276/1278.

SteveHawk, 2025-09-03:
Spent some time digging into the codes, and realized something very weird.

The 17dbm limit actually comes from radiolib (RadioLib/src/modules/SX127x/SX1278.cpp, `setOutputPower`/`checkOutputPower`), it only allows output power of <17dbm or 20dbm, not in between (ie 18/19dbm). This behavior has been there since many years ago. After temporarily lifting the 19dbm regulatory limit in firmware, I successfully init'd tbeam's sx1278 at 17dbm and 20dbm.

And the above "success" logs @19dbm all show this line: `RF95 init result -13`, which means `RADIOLIB_ERR_INVALID_OUTPUT_POWER`. However, somehow versions before v2.0.20 ignores this and thinks it inits successfully, and also is somehow able to actually use the radio in that way.

Not entirely sure yet, but my gut feeling is that it worked before v2.0.20 because of some kind of bug+bug=works magic, and v2.0.21 fixed one of the bug and breaks it.

SteveHawk, 2025-09-03:
For sx1278 yes, I think so.
(maybe also sx1276 too, seems like codes in radiolib inherits from 1278)