meshtastic/firmware #3799: [Bug]: CH341 + SX1262 Packet size limited

Pigpog, 2024-05-05:
### Category

Hardware Compatibility, Other

### Hardware

Other

### Firmware Version

2.3.9.70712d85

### Description

Known issue but can't find an existing GH issue for this. Not sure what the status is on resolution.

I have been trying to run Meshtastic on Linux with my PineDIO USB adapter, which uses CH341 & SX1262.
I noticed the [footnote](https://meshtastic.org/docs/hardware/devices/linux-native-hardware/#user-content-fnref-2) about these devices being unable to send more than a few bytes at a time.

My experience with the PineDIO LoRa USB tells me that this bug is due to writing too many bytes to the SX1262 at once.
I have used a different program with the adapter that has overcome this issue by transferring blocks of ten bytes to the SX1262 at a time, waiting for the device to stop being busy before writing the next block.
Here is the patch solving it for that program for reference: https://codeberg.org/JF002/pinedio-lora-driver/issues/5
I recall that the symptoms of the unpatched program are similar to what I'm seeing with Meshtastic . Though it's entirely possible I am encountering an unrelated issue.

I'm not (yet) familiar enough with the Meshtastic source code to make any progress myself.
I am quite motivated to get this working, I will gladly contribute logs, retrace my steps if needed, and answer any questions.
Spam me about this!

### Relevant log output

```Shell
Portduino is starting, HWID=1, VFS root at /root/.portduino/default
Setting up Meshtastic on Portduino...
Using local config.yaml as config file
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  /. (0 Bytes)
DEBUG | ??:??:?? 0  /config.yaml (2823 Bytes)
DEBUG | ??:??:?? 0  /.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (91 Bytes)
DEBUG | ??:??:?? 0  /prefs/.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (506 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (104 Bytes)
INFO  | ??:??:?? 0 No I2C device configured, skipping.
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 Meshtastic hwvendor=37, swver=2.3.9.70712d85
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded /prefs/db.proto successfully
INFO  | ??:??:?? 0 Loaded saved devicestate version 22, with nodecount: 1
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded /prefs/config.proto successfully
INFO  | ??:??:?? 0 Loaded saved config version 22
INFO  | ??:??:?? 0 Loading /prefs/module.proto
INFO  | ??:??:?? 0 Loaded /prefs/module.proto successfully
INFO  | ??:??:?? 0 Loaded saved moduleConfig version 22
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
INFO  | ??:??:?? 0 File /oem/oem.proto not found
DEBUG | ??:??:?? 0 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 0 Using nodenum 0x20981da7 
DEBUG | ??:??:?? 0 Expanding short PSK #0
INFO  | ??:??:?? 0 Wanted region 1, using US
DEBUG | ??:??:?? 0 Set Timezone to GMT0
[... 433 lines cut ...]
DEBUG | ??:??:?? 95 [ServerAPI] Setting local position: latitude=0, longitude=0, time=1714938583, timestamp=0
DEBUG | ??:??:?? 95 [ServerAPI] POSITION node=20981da7 l=5 latI=0 lonI=0 msl=0 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=0 fxq=0 fxt=0 pts=0 time=1714938583
DEBUG | ??:??:?? 95 [ServerAPI] Upgrading time to quality NTP
DEBUG | 19:49:43 95 [ServerAPI] Read RTC time as 1714938583
DEBUG | 19:49:43 95 [ServerAPI] updatePosition SPECIAL time setting time=1714938583
DEBUG | 19:49:43 95 [ServerAPI] Node status update: 0 online, 1 total
DEBUG | 19:49:43 95 [ServerAPI] Module 'position' considered
DEBUG | 19:49:43 95 [ServerAPI] Module 'routing' wantsPacket=1
INFO  | 19:49:43 95 [ServerAPI] Received routing from=0x0, id=0xff61686a, portnum=3, payloadlen=5
DEBUG | 19:49:43 95 [ServerAPI] Routing sniffing (id=0xff61686a fr=0x00 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3)
DEBUG | 19:49:43 95 [ServerAPI] Module 'routing' considered
DEBUG | 19:49:43 95 [ServerAPI] localSend to channel 0
DEBUG | 19:49:43 95 [ServerAPI] Add packet record (id=0xff61686a fr=0x00 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3)
DEBUG | 19:49:43 95 [ServerAPI] Expanding short PSK #0
DEBUG | 19:49:43 95 [ServerAPI] Using AES0 key!
DEBUG | 19:49:43 95 [ServerAPI] Installing AES0 key!
DEBUG | 19:49:43 95 [ServerAPI] enqueuing for send (id=0xff61686a fr=0xa7 to=0xff, WantAck=0, HopLim=3 Ch=0xa encrypted hopStart=3)
DEBUG | 19:49:43 95 [ServerAPI] txGood=0,rxGood=0,rxBad=0
INFO  | 19:49:43 95 [ServerAPI] Disconnecting from phone
INFO  | 19:49:43 95 [ServerAPI] Client dropped connection, suspending API service
WARN  | 19:49:43 95 [RadioIf] Can not send yet, busyTx
ERROR | 19:49:43 95 [RadioIf] Hardware Failure! busyTx for more than 60s
ERROR | 19:49:43 95 [RadioIf] NOTE! Recording critical error 8 at src/mesh/RadioLibInterface.cpp:94
ERROR | 19:49:43 95 [RadioIf] A critical failure occurred, portduino is exiting...
```

jp-bennett, 2024-05-05:
I would love to get this working. I'll see about adapting this code.

markbirss, 2024-05-06:
@Pigpog 

Your radio not init - error -1 ?

ERROR | ??:??:?? 30 \[RadioIf\] Radiolib error -1 when attempting SX126X scanChannel!

Below my DIY-CH341 SX1262 log with 

Logging:   
 LogLevel: debug  
 

```
Setting up Meshtastic on Portduino...
Using /usr/sbin/config.yaml as config file
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (94 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (699 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (107 Bytes)
DEBUG | ??:??:?? 0  /.. (0 Bytes)
DEBUG | ??:??:?? 0  /. (0 Bytes)
INFO  | ??:??:?? 0 Using �˓� as I2C device.
INFO  | ??:??:?? 0 Scanning for i2c devices...
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 Meshtastic hwvendor=37, swver=2.3.7.95e9b2e0
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded /prefs/db.proto successfully
INFO  | ??:??:?? 0 Loaded saved devicestate version 22, with nodecount: 3
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded /prefs/config.proto successfully
INFO  | ??:??:?? 0 Loaded saved config version 22
INFO  | ??:??:?? 0 Loading /prefs/module.proto
INFO  | ??:??:?? 0 Loaded /prefs/module.proto successfully
INFO  | ??:??:?? 0 Loaded saved moduleConfig version 22
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
INFO  | ??:??:?? 0 File /oem/oem.proto not found
DEBUG | ??:??:?? 0 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 0 Using nodenum 0x67d60b4d 
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 3, using EU_868
DEBUG | ??:??:?? 0 [DeviceUI] LVGL: allocating 499840 bytes heap memory for draw buffer
DEBUG | ??:??:?? 0 [DeviceUI] DeviceScreen::init()...
DEBUG | ??:??:?? 0 [DeviceUI] TFTView_320x240 init...
DEBUG | ??:??:?? 0 [DeviceUI] X11Driver::init...
DEBUG | ??:??:?? 0 [DeviceUI] DisplayDriver init...
DEBUG | ??:??:?? 0 [DeviceUI] LV init...
DEBUG | ??:??:?? 0 [DeviceUI] init_hal...
DEBUG | ??:??:?? 0 [DeviceUI] ui_init...
DEBUG | ??:??:?? 0 [DeviceUI] DeviceScreen::init() done.
DEBUG | ??:??:?? 0 Set Timezone to GMT0
DEBUG | ??:??:?? 0 Read RTC time as 1714983483
DEBUG | ??:??:?? 0 NeighborInfoModule is disabled
INFO  | ??:??:?? 0 File /prefs/cannedConf.proto not found
INFO  | ??:??:?? 0 CannedMessageModule is enabled
DEBUG | ??:??:?? 0 Attempting to activate sx1262 radio on SPI port /dev/spidev0.1
DEBUG | ??:??:?? 0 SX126xInterface(cs=0, irq=6, rst=2, busy=4)
DEBUG | ??:??:?? 0 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO  | ??:??:?? 0 Starting meshradio init...
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
[... 186 lines cut ...]
INFO  | ??:??:?? 35 [Router] Received routing from=0x818c6ff3, id=0x3d542195, portnum=4, payloadlen=44
DEBUG | ??:??:?? 35 [Router] Routing sniffing (id=0x3d542195 fr=0xf3 to=0x4d, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 requestId=6b8b4569 rxSNR=7 rxRSSI=-46)
DEBUG | ??:??:?? 35 [Router] Received an ack for 0x6b8b4569, stopping retransmissions
DEBUG | ??:??:?? 35 [Router] Delivering rx packet (id=0x3d542195 fr=0xf3 to=0x4d, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 requestId=6b8b4569 rxSNR=7 rxRSSI=-46)
DEBUG | ??:??:?? 35 [Router] Update DB node 0x818c6ff3, rx_time=0
DEBUG | ??:??:?? 35 [Router] Forwarding to phone (id=0x3d542195 fr=0xf3 to=0x4d, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 requestId=6b8b4569 rxSNR=7 rxRSSI=-46)
DEBUG | ??:??:?? 35 [Router] Module 'routing' considered
INFO  | ??:??:?? 35 Telling client we have new packets 3
INFO  | ??:??:?? 35 [PacketAPI] getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 35 [PacketAPI] phone downloaded packet (id=0x3d542195 fr=0xf3 to=0x4d, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 requestId=6b8b4569 rxSNR=7 rxRSSI=-46)
DEBUG | ??:??:?? 35 [PacketAPI] encoding toPhone packet to phone variant=2, 90 bytes
DEBUG | ??:??:?? 35 [DeviceUI] handleFromRadio variant 2
DEBUG | ??:??:?? 35 [DeviceUI] received packet from 0x818c6ff3(2173464563), portnum=4
DEBUG | ??:??:?? 36 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=66, time 722 ms
DEBUG | ??:??:?? 36 [RadioIf] Lora RX (id=0x6b8b4569 fr=0x4d to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=6 rxRSSI=-46)
DEBUG | ??:??:?? 36 [RadioIf] AirTime - Packet received : 722ms
DEBUG | ??:??:?? 36 [Router] Rx someone rebroadcasting for us (id=0x6b8b4569 fr=0x4d to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=6 rxRSSI=-46)
DEBUG | ??:??:?? 36 [Router] didn't find pending packet
DEBUG | ??:??:?? 36 [Router] Found existing packet record for fr=0x67d60b4d,to=0xffffffff,id=0x6b8b4569
DEBUG | ??:??:?? 36 [Router] Found existing packet record for fr=0x67d60b4d,to=0xffffffff,id=0x6b8b4569
DEBUG | ??:??:?? 36 [Router] Add packet record (id=0x6b8b4569 fr=0x4d to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=6 rxRSSI=-46)
DEBUG | ??:??:?? 36 [Router] Ignoring incoming msg, because we've already seen it (id=0x6b8b4569 fr=0x4d to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=6 rxRSSI=-46)
DEBUG | ??:??:?? 36 [Router] cancelSending id=0x6b8b4569, removed=0
DEBUG | ??:??:?? 36 [Router] Incoming message was filtered 0x67d60b4d
```

Pigpog, 2024-05-06:
@markbirss This is on the PineDIO hardware? Mind sharing which CH341 driver you used, mine might be problematic.
It's also worth mentioning that im not entirely sure of the pins defined in my config.yaml. So if you have a working setup, I'd love to try and replicate it.
Thanks!

ndoo, 2024-05-07:
I'm planning to design and build an EBYTE E22 to USB board for use with Portduino Meshtastic. 

Do we know if this issue is specific to the CH341 (i.e. does not happen on other vendors chips, e.g. FT232H, MCP2210)?

If there isn't anything conclusive, then perhaps I am better off buying a Pi Zero and wiring to the hardware SPI.

markbirss, 2024-05-07:
@Pigpog 

No, Pine dont ship to my country. Using the DIP CH341 package IC, pinedio uses the smaller ic

driver: https://github.com/frank-zago/ch341-i2c-spi-gpio

Im using any one of these 3 below wired to various LoRa Modules

[image]
[image]
[image]

markbirss, 2024-05-07:
@ndoo 

issues confirmed both on PineDio and DIY-CH341 regarding only able to send around 50 chars reliably. 

Yes, Raspberry Pi at the moment the better compatible option

Not been able to find good support for either FT232H, MCP2210

But im am using e-ink/tft on FT232H with Adafruit Blinka
[image]

Pigpog, 2024-05-07:
I investigated my software more and found that my pins were definitely set wrong. 
After configuring it properly, my PineDIO USB can successfully be used with the firmware.

I am now encountering the packet size limit issue. Apologies for the unrelated discussion my configuration caused.

Pigpog, 2024-05-09:
I have managed to apply a workaround that solves this bug.
The single file I modified is actually in RadioLib's source tree, not Meshtastic.
I'm not sure how this could be upstreamed, but here is my hacky patch.
RadioLib is not tracked by git in Meshtastic's firmware tree, so you'll wanna apply this by hand.

```patch
diff --git a/RadioLib/src/Module.cpp b/RadioLib/src/Module.cpp
index c46e26e9..a5a4c85e 100644
--- a/RadioLib/src/Module.cpp
+++ b/RadioLib/src/Module.cpp
@@ -304,7 +304,35 @@ int16_t Module::SPItransferStream(uint8_t* cmd, uint8_t cmdLen, bool write, uint
   // do the transfer
   this->hal->spiBeginTransaction();
   this->hal->digitalWrite(this->csPin, this->hal->GpioLevelLow);
-  this->hal->spiTransfer(buffOut, buffLen, buffIn);
+
+
+  uint16_t offset = 0;
+
+  while (buffLen) {
+    uint8_t block_size = (buffLen < 10 ? buffLen : 10);
+
+    this->hal->spiTransfer((buffOut != NULL ? buffOut + offset : NULL) , block_size, (buffIn != NULL ? buffIn + offset : NULL));
+
+    // ensure GPIO is low
+    if(this->gpioPin == RADIOLIB_NC) {
+      this->hal->delay(1);
+    } else {
+      uint32_t start = this->hal->millis();
+      while(this->hal->digitalRead(this->gpioPin)) {
+        this->hal->yield();
+        if(this->hal->millis() - start >= timeout) {
+          RADIOLIB_DEBUG_BASIC_PRINTLN("GPIO mid-transfer timeout, is it connected?");
+          #if !RADIOLIB_STATIC_ONLY
+            delete[] buffOut;
+            delete[] buffIn;
+          #endif
+          return(RADIOLIB_ERR_SPI_CMD_TIMEOUT);
+        }
+      }
+    }
+    offset += block_size;
+    buffLen -= block_size;
+  }
   this->hal->digitalWrite(this->csPin, this->hal->GpioLevelHigh);
   this->hal->spiEndTransaction();
```

jp-bennett, 2024-05-09:
@Pigpog This is excellent. I bet we can actually introduce an equivalent in our HAL code at https://github.com/meshtastic/firmware/blob/0c89aff0f64d7d93321238222d27b2fb32ad4c3f/src/mesh/RadioLibInterface.cpp#L28

Pigpog, 2024-05-10:
@jp-bennett Not sure. The fix has to be pretty low-level, since it needs to check if the BUSY pin is high between data chunks. Maybe the `canSendImmediately` function performs the same check?

jp-bennett, 2024-05-10:
> @jp-bennett Not sure. The fix has to be pretty low-level, since it needs to check if the BUSY pin is high between data chunks. Maybe the `canSendImmediately` function performs the same check?

https://github.com/meshtastic/firmware/pull/3847 is my initial stab at it. Not yet tested on real hardware, will break any other radio type.

markbirss, 2024-05-10:
@Pigpog 

Thank you, I was able to test the hotfix @jp-bennett made with my own DIY hardware and it seems to be working in one of the scenarios where my device is configured on /dev/spi0.1 does not work on /dev/spi0.0 (handling of CS by kernel/radiolib)

jp-bennett, 2024-05-10:
I think this is fixed, then with #3847 . Feel free to re-open if it continues to be a problem.

closed 2024-05-10 by jp-bennett

Pigpog, 2024-05-10:
Thanks, glad you were able to find a user-friendly way to implement it!

I did some testing, and it definitely helps but I still can't send messages longer than ~103 bytes (Meshtastic header and all) on LongFast, or ~150 on ShortFast. Messages that long do not get received on my T-Beam.

Turns out that my original modification couldn't do it either, and even my simple reimplementation of Meshtastic can't do it. Shorter messages work reliably though.

I have an SDR, so I'd be able to analyze any physical layer issues. But I think i'll come back to this later.

andrewthetechie, 2024-06-08:
@Pigpog any luck on resolving those message size limits?

Pigpog, 2024-06-08:
@andrewthetechie Unfortunately no. After using the USB for a short time, Meshtastic always crashes. I tried quite a few different configurations and code changes with no luck.
Making matters worse, each time Meshtastic crashes, it leaves my ch341 driver in an unrecoverable state, forcing me to reboot after each failed attempt. 

With transmit disabled, it has run error-free for hours. The crashing might be a symptom of this packet length issue.

I will run some more tests now and share the configurations and logs in a new comment.

andrewthetechie, 2024-06-08:
Well drat. I've got a PineDIO sitting around. It would be cool to run meshtastic in my k8s cluster with the USB radio.

I'll follow this issue, but I'm out of my depth on actually trying to fix the issue :)

Pigpog, 2024-06-08:
I fetched master and tested it. Configuration info and logs here: https://pastebin.com/yjKYERJb
I did modify RadioLibInterface to send chunks of 10 bytes over SPI instead of 20, since this is what I use for my other PineDIO application. It crashes regardless.
During my test, no other Meshtastic devices were present, and I was simply commanding it to send "test" and "test message" over the default channel with default configuration.
It crashed in minutes with Radiolib error -705 (RADIOLIB_ERR_SPI_CMD_TIMEOUT), which I saw very often during my previous tests.