meshtastic/firmware #3806: Unsanitized MQTT password

grizleeeee, 2024-05-06:
### Category

Other

### Hardware

T-Lora v2 1.6, T-Deck, T-Echo, Rak4631, Heltec V3

### Firmware Version

2.3.6, 2.3.7

### Description

When using a password that only used pipes as separators, the entire MQTT settings page became unreadable to iOS clients, as well as sometimes the WebUI (not always reproducible, unclear what would cause this).  Furthermore, the device was unable to utilize MQTT to connect.  The settings were always visible from the command line, but inside the iOS app, the settings were unreadable and greyed out. The screen wouldn't even scroll.  All other settings were accessible. 

The obvious issues with the mobile/web client and it actually breaking the functionality were resolved immediately once the password was changed to something other than one that closely resembles a RegEx query. 

Deprecated Password that caused the issue: `CREED|hunter|AMPLE|outlet|STEED`

garthvh, 2024-05-06:
iOS could care less about pipes so not sure that is the actual issue, oddly it will save CREED|hunter|AMPLE|outlet|STEE

grizleeeee, 2024-05-07:
So the password would save when set from the CLI, but then the MQTT settings page in the iOS client would be unreadable and completely greyed out.  I've attached an example of the settings being inaccessible, and it appears to be isolated to the MQTT pane.  https://github.com/meshtastic/firmware/assets/2353329/1b45c0f6-a6d3-42d6-8b1d-0c23e3ef0773

fifieldt, 2024-09-17:
Can't reproduce with 2.5.0 && Android - saves correctly.

garthvh, 2024-09-17:
I think this was reported when we increased the size

closed 2024-09-17 by garthvh