meshtastic/firmware #3810: [Feature Request]: Include a web server in the Pico W firmware

clwgh, 2024-05-06:
### Platform

ESP32

### Description

This is a feature request for the Pico W RP2040 (not ESP32, dropdown had no option).

I have a Pico W based node, in wifi range but away from the computer. It would be very useful to have access to the web client over wifi, as one can with (say) a Heltec. I believe the firmware supports the web client but there is no web server running to be able to access it.

As it stands the only way to interact nicely with the Pico W is to connect to to a Chromium-based browser and use the hosted web client to access it over serial.

Is it possible to enable a simple web server in order to access the web client over wifi?

madeofstown, 2024-05-06:
Would also be nice if the Pico W with WiFi enabled would automatically create a hotspot if the preferred SSID isn't in range for a while. I realize that it can be managed over the admin channel but I've found the admin channel a bit unreliable.

clwgh, 2024-05-06:
> Would also be nice if the Pico W with WiFi enabled would

Would you be able to open that as a separate feature request please? I want to keep this one focused on this specific request. Thanks.

madeofstown, 2024-05-07:
> 
> As it stands the only way to interact nicely with the Pico W is to connect to to a Chromium-based browser and use the hosted web client to access it over serial.
> 

You can also use the Android app to connect to the Pico W with it's IP. I'm not sure about iOS because I heard in a video that it was Bluetooth only, but I'm not sure if that is still the case, the video was from a few months ago.