meshtastic/firmware #3811: [Feature Request]: Support Ublox M9N

hdngr, 2024-05-06:
### Platform

NRF52, ESP32

### Description

# Overview
Canaryone is designed with the Ublox M9N.

# Problem
M6,7,8,9 and especially M10 all run different versions of firmware, and may communicate with dramatically different protocols.

Currently, the GPS module handles all chips that are `!M10`.

# Solution
Refactor code to selectively apply commands to chips given there version and protocol needs.

**As of May 6th, 2024 - DRAFT PRs to follow**

hdngr, 2024-05-06:
cc @lolsborn @GPSFan

GPSFan, 2024-05-07:
The u-blox standard precision product line can be roughly divided into 2 categories, Pre F9/M9 and post F9/M9. The F9/M9 generation (Chip type 9140) introduced the VALSET/VALGET/VALDEL key value pair configuration framework. It also retained some of the older configuration protocol. As the newer versions of the protocol were introduced, the older configuration method has been deprecated and in newer generations much has been removed. The Neo-M9N firmware is currently 4.04, and hasn't been updated since lat Sept, 2020.
The M9 is a hungry beast, but also is a very good receiver, combining an M9 and an NRF52 is somewhat puzzling as even in power save mode the M9 consumes more than an M10 in full acquisition mode.
The current GPS.cpp handles u-blox generations 6,7,8 and 10. Although the M10 support is not as complete as I would like it to be, and the support has been sort of "patched together" over time. A more structured approach might be a goal for Meshtastic V3. There is also the support for non-u-blox receivers, as they can be much more cost effective GNSS solutions, and there are a lot of variations in those receivers.

thebentern, 2024-10-07:
Closing this for now

closed 2024-10-07 by thebentern