meshtastic/firmware #3813: [Feature Request]: Implement sane MQTT defaults (PR already pending)

tropho23, 2024-05-07:
### Platform

NRF52, ESP32

### Description

Request making the default for MQTT to 'ignore MQTT' and maybe even set hop count for inbound MQTT packets to zero, if that is possible. The core issue here is people turning on MQTT, then either forgetting it's enabled or not caring about the impact it has to not only their local mesh but anyone connected to it via MQTT.

We need MQTT to be a tool for testing, a fun feature, and a nice capability to help join disparate meshes; right now without the proper attention and deliberate actions MQTT becomes a liability and burden for many users.  Some boards still get overloaded and freeze/reset due to out of control traffic caused by mesh flooding by MQTT packets.

I think this might already be addressed by @GUVWAF's PR that is still pending review:
https://github.com/meshtastic/firmware/pull/3591

thebentern, 2024-05-07:
Yes, this one is tricky because even in the US without the duty cycle default of off, we get a ton of support requests where users are confused by MQTT not working once enabled, not realizing the Channel up/downlink settings must be enabled first. Heck, it gets me sometimes

GUVWAF, 2024-05-07:
> We've had reports from Germany by people being confused about MQTT not working after they enabled it, because on duty cycle limited regions config.lora.ignore_mqtt defaults to OFF already.

With my PR this actually is solved. Ignore MQTT will be on be default, but MQTT packets will be received if downlink enabled. Only when you disable "ignore MQTT" it will also rebroadcast it over LoRa.

closed 2024-09-16 by fifieldt