meshtastic/firmware #3843: RAK14500 ink display is broken

miodzicho, 2024-05-09:
I habe meshtastic starter, connected RAK14500 with ink display, but it seems all wrong (attached photo). No GPS connected. [image]

caveman99, 2024-05-09:
Looks like you got the three-color version of that device. I don't think this is supported yet.

miodzicho, 2024-05-09:
Seems indeed that could be not supported yet. I hope that it will be supported at some point.

markbirss, 2024-05-09:
> Seems indeed that could be not supported yet. I hope that it will be supported at some point.

Tricolor display can't refresh fast enough to be usable with meshtastic

They don't support partial or fast updates and takes upto 18s to just to full refresh a single page

markbirss, 2024-05-09:
not planned

closed 2024-05-09 by markbirss