meshtastic/firmware #3848: [Bug]: iOS Blutooth - heltec v3 board not discovered at all

JSP6668, 2024-05-10:
### Category

BLE

### Hardware

Heltec V3

### Firmware Version

2.3.7

### Description

Hi All, 
the heltec v3 does not even show up in the iOS Blutooth discovery, on android the same device is dicovered and can be connected successfully. 
Is it possible that this board is not compatible with iOS?

Best regards

caveman99, 2024-05-10:
You can use either wifi or bluetooth. Not both at the same time.

JSP6668, 2024-05-10:
I know, and Wifi is disabled, and I can connect with an Android device via Blutooth.

garethhowell, 2024-05-10:
I know it doesn't help you, but I've flashed two Heltec V3 units in the last couple of days with the latest firmware and both were found immediately by the iOS client.

JSP6668, 2024-05-10:
Thank you, that does help! So it is not incompatible at all.

caveman99, 2024-05-10:
Have you tried unpairing from android (os level, not app) before trying to connect from iOS?

JSP6668, 2024-05-10:
Yes, and I tried it with my ipad and my iphone, no success.
I double checked that it is a heltec v3 device, as well.
It comes up on Linux and macOS

JSP6668, 2024-05-10:
Thank You @garethhowell ,

My fault I alway tried to first pair the device and then I wanted to use the app.
After Your comment I tried the app and bingo it wanted to pair the device!

So it is not a Bug!

closed 2024-05-10 by JSP6668