meshtastic/firmware #3852: [Bug]: Heltec LoRa setting mismatch

memo-567, 2024-05-10:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.9.f06c56a alpha

### Description

After updating to FW 2.3.9.f06c56a Alpha, LoRa setting on Heltec V3 is EU_868 MHz instead of EU_433 MHz.

thebentern, 2024-05-11:
What happens if you set the LoRA region back to EU_433MHz?
I don't understand how this is possible with just a firmware upgrade, especially considering that the part of the codebase that deals with this is unaffected.

memo-567, 2024-05-11:
> "What happens if you set the LoRA region back to EU_433MHz?"

Of course I did that. I will continue to monitor this and inform you after the next update or close this issue.

memo-567, 2024-05-22:
It happened again  ¯\\\_(ツ)_/¯
After updating to Meshtastic Firmware 2.3.10.d19607b Alpha, the LoRa frequency was reset to EU_868 on both Heltecs.
It's not a big deal for me to change it manually again - but maybe there's a better way?

thebentern, 2024-05-22:
I wonder if you downgrade to 2.3.9 again... if you can consistently reproduce. If so, could you grab some serial logs? If you use the web flasher we capture the log output immediately after resetting flashed device.

memo-567, 2024-05-22:
I can do that right away. I'll start the downgrade to 2.3.9 now.

memo-567, 2024-05-22:
Downgrading doesn't affect LoRa settings. I'm going to upgrade again now to 2.3.10.

memo-567, 2024-06-02:
After updating to FW 2.3.11 everything was fine again.

closed 2024-06-02 by memo-567