meshtastic/firmware #3854: [Bug]: Device screen: non-english characters

panayotoff, 2024-05-10:
### Category

Other

### Hardware

T-Beam S3, Heltec V3

### Firmware Version

2.3.6.7a3570a Beta

### Description

When users send non-english characters ( ex. Cyrillic letters ) the device display does not show them. Tested with T-Beam supreme / Heltec v3. Is it possible to have UTF-8 encoding for the screens? 
The iOS/Android app displays properly the characters.

todd-herbert, 2024-05-11:
I remember someone added support for Ukrainian at some point. I think you have to [build the firmware](https://meshtastic.org/docs/development/firmware/build/) yourself, adding an extra line of code to enable it:

[image]
(Screenshot from discord)

thebentern, 2024-05-24:
@todd-herbert is correct. You will have to compile your with the font macro

closed 2024-05-24 by thebentern