meshtastic/firmware #3855: [Bug]: Documentation issue regarding channel keys

KyleMaas, 2024-05-10:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.3.6

### Description

I'm not sure if this is the correct place to post this, but in the documentation for channel keys on this page:

https://meshtastic.org/docs/configuration/radio/channels/

...it refers to keys as 0, 16, or 32 bytes but doesn't specify how they're supposed to be entered.  It has the hex version of the primary default channel ("0x01") right next to a Base64-encoded "AQ==" key but doesn't specify which one you're supposed to use.  It took me a bit of trial and error to generate a 32 byte key, because I assumed it was probably hex and saving the channel would fail and not tell me why.  Once I figured out I needed to Base64-encode it, everything worked fine, but there's nothing in the interface or the docs as far as I've found which tells you this is the format it expects.

garthvh, 2024-05-11:
The apps are synced up so that you can paste keys between the iOS and android app, not sure any users can really type in base64, not sure how useful it is to put in a warning.

KyleMaas, 2024-05-11:
I did.  I wanted to generate my own keys, and was not using the app but the web UI.  There was no error message when I used a key that was too long (hex on my first attempt, so more digits).  Even just a quick clarification in the docs would have been exceptionally helpful.