meshtastic/firmware #3856: [Bug]: T-Beam Supreme No USB CLI

joneyes, 2024-05-10:
### Category

Serial

### Hardware

T-Beam S3

### Firmware Version

2.3.9.f06c56a Alpha

### Description

T-Beam supreme.
Noting this bug fix: Revert "Fix T-Beam S3 USB CLI https://github.com/meshtastic/firmware/issues/3736" by @thebentern in https://github.com/meshtastic/firmware/pull/3809, to fix the loss of CLI comms on 2.3.6 & 2.3.7
Even with this version flashed there is no USB CLI.
As the previous poster noted, reverting to 2.3.4 works
Flashing to 2.3.9 via the webflasher also put the device in a state where the webflasher could not flash the device - i had to flash via CLI to get 2.3.4b on to it.

joneyes, 2024-05-11:
As an additional note - the tbeam is stuck on the comport used for flashing. So stuck in DFU mode?

Running ESP tool detects it on com 27 but doesn't manage to reset it back to the port normally used - com 28 in this case:
This is the output running from being plugged straight in, without holding the user button.

esptool.py v4.7.0
Found 3 serial ports
Serial port COM27
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded Flash 8MB (GD)
Crystal is 40MHz
MAC: 48:27:e2:e5:ff:dc
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-S3 has no Chip ID. Reading MAC instead.
MAC: 48:27:e2:e5:ff:dc
Hard resetting via RTS pin...

mkgin, 2024-05-16:
I have reverted to 2.3.4.

Here is my =dmesg=  plugging in with 2.3.6 after reset. (Where the python cli hangs)  Same behavior with  2.3.9.

```
[1441467.412014] usb 3-1.2: new full-speed USB device number 55 using ehci-pci
[1441467.522433] usb 3-1.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[1441467.522444] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1441467.522449] usb 3-1.2: Product: USB JTAG/serial debug unit
[1441467.522464] usb 3-1.2: Manufacturer: Espressif
[1441467.522466] usb 3-1.2: SerialNumber: __SNIP__
[1441467.522997] cdc_acm 3-1.2:1.0: ttyACM3: USB ACM device
```

After reinstalling 2.3.4

```
[1441492.764044] usb 3-1.2: new full-speed USB device number 56 using ehci-pci
[1441492.879189] usb 3-1.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.00
[1441492.879201] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1441492.879206] usb 3-1.2: Product: LilyGo TBeam-S3-Core
[1441492.879220] usb 3-1.2: Manufacturer: Espressif Systems
[1441492.879222] usb 3-1.2: SerialNumber: __SNIP__
[1441492.880394] cdc_acm 3-1.2:1.0: ttyACM4: USB ACM device
```

And python CLI works with 2.3.4 ... Of course I'd prefer to run a newer version of stable, but I prefer configuring with the CLI more...

oherrala, 2024-05-17:
This might have been fixed in #3912 ?

mkgin, 2024-05-18:
> This might have been fixed in #3912 ?

Seems likely... next release I guess.  (the update bit in purple)

https://github.com/meshtastic/firmware/releases/tag/v2.3.9.f06c56a

joneyes, 2024-05-22:
I can confirm this issue is closed in 2.3.10a

GUVWAF, 2024-06-21:
Seems resolved in 2.3.10.

closed 2024-06-21 by GUVWAF