meshtastic/firmware #3857: [Feature Request]: add support for linux iio (Industrial I/O) subsystem for sensors

Jorropo, 2024-05-11:
### Platform

ESP32

### Description

Linux iio provide many drivers for different kind of sensors:
https://www.kernel.org/doc/html/latest/driver-api/iio/intro.html
> - analog to digital converters (ADCs)
> - accelerometers
> - capacitance to digital converters (CDCs)
> - digital to analog converters (DACs)
> - gyroscopes
> - inertial measurement units (IMUs)
> - color and light sensors
> - magnetometers
> - pressure sensors
> - proximity sensors
> - temperature sensors

It would be nice if sensors that are already supported in meshtastic (pressure, temperature, ...) could fetch data through `/sys/bus/iio` so any pressure or temperature sensor with a linux driver would be usable in meshtastic.

jp-bennett, 2024-05-11:
@Jorropo have a suggested affordable sensor to start with?

Jorropo, 2024-05-11:
There a couple sensors supported by mesthastic already would allow to confirm the feature.
BMP and BME family are the one that come to my mind. https://github.com/torvalds/linux/blob/master/drivers/iio/pressure/bmp280-spi.c#L120-L141

rexbyrns, 2024-06-04:
1-wire is very rich....  I2C to 1-wire bridges are cool..

thebentern, 2024-06-04:
As much as I love the Dallas 1-wire sensors and use them a lot in other projects, we removed support for those in Meshtastic in favor of a rich i2c ecosystem of sensors where we have a clearly defined bus and don't have to add yet-another-gpio pin for a user to hunt down in potentially problematic boards.