meshtastic/firmware #3864: [Bug]: T-Beam S3 Supreme OLED "Shutting down..."

joneyes, 2024-05-11:
### Category

Hardware Compatibility

### Hardware

T-Beam S3

### Firmware Version

2.3.9.f06c56a

### Description

The OLED periodically displays "Shutting down...." and the display is not woken with the user button.
The device is still active and responds to queries on Longfast and can be accessed by ble.
Receiving beacons where the display should wake up with the Rx'd nodes data, the screen does wake up, but does not refresh and just says "Shutting down..."
Device has been factory reset - and configured from defaults, rather than an import.
Power saving is disabled.
Symptom occurs on USB power, or on battery or both
Issuing a reboot will temporarily restore operation.
I see the same issue on v2.3.4
Happy to provide a config file (can't geta yaml file out of it because of the USB CLI bug)

todd-herbert, 2024-05-11:
It sounds a little bit like your user button may be stuck down / shorted? Is there any chance you could test it with a multimeter or anything?

joneyes, 2024-05-11:
I don't think that is the case - as the user button behaves normally when the screen is not showing Shutting down... and toggles through the screens. if triple click is enabled the GPS can be turned on and off, and it beacons with a double click
The board is not cased - yet
I took the core card off and can confirm that the buttons are not shorted.

todd-herbert, 2024-05-12:
Still no luck with the CLI on 2.3.9 hey? Might be best to try troubleshoot on 2.3.4 for now? Do you think you could take a look at the debug output on 2.3.4, to see if there are any clues? There's a serial monitor button at https://flasher.meshtastic.org, if you don't have another preferred method.

I don't personally have this hardware sorry, so it makes it tricky to problem solve, but I'll do what I can!

joneyes, 2024-05-12:
Thanks for the input.
So I have rolled back to 2.3.4 and the problem still exists.
I have attached two log files, and it has taken around 30~40 minutes from start for it to show.
Looking at both logs, the even is logged - but there is no hint as to why.
First log, event is at 10:17:44 1976 (approx. 33 minutes from boot)
Second log, event is at 11:00:16 2318 (approx 38 minutes from start)
I also note, that there doesn't appear to be any logging of the user button presses. In the later log, i have pressed the button several times, to try and wake the screen (which it doesn't succeed in doing). I can see from the log, that the sending of the manual beacon appears to be logged.

[meshtastic-log-2024-05-12T10-18-28.813Z.log](https://github.com/meshtastic/firmware/files/15285967/meshtastic-log-2024-05-12T10-18-28.813Z.log)
[meshtastic-log-2024-05-12T11-04-00.697Z.log](https://github.com/meshtastic/firmware/files/15285968/meshtastic-log-2024-05-12T11-04-00.697Z.log)