meshtastic/firmware #3867: [Bug]: messages between contacts are relayed to mqtt without an opt-in to uplink that destination

quimnut, 2024-05-12:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.3.6

### Description

I have a client_mute mqtt gateway to collect my telemetry's, as it's client_mute it does not route. It has uplink enabled for channel 0 / primary to my local mqtt to process any mentions or actions from public chatter.

When I send a message to a contact (not the channel) (many hops away) from a my client device, which hops through a 3rd device (rooftop with peers, client mode), the client_mute node relays that message to mqtt.

In the logs example the message is from 3204 (3771740676) to cf20 (2086326048) (truncated), the from address d37c is the client_mute mqtt gateway.

I feel choosing to uplink channel 0 should not include anything not addressed to: 0xffffffff (the channel).

Thanks to everyone ,bless.

### Relevant log output

```Shell
client_mute mqtt gateway d37c 
    "role": "CLIENT_MUTE",
    "proxyToClientEnabled": false,
  Index 0: PRIMARY psk=default { "psk": "AQ==", "uplinkEnabled": true, "moduleSettings": { "positionPrecision": 32, "isClientMuted": false }, "channelNum": 0, "name": "", "id": 0, "downlinkEnabled": false }

DEBUG | 01:12:43 21110 [Router] MQTT onSend - Publishing encrypted message
DEBUG | 01:12:43 21110 [Router] MQTT Publish msh/ANZ/2/e/LongFast/!7c5ad37c, 78 bytes
INFO  | 01:12:43 21110 [Router] serialized json message: {"channel":0,"from":3771740676,"hops_away":0,"id":92611553,"rssi":-77,"sender":"!7c5ad37c","snr":6,"timestamp":1715505899,"to":208632
INFO  | 01:12:43 21110 [Router] JSON publish message to msh/ANZ/2/json/LongFast/!7c5ad37c, 148 bytes: {"channel":0,"from":3771740676,"hops_away":0,"id":92611553,"rssi":-77,"sender":"!7c5ad37
```

garthvh, 2024-05-12:
Those are settings for lora re-broadcasting that I don't think are going to have any MQTT impact.

quimnut, 2024-05-12:
I caught up on with discord and we discussed this, discussed this existing draft pr https://github.com/meshtastic/firmware/pull/1509 which is the direction I saw.

Thank you to those that replied on discord too.

closed 2024-05-12 by quimnut