meshtastic/firmware #3875: [Bug]: immediate resume after shutdown

roha-github, 2024-05-13:
### Category

Hardware Compatibility

### Hardware

T-Lora v2 1.6

### Firmware Version

2.3.9.f06c56a

### Description

Device: T-LoRa V2.1-1.6
Firmware: 2.3.9.f06c56a Alpha

Error: immediate resume after shutdown
Expected: shutdown forever (until reset button pressed)

The device does not have a push button, only a reset button and power switch for the battery. The shutdown worked up to version 2.3.4. This issue also discussed at https://meshtastic.discourse.group/t/dodeepsleep-with-msectowake/12338/12

### Relevant log output

```Shell
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (823) esp_core_dump_flash: No core dump▒▒ѥѥ▒▒▒found!
E (824) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
▒▒@INFO  | ??:??:?? 0

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (491520/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (99 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (316 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (106 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-BMhLjTmL.css.gz (16270 Bytes)
DEBUG | ??:??:?? 0  /static/index-Bt2MYVVB.js.gz (203021 Bytes)
DEBUG | ??:??:?? 0  /static/index-CGqDWSD_.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (548 Bytes)
DEBUG | ??:??:?? 0  /static/maplibre-gl-BXZZhNda.js.gz (210451 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (197 Bytes)
[   556][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
DEBUG | ??:??:?? 0 Using analog input 35 for battery level
INFO  | ??:??:?? 0 ADCmod: ADC characterization based on reference voltage stored in eFuse
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   584][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 0 I2C device found at address 0x3c
INFO  | ??:??:?? 0 ssd1306 display found
INFO  | ??:??:?? 0 ssd1306 display found
[... 573 lines cut ...]
INFO  | ??:??:?? 14 From Radio onread
INFO  | ??:??:?? 14 getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 14 encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 14 From Radio onread
INFO  | ??:??:?? 14 getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 14 encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 14 From Radio onread
INFO  | ??:??:?? 14 getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 14 encoding toPhone packet to phone variant=9, 13 bytes
INFO  | ??:??:?? 14 From Radio onread
INFO  | ??:??:?? 14 getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 14 encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 15 From Radio onread
INFO  | ??:??:?? 15 getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 15 encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 15 From Radio onread
INFO  | ??:??:?? 15 getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 15 encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 15 From Radio onread
INFO  | ??:??:?? 15 getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 15 encoding toPhone packet to phone variant=9, 7 bytes
INFO  | ??:??:?? 15 From Radio onread
INFO  | ??:??:?? 15 getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 15 encoding toPhone packet to phone variant=9, 12 bytes
```

todd-herbert, 2024-05-13:
I remember we spoke about this on the [Discourse Server](https://meshtastic.discourse.group/t/dodeepsleep-with-msectowake/12338). Was hoping to hear back from you! I don't have the hardware myself, but I've reached out to hopefully do some testing with someone who does. Still suspecting it's either due to the defined PIN_BUTTON (GPIO12) for this variant requesting an internal pull-up, or maybe due to its role as a bootstrapping pin (fails if pulled HIGH at boot). If all else fails, there may be a case to disable the PIN_BUTTON for this variant, and rely on the reset instead.

todd-herbert, 2024-05-13:
> The shutdown worked up to version 2.3.4

If you're wondering what the change was, the EXT0 interrupt was reassigned to listen for interrupts from the radio during sleep, which meant the button interrupt was shifted to EXT1. I think we've caught most of the edge cases at this point, but seems there's still one or two out there!

roha-github, 2024-05-13:
@Herbert sorry for the late response. I didn't have access to a computer with VS Code this past week to try out your hint. Today I tested the shutdown on different hardware variants on firmware: 2.3.9.f06c56a.

The T-Lora 2.1-1.6 device don't have a push button, only one reset button and power switch. 
https://github.com/LilyGO/TTGO-LORA32/blob/master/schematic1in6.pdf

I picked up project environment "env:tlora-v2-1-1_6" and Line 235 to 238 of sleep.cpp are grayed out.
Line 267 is enabled, meaning BUTTON_PIN is defined in variant.h (BUTTON_PIN 35).

Try comment out line 267 and 268 solve the issue for TLORA_2_1_1P6
```
#ifdef BUTTON_PIN
    // Avoid leakage through button pin
    if (GPIO_IS_VALID_OUTPUT_GPIO(BUTTON_PIN)) {
        // pinMode(BUTTON_PIN, INPUT);
        // gpio_hold_en((gpio_num_t)BUTTON_PIN);
    }
#endif
```
Perhaps a precompiler variable HAS_NO_BUTTON in variant.h would help to exclude all devices without a push button from such operations in future. There may also be (repeater) devices without PUSH_BUTTON in the future.

todd-herbert, 2024-05-13:
> The T-Lora 2.1-1.6 device don't have a push button, only one reset button and power switch.

> Perhaps a precompiler variable HAS_NO_BUTTON in variant.h would help to exclude all devices without a push button from such operations in future. There may also be (repeater) devices without PUSH_BUTTON in the future.

It is perfectly valid for a variant to not have a `PIN_BUTTON` specified, but I'm not actually sure what the original reason for adding this for the T-Lora was, and I'd hate to break someone else's device by removing it. My concern is that some T-Lora users may have DIY builds which do use a button connected to GPIO12.

> Try comment out line 267 and 268 solve the issue for TLORA_2_1_1P6

Just to make sure I understand correctly: you tested commenting out line 267 and 268, and it resolved the issue? If so, that's *very* helpful information.

If commenting out 267 and 268 worked for you, could I possibly bother you for a tiny bit more testing? I would be interested to know if it is specifically line 267 or 268 which is the issue. My hope is that changing line 267 to `INPUT_PULLUP` will resolve the issue both on your device (without a button) and on any other user's builds (which may have a button connected to GPIO12?)

```cpp
#ifdef BUTTON_PIN
    // Avoid leakage through button pin
    if (GPIO_IS_VALID_OUTPUT_GPIO(BUTTON_PIN)) {
        pinMode(BUTTON_PIN, INPUT_PULLUP);
        gpio_hold_en((gpio_num_t)BUTTON_PIN);
    }
#endif
```

I'm also suspicious that [lines 212-214 of src/platform/esp32/main-esp32.cpp](https://github.com/meshtastic/firmware/blob/a9a208de7336903fa466ebd3d749f317c0ae69ce/src/platform/esp32/main-esp32.cpp#L212C1-L214C7) may need to be removed..

I'm sorry to ask this of you. Ideally I wouldn't bother you, except that I'm not able to test on my own.

caveman99, 2024-05-13:
> I believe it is already acceptable to simply omit the `PIN_BUTTON` definition when defining a variant without a button. I'm not actually sure what the original reason for adding a button definition for this model was, and I'd hate to break someone else's device by removing it. My concern is that some T-Lora users may have DIY builds which do use a button connected to GPIO12.

The best practise to tell the firmware the board does not habe a user button is to not define that macro. This is one of the oldest parts of the code and comes from the idea, to have fixed pins assigned for optional peripherals. we have since removed the fixed GPS pins from those devices, and in theory the user button can be removed and replaced by the protobuf setting. We just need to put this into release nodes. Fixed definitions shold only exist if a button is physically present on the shipped devices.

todd-herbert, 2024-05-13:
Makes sense to me. I would still quite like to figure out what changes would resolve this particular situation though if possible,  because I notice there are a few other variants which also have `BUTTON_NEED_PULLUP`, and I'm suspicious that they might be affected by the same bug right now.

roha-github, 2024-05-13:
Yes, I  tested commenting out line 267 and 268, and it resolved the issue.

I repeated the test with pinMode INPUT_PULLUP, and it also resolved the issue.
I don't change the main-esp32.cpp and think gpio_pullup_en alone has no impact, 
because pinMode is a Arduino wrapper around gpio_set_direction and gpio_set_pull_mode.
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/gpio.html

roha-github, 2024-05-13:
> there are a few other variants which also have `BUTTON_NEED_PULLUP`
There may be models whose push button is connected to GND.

todd-herbert, 2024-05-13:
Thank you very much for confirming that!

> I don't change the main-esp32.cpp and think gpio_pullup_en alone has no impact,
because pinMode is a Arduino wrapper around gpio_set_direction and gpio_set_pull_mode.

I had a slight suspicion that maybe there was something odd happening when calling `gpio_pullup_en` after `gpio_hold_en`, but from the results of your testing, it seems that it's no problem!

> There may be models whose push button is connected to GND.

Specifically the ones without external pull-ups, too.

Just to absolutely confirm: replacing like 267 of sleep.cpp with `pinMode(BUTTON_PIN, INPUT_PULLUP);` resolves the issue for you; no extra changes required? I ask because I'll make that change for any devices with `BUTTON_NEED_PULLUP`, unless of course @caveman99 is happy to make the changes? I can also go through and purge any `PIN_BUTTON` definitions for devices without an onboard user button, if that helps tidy things up.

roha-github, 2024-05-13:
replacing like 267 of sleep.cpp with pinMode(BUTTON_PIN, INPUT_PULLUP); resolves the issue for me; no extra changes required
```
#ifdef BUTTON_PIN
    // Avoid leakage through button pin
    if (GPIO_IS_VALID_OUTPUT_GPIO(BUTTON_PIN)) {
        pinMode(BUTTON_PIN, INPUT_PULLUP); // bugfix #3875 pinMode(BUTTON_PIN, INPUT);
        gpio_hold_en((gpio_num_t)BUTTON_PIN);
    }
#endif
```

GUVWAF, 2024-06-21:
Seems fixed by #3884.

closed 2024-06-21 by GUVWAF

roha-github, 2024-06-21:
Get TLORA 2.1-1.6 down to 132uA at deep sleep. Works best for cheap sensor nodes.