meshtastic/firmware #3876: [Bug]: GPIO output gpio_num error

roha-github, 2024-05-13:
### Category

Hardware Compatibility

### Hardware

Heltec v2.1

### Firmware Version

2.3.9.f06c56a

### Description

Device: Heltec V2.1
Firmware: 2.3.9.f06c56a Alpha

Error: E (29334) gpio: gpio_set_level(226): GPIO output gpio_num error
Expected: no error message regarding GPIO's after shutdown

During shutdown there is an error message regarding invalid IO numbers. It is unclear whether this could have negative effects after the resume. This error message does not appear on other devices.

### Relevant log output

```Shell
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (822) esp_core_dump_flash: No core dump▒▒ѥѥ▒▒▒found!
E (822) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, A                                                                                                                                                             PB: 80000000 Hz
▒▒@INFO  | ??:??:?? 0

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (491520/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (103 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (396 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (106 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-BMhLjTmL.css.gz (16270 Bytes)
DEBUG | ??:??:?? 0  /static/index-Bt2MYVVB.js.gz (203021 Bytes)
DEBUG | ??:??:?? 0  /static/index-CGqDWSD_.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (548 Bytes)
DEBUG | ??:??:?? 0  /static/maplibre-gl-BXZZhNda.js.gz (210451 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (197 Bytes)
[   554][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=4 scl=15                                                                                                                                                              freq=100000
DEBUG | ??:??:?? 0 Using analog input 37 for battery level
INFO  | ??:??:?? 0 ADCmod: ADC characterization based on reference voltage store                                                                                                                                                             d in eFuse
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   581][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 0 I2C device found at address 0x3c
DEBUG | ??:??:?? 0 0xf subtype probed in 2 tries
INFO  | ??:??:?? 0 1 I2C devices found
[... 267 lines cut ...]
DEBUG | ??:??:?? 22 [Screen] showing shutdown screen
DEBUG | ??:??:?? 22 [Screen] showing shutdown screen
INFO  | ??:??:?? 22 [Screen] From Radio onread
DEBUG | ??:??:?? 22 [Screen] showing shutdown screen
DEBUG | ??:??:?? 23 [Screen] showing shutdown screen
DEBUG | ??:??:?? 23 [GPS] Probing for GPS at 9600
DEBUG | ??:??:?? 23 [GPS] Setting Baud to 9600
WARN  | ??:??:?? 26 [GPS] Failed to find UBlox & MTK GNSS Module using baudrate 9600
DEBUG | ??:??:?? 26 [Screen] showing shutdown screen
DEBUG | ??:??:?? 27 [Screen] showing shutdown screen
INFO  | ??:??:?? 27 Shutting down from admin command
INFO  | ??:??:?? 27 Shutting down
INFO  | ??:??:?? 27 Entering deep sleep forever
INFO  | ??:??:?? 27 Disable bluetooth until reboot
INFO  | ??:??:?? 27 BLE disconnect
E NimBLEAdvertising: Error enabling advertising; rc=30,
INFO  | ??:??:?? 27 Turning off screen
INFO  | ??:??:?? 27 Saving /prefs/db.proto
INFO  | ??:??:?? 27 Saving /prefs/config.proto
INFO  | ??:??:?? 28 Saving /prefs/module.proto
INFO  | ??:??:?? 29 Saving /prefs/channels.proto
INFO  | ??:??:?? 29 Setting GPS power=0
DEBUG | ??:??:?? 29 GPS powerdown using GPS_EN_ACTIVE
E (29334) gpio: gpio_set_level(226): GPIO output gpio_num error
```

roha-github, 2024-05-13:
Device: Heltec V3
Firmware: 2.3.9.f06c56a Alpha

Error:
E NimBLEAdvertising: Error enabling advertising; rc=30,
Expected: no error message regarding BLE during shutdown

Error:
E (29446) RTCIO: rtc_gpio_isolate(171): RTCIO number error
E (29447) RTCIO: rtc_gpio_isolate(171): RTCIO number error
E (29447) RTCIO: rtc_gpio_isolate(171): RTCIO number error
Expected: no error message regarding GPIO's after shutdown

Similar error messages with Heltec V3 plus BLE error.

```ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (353) esp_core_dump_flash: No core dump partition found!
E (353) esp_core_dump_flash: No core dump partition found!
▒▒@INFO  | ??:??:?? 0

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (491520/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (99 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (314 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (107 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-BMhLjTmL.css.gz (16270 Bytes)
DEBUG | ??:??:?? 0  /static/index-Bt2MYVVB.js.gz (203021 Bytes)
DEBUG | ??:??:?? 0  /static/index-CGqDWSD_.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (548 Bytes)
DEBUG | ??:??:?? 0  /static/maplibre-gl-BXZZhNda.js.gz (210451 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (197 Bytes)
[   717][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=41 scl=42 freq=100000
[   719][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=17 scl=18 freq=100000
DEBUG | ??:??:?? 0 Using analog input 1 for battery level
INFO  | ??:??:?? 0 ADCmod: ADC Characterization based on Two Point values and fitting curve coefficients stored in eFuse
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   751][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 2
[   781][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 0 I2C device found at address 0x3c
INFO  | ??:??:?? 0 ssd1306 display found
INFO  | ??:??:?? 0 ssd1306 display found
DEBUG | ??:??:?? 0 0x3 subtype probed in 2 tries
INFO  | ??:??:?? 0 1 I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 Meshtastic hwvendor=43, swver=2.3.9.f06c56a
DEBUG | ??:??:?? 0 Setting random seed 2487372047
DEBUG | ??:??:?? 0 Total heap: 293720
DEBUG | ??:??:?? 0 Free heap: 257220
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
[... 232 lines cut ...]
DEBUG | ??:??:?? 23 [Screen] showing shutdown screen
DEBUG | ??:??:?? 23 [Screen] phone downloaded packet (id=0x33eef3a8 fr=0x44 to=0x44, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=d3e8f203 priority=120)
DEBUG | ??:??:?? 23 [Screen] showing shutdown screen
DEBUG | ??:??:?? 23 [Screen] encoding toPhone packet to phone variant=2, 34 bytes
DEBUG | ??:??:?? 23 [Screen] showing shutdown screen
DEBUG | ??:??:?? 23 [Screen] showing shutdown screen
INFO  | ??:??:?? 23 [Screen] From Radio onread
DEBUG | ??:??:?? 23 [Screen] showing shutdown screen
DEBUG | ??:??:?? 28 [Screen] showing shutdown screen
INFO  | ??:??:?? 28 Shutting down from admin command
INFO  | ??:??:?? 28 Shutting down
INFO  | ??:??:?? 28 Entering deep sleep forever
INFO  | ??:??:?? 28 Disable bluetooth until reboot
INFO  | ??:??:?? 28 BLE disconnect
E NimBLEAdvertising: Error enabling advertising; rc=30,
DEBUG | ??:??:?? 28 SX126x entering sleep mode (FIXME, don't keep config)
INFO  | ??:??:?? 28 Turning off screen
INFO  | ??:??:?? 28 Saving /prefs/db.proto
INFO  | ??:??:?? 29 Saving /prefs/config.proto
INFO  | ??:??:?? 29 Saving /prefs/module.proto
INFO  | ??:??:?? 29 Saving /prefs/channels.proto
E (29446) RTCIO: rtc_gpio_isolate(171): RTCIO number error
E (29447) RTCIO: rtc_gpio_isolate(171): RTCIO number error
E (29447) RTCIO: rtc_gpio_isolate(171): RTCIO number error
```

garthvh, 2024-05-13:
This is no longer a supported device

roha-github, 2024-05-13:
@garthvh The error occurs in several models. Should I post separate bugs in future? 

* Heltec V2
* Heltec V3
* Heltec Wireless Stick

Yes, the Heltec V2 are older, but the power consumption with active ESP32 is only 70mA compared to the Heltect V3 with 108mA. Do you mean Heltec V3 is outdated?

garthvh, 2024-05-13:
It will get closed if logged against an unsupported device, so if it happens on a supported one that would be better.

closed 2024-10-07 by thebentern