meshtastic/firmware #3879: [Feature Request]: Anemometer support

mikechristiansen, 2024-05-13:
### Platform

NRF52, ESP32

### Description

Hi, Just wondering what the best approach might be to add Anemometer support for environmental data on Meshtastic. The use case of a remote weather station is exciting due to Meshtastics hardware support and data routing capability. 
I understand a co-processor could be used and the serial module to forward data but using the hardware meshtastic is running on would seem to be more power efficient. Data could then be routed to other systems using MQTT or other integrations, and having the data available to other nodes on the default channel through telemetry packets might be useful for other nodes on the network. 
Many anemometer sensors are on the market, most use an analog read pin for wind direction and a digital pin for wind speed. the DAVIS anemometers are widespread and of good quality. Code is available on Arduino to take measurements from this sensor https://github.com/wrybread/weewx-ArduinoWeatherStation/blob/master/ArduinoWeather.ino 
Is there any work on adding this sensor type? Or any desire or willingness from developers to add this to Meshtastic? 
Many thanks for the hard work on this fantastic project! 
Regards, Mike.

garthvh, 2024-05-13:
This is the first step https://github.com/meshtastic/protobufs/issues/491

Using serial to pass data from a weather station is probably going to be the easiest way go get started.

Analog sensors that require processing need to have a second MCU to do that processing, there are no I2C Anemometers that we have found.

garthvh, 2024-05-13:
The serial module has a protobufs mode, which is probably what we would want to use to pass data in as an environment telemetry packet.

garthvh, 2024-05-13:
We could also try and make a I2C adaptor / MCU for some of these analog sensors, or use serial from weather stations.  I suspect weather stations may be the first step as a lot of people have fancy weather stations already

AllanRomanato, 2024-05-14:
Wondering here..

What if we create something like an "stationary" module which would have 2 MCUs (one for LoRa and one for analog sensors) or a outside module.

We could have some #defines to handle if it is an stationary module or not and another code to the new MCU which could transfer the data to LoRa via Serial.

mikechristiansen, 2024-05-14:
@AllanRomanato I was thinking something like the ATTINY85 might do it, low power consumption if running from 3.3v or vbat.

caveman99, 2024-05-14:
the key is to code the intermediate mcu so that the meshtastic firmware does not need to poll it for a few minutes. Average sensors report every 900 seconds, the wind sensor can go lower, but should at lease buffer/average measurements for about 5 minutes. One data transmission from the new sensor type can be crafted so to carry all those values. What about momentary value every 30 seconds, then average and peak values. Both for direction and speed.

mikechristiansen, 2024-08-17:
Is anyone able to help me emulate the LARK to use on a co-processor to meshtastic in order to connect standard anamemoters? This is a bireif start I have made, meshtastic sees a connected LARK but I cant get the values to show up in meshtastic - it returns all zeros.
https://github.com/mikechristiansen/Msh_Weather_bridge

fifieldt, 2024-09-16:
https://github.com/meshtastic/firmware/pull/4032 implemented I2C Anemometer support.

Converting from bug to discussion so this can continue.