meshtastic/firmware #3881: [Feature Request]: Keyboard shortcuts documentation for t-deck.

bsouth00, 2024-05-14:
### Platform

ESP32

### Description

Running [2.3.9.f06c56a Alpha](https://github.com/meshtastic/firmware/releases/tag/v2.3.9.f06c56a) and I believe that some t-decks are being shipped from lilygo with different keyboard firmware.

For the examples below keys separated by a comma (',') indicate that the keys are pressed in sequence, releasing the previous key bafore pressing the next while keys separated by a plus ('+') indicate that the keys before is held and released after the second key is pressed.

The below keyboard shortcuts are working on my board:

- Shift+mic does NOT seem to work to enable the function layer, but the sequence "Sym,Shift+N" does
- I've been using TAB successfully since at least 2.3.7.30fbcab, maybe earlier.  The sequence is "Sym,Shift+Y"
- "Sym,Shift+N,m" - Mute (does not show the message "Notifications [Enabled/Disabled]" unless the external notification module is enabled in the module config.
- "Sym,Shift+N,g" - Toggles GPS
- "Sym,Shift+N,q" - "Quit" (is this the equivalent of an escape key?
- "Sym,Shift+J" - Also quit?
- "Sym,Shift+N,[space]" - "Position update sent" (but where was it sent? only to the primary channel?)
- "Sym,Shift+N,i" - lower screen brightness level
- "Sym,Shift+N,o" - raise screen brightness level
- After reading the rest of [This PR #3668](https://github.com/meshtastic/firmware/pull/3668) it seems that all of the above "Sym,Shift+N" combinations also work with "Alt+C" as the Fn modifier.

Thanks to all the contributors for the frequent improvements!

bsouth00, 2024-05-14:
I should read the fine manual before posting these things...
https://meshtastic.org/docs/hardware/devices/tdeck/

Still, leaving this open for a bit for the contributors awareness that some differences in the t-deck keyboards may be occurring.

bsouth00, 2024-05-14:
And I just found two more:
-"Sym,Shift+W" - Raise screen brightness
-"Sym,Shift+E" - Lower screen brightness

bsouth00, 2024-05-14:
Yeah, I wrote this up after loosely following some commits over the last few weeks where the given Fn modifier in the commit was not working on my t-deck.

I was testing keypresses and following along in the thread of that PR after flashing the latest alpha.

By the time that I saw the alt & 'c' combination as an option I'd already written the rest and then noticed after submitting that the documentation had been updated with alt & 'c' as the Fn modifier.

Perfectly satisfied if this is closed as "won't fix" or "not planned", just having the alternates visible somewhere in the issue tracker may still be of use later.  Probably not useful for most users.

itstayyabniazi, 2024-08-24:
How do I enable Mic ??
"Sym/Shift , Mic" is not working...