meshtastic/firmware #3883: [Feature Request]: use internal ESP32, NRF52 temperature sensor of RTC

roha-github, 2024-05-14:
### Platform

NRF52, ESP32, RP2040

### Description

Idea: use the internal temperature sensor of the Real Time Clock (RTC) as a temperature sensor. 

RTC chips have an internal temperature sensor for ongoing calibration of the Real Time Clock. The internal sensor could be used to ( approximately) determine the temperature. 

https://www.instructables.com/Temperature-Measuring-Without-Any-Sensors-in-NRF52/
https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/peripherals/temp_sensor.html
https://learnembeddedsystems.co.uk/using-the-rp2040-on-board-temperature-sensor

roha-github, 2024-05-14:
> You'll get a temperature that's (much) higher than the actual ambient temperature, and it's unknown how to compensate for it. I'm not sure how useful this really is.

It depends ... I have already used the RTC temperature sensor in projects. With a correction value you can adjust the measurement temperature to the little heating of the chip.

But with pure sensors, the MCU is almost always in sleep mode anyway and there is de facto no heating.  The values are then +/-1°C and a good indicator of heat.

Btw you can also use LEDs as light sensors for certain circuits.

roha-github, 2024-05-14:
@caveman99 This is about the possibility of enabling a temperature sensor WITHOUT additional hardware. I would leave it up to the user to decide exactly what this is and for what purposes it can be used.

Have you had any practical experience with the temperature sensors from RTC? What were your deviations from reality?

thebentern, 2024-05-14:
A dirt cheap supported BMP180 or SHT temp i2c sensor can be added for little cost and effort. Trying to use problematic CPU temperatures isn't something that seems like a fruitful effort in the face of relatively easy alternatives.

closed 2024-05-14 by thebentern

GUVWAF, 2024-05-14:
> What I particularly like about Meshtastic is that it's not just made for nerds. Without IT knowledge, users can buy one of the inexpensive devices and simply flash it via the web UI.

That's indeed nice. However, if a temperature sensor is included by default, those people expect it to be accurate. You can’t really say that you shouldn’t rely on it when it’s -20 degrees outside, but if it’s +20 degrees it _might_ be accurate, depending on how busy your mesh and thus processor and radio is, etc.

roha-github, 2024-05-17:
As caveman99 says, the real temperature and the RTC temperature are not directly identical. 

Instead of a simple correction value, you could also determine how the RTC temperature behaves depending on the proportion of deep / light sleep. The deviation depends on how much energy the MCU consumes in heat. 

For the device type sensor or as an indicator for the temperature, this should be negligible. Btw, the battery voltage and the percentage of charge are also only approximated.

blieb, 2025-03-08:
I think it would be a nice feature to monitor the board temperature, but this should be separate than the environment temperature.