meshtastic/firmware #3886: [Bug]: No PSRAM reported for RAK11200 (ESP32-WROVER)

erikdewildt, 2024-05-14:
### Category

Hardware Compatibility

### Hardware

Rak11200

### Firmware Version

2.3.6.7a3570a

### Description

Hi,

I have some RAK11200 based nodes. However the Store & Forward module can not be enabled as the device reports no PSRAM. See the attached logging output.

[According to RAK](https://store.rakwireless.com/products/wiscore-esp32-module-rak11200) these boards should have "8 MB SPI Pseudo static RAM (PSRAM)".

Hope this can be resolved in a future firmware release so we can enable S&F on our nodes.

Thanks,
Regards,

Erik de Wildt

### Relevant log output

```Shell
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (818) esp_core_dump_flash: No core dump��ѥѥ���found!
E (818) esp_core_dump_flash: No core dump partition found!
[ 12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
��@INFO | ??:??:?? 0

//\ E S H T /\ S T / C

INFO | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (20480/1048576 Bytes):
DEBUG | ??:??:?? 0 /prefs/channels.proto (73 Bytes)
DEBUG | ??:??:?? 0 /prefs/config.proto (124 Bytes)
DEBUG | ??:??:?? 0 /prefs/db.proto (747 Bytes)
DEBUG | ??:??:?? 0 /prefs/module.proto (118 Bytes)
[ 127][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=4 scl=5 freq=100000
DEBUG | ??:??:?? 0 Using analog input 36 for battery level
INFO | ??:??:?? 0 ADCmod: ADC characterization based on reference voltage stored in eFuse
INFO | ??:??:?? 0 Scanning for i2c devices...
[ 154][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
INFO | ??:??:?? 0 No I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO | ??:??:?? 0 Meshtastic hwvendor=13, swver=2.3.6.7a3570a
DEBUG | ??:??:?? 0 Setting random seed 2237718514
DEBUG | ??:??:?? 0 Total heap: 197896
DEBUG | ??:??:?? 0 Free heap: 163508
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
DEBUG | ??:??:?? 0 NVS: UsedEntries 193, FreeEntries 437, AllEntries 630, NameSpaces 7
DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 0 Number of Device Reboots: 27

INFO | ??:??:?? 0 *** Initializing Store & Forward Module in Router mode
INFO | ??:??:?? 0 *** Device doesn't have PSRAM.
INFO | ??:??:?? 0 *** Store & Forward Module - disabling server.
```

caveman99, 2024-05-14:
can you check the firmware artefact built from https://github.com/meshtastic/firmware/pull/3887 once the CI run finishes. That should fix PSRAM reporting on that board.

erikdewildt, 2024-05-14:
Hi Thomas,

Your fixed worked perfectly:

`
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:1184

load:0x40078000,len:13192

load:0x40080400,len:3028

entry 0x400805e4

E (827) esp_core_dump_flash: No core dum���ѥѥ���found!

E (827) esp_core_dump_flash: No core dump partition found!

[ 12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

[ 464][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

��@INFO | ??:??:?? 0

//\ E S H T /\ S T / C

INFO | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset

DEBUG | ??:??:?? 0 Filesystem files (20480/1048576 Bytes):

DEBUG | ??:??:?? 0 /prefs/channels.proto (73 Bytes)

DEBUG | ??:??:?? 0 /prefs/config.proto (124 Bytes)

DEBUG | ??:??:?? 0 /prefs/db.proto (747 Bytes)

DEBUG | ??:??:?? 0 /prefs/module.proto (118 Bytes)

[ 570][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=4 scl=5 freq=100000

DEBUG | ??:??:?? 0 Using analog input 36 for battery level

INFO | ??:??:?? 0 ADCmod: ADC characterization based on reference voltage stored in eFuse

INFO | ??:??:?? 0 Scanning for i2c devices...

[ 596][W][Wire.cpp:301] begin(): Bus already started in Master Mode.

DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1

INFO | ??:??:?? 0 No I2C devices found

DEBUG | ??:??:?? 0 acc_info = 0

INFO | ??:??:?? 0 Meshtastic hwvendor=13, swver=2.3.10.ed86660

DEBUG | ??:??:?? 0 Setting random seed 1556584251

DEBUG | ??:??:?? 0 Total heap: 192888

DEBUG | ??:??:?? 0 Free heap: 158728

DEBUG | ??:??:?? 0 Total PSRAM: 4192107

DEBUG | ??:??:?? 0 Free PSRAM: 4191847

DEBUG | ??:??:?? 0 NVS: UsedEntries 193, FreeEntries 437, AllEntries 630, NameSpaces 7

DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage

DEBUG | ??:??:?? 0 Number of Device Reboots: 33

...

INFO | ??:??:?? 1 *** Initializing Store & Forward Module in Router mode

DEBUG | ??:??:?? 1 *** Before PSRAM initialization: heap 149148/191904 PSRAM 4177031/4192091

DEBUG | ??:??:?? 1 *** After PSRAM initialization: heap 149148/191904 PSRAM 1390383/4192059

DEBUG | ??:??:?? 1 *** numberOfPackets for packetHistory - 11033

`

Thanks for the quick fix!
Regards,

Erik

PR #3887 (2024-05-14): add psram flag on RAK11200 board definition
fixes #3886
closed 2024-05-14 by erikdewildt