meshtastic/firmware #3897: since 15.05.2025 midnight (CEST) 500kbit/s MQTT traffic

hb9eue, 2024-05-15:
### Category

WiFi

### Hardware

Not Applicable

### Firmware Version

0

### Description

I use a mosquitto server in Bridge Mode to mqtt.meshtastic.org:1883 to connect my Meshtastic devices running on a Arm64 device.

Since last night, it crashed twice with out of memory.

My snmp monitoring shows, that last night I contantly had about 500kbit/s traffic to that instance and memory consumption steadily increased.
After short investigation I found that several thausend of topics are being created within minutes only in the meshtastic subtopic.

I wonder, is there an ongoing attack to MQTT? Or was there any kind of firmware change pushed last night causing the issue?

-Benoit-

### Relevant log output

```Shell
none
```

caveman99, 2024-05-15:
The public MQTT is exactly that - a public service. We will be implementing protective measures in the future but these changes are still in testing. This is not related to the firmware, so closing for now.

closed 2024-05-15 by caveman99

thebentern, 2024-05-15:
I believe it got into a bad state. It was power cycled this morning and seems to be behaving nicer :-)