meshtastic/firmware #3909: [Bug]: GPS on Portduino works on first application start, subsequent GPS checks fail indefinately.

Talie5in, 2024-05-16:
### Category

Other

### Hardware

Other

### Firmware Version

2.3.9

### Description

Portduino (rPi 3 with Waveshare SX1262 LoRaWAN/GNSS HAT)
https://www.waveshare.com/sx1262-lorawan-hat.htm?sku=24654

On meshtasticd start, initial aquisition of GPS via /dev/ttyS0 is successful
- After first aquisition "WANT GPS=0 and GPS power=0" gets set.
- At next poll, "WANT GPS=1 and GPS power=1" you get a 900 second timeout on getting a lock
- - During this period you can check data on /dev/ttyS0 and still see GPS position data coming in (have also tested this while meshtasticd is running and not running, always have data coming in)
- Every subsequent poll, "WANT GPS=1 and GPS power=1" you get a 900 second timeout on getting a lock

Another report of this has appeared in Discord by 'madeofstown' - so not a single occurrence.

### Relevant log output

```Shell
DEBUG | ??:??:?? 0 [GPS] Probing for GPS at 9600
INFO  | ??:??:?? 0 [GPS] L76K GNSS init succeeded, using L76K GNSS Module
DEBUG | ??:??:?? 2 [GPS] NMEA GPS time 2024-05-15 13:35:14
DEBUG | ??:??:?? 2 [GPS] Upgrading time to quality GPS
DEBUG | 13:35:14 2 [GPS] Read RTC time as 1715780114
DEBUG | 13:35:14 2 [GPS] hasValidLocation RISING EDGE
DEBUG | 13:35:14 2 [GPS] WANT GPS=0
DEBUG | 13:35:14 2 [GPS] GPS Lock took 2, average 0
INFO  | 13:35:14 2 [GPS] Setting GPS power=0
DEBUG | 13:35:14 2 [GPS] publishing pos@6644ba12:2, hasVal=1, Sats=29, GPSlock=1
DEBUG | 13:35:14 2 [GPS] New GPS pos@6644ba12:3 lat=-35.092627, lon=138.872620, alt=349, pdop=0.84, track=230.36, speed=0.00, sats=29
DEBUG | 13:35:14 2 [GPS] onGPSChanged() pos@6644ba12, time=1715780114, lat=-350926275, lon=1388726198, alt=349
INFO  | 13:35:14 2 [GPS] updatePosition LOCAL pos@6644ba12, time=1715780114, latI=-350926275, lonI=1388726198, alt=349
DEBUG | 13:35:14 2 [GPS] Setting local position: latitude=-350926275, longitude=1388726198, time=1715780114, timestamp=1715780114
<snip>
DEBUG | 13:37:14 122 [GPS] WANT GPS=1
INFO  | 13:37:14 122 [GPS] Setting GPS power=1
DEBUG | 13:52:14 1022 [GPS] hasValidLocation FALLING EDGE (last read: 0)
DEBUG | 13:52:14 1022 [GPS] WANT GPS=0
DEBUG | 13:52:14 1022 [GPS] GPS Lock took 900, average 900
DEBUG | 13:52:14 1022 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=1
DEBUG | 13:52:14 1022 [GPS] No GPS lock
DEBUG | 13:52:14 1022 [GPS] onGPSChanged() pos@0, time=1715781134, lat=0, lon=0, alt=0
INFO  | 13:52:14 1022 [GPS] updatePosition LOCAL pos@0, time=1715781134, latI=0, lonI=0, alt=0
DEBUG | 13:52:14 1022 [GPS] Setting local position: latitude=0, longitude=0, time=1715781134, timestamp=0
DEBUG | 13:52:14 1022 [GPS] Node status update: 9 online, 15 total
DEBUG | 13:54:14 1142 [GPS] WANT GPS=1
INFO  | 13:54:14 1142 [GPS] Setting GPS power=1
DEBUG | 14:09:14 2042 [GPS] WANT GPS=0
DEBUG | 14:09:14 2042 [GPS] GPS Lock took 900, average 899
DEBUG | 14:09:14 2042 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=1
DEBUG | 14:09:14 2042 [GPS] onGPSChanged() pos@0, time=1715782154, lat=0, lon=0, alt=0
INFO  | 14:09:14 2042 [GPS] updatePosition LOCAL pos@0, time=1715782154, latI=0, lonI=0, alt=0
DEBUG | 14:09:14 2042 [GPS] Setting local position: latitude=0, longitude=0, time=1715782154, timestamp=0
DEBUG | 14:09:14 2042 [GPS] Node status update: 10 online, 15 total
DEBUG | 14:11:14 2162 [GPS] WANT GPS=1
INFO  | 14:11:14 2162 [GPS] Setting GPS power=1
DEBUG | 14:26:14 3063 [GPS] WANT GPS=0
DEBUG | 14:26:14 3063 [GPS] GPS Lock took 900, average 898
DEBUG | 14:26:14 3063 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=1
DEBUG | 14:26:14 3063 [GPS] onGPSChanged() pos@0, time=1715783174, lat=0, lon=0, alt=0
INFO  | 14:26:14 3063 [GPS] updatePosition LOCAL pos@0, time=1715783174, latI=0, lonI=0, alt=0
DEBUG | 14:26:14 3063 [GPS] Setting local position: latitude=0, longitude=0, time=1715783174, timestamp=0
DEBUG | 14:26:14 3063 [GPS] Node status update: 11 online, 16 total

<SEPARATE TEST>
Commentary: - Setting Get GPS every 30 seconds to attempt to stop powerdown fails as a forced power=0 set right after initialization>

INFO  | ??:??:?? 0 [GPS] L76K GNSS init succeeded, using L76K GNSS Module
INFO  | 15:10:00 1 [GPS] Setting GPS power=0
INFO  | 15:10:00 1 [GPS] updatePosition LOCAL pos@6644d048, time=1715785800, latI=-350926183, lonI=1388726060, alt=346
INFO  | 15:10:00 1 [GPS] Sending pos@6644d048:6 to mesh (wantReplies=1)
INFO  | 15:10:00 1 [GPS] Providing time to mesh 1715785800
INFO  | 15:10:00 1 [GPS] Position reply: time=1715785800, latI=-350912512, lonI=1388740608
INFO  | 15:10:00 1 [GPS] serialized json message: {"channel":0,"from":3958966453,"hops_away":0,"id":1804289385,"payload":{"PDOP":84,"altitude":346,"ground_track":23036000,"latitude_i"
INFO  | 15:10:00 1 [GPS] JSON publish message to msh/ANZ/SA/2/json/LongFast/!ebf908b5, 305 bytes: {"channel":0,"from":3958966453,"hops_away":0,"id":1804289385,"payload":{"PDOP":84,"al
INFO  | 15:10:30 31 [GPS] Setting GPS power=1
INFO  | 15:25:30 931 [GPS] updatePosition LOCAL pos@0, time=1715786730, latI=0, lonI=0, alt=0
INFO  | 15:26:05 966 [GPS] Setting GPS power=1
INFO  | 15:41:05 1867 [GPS] updatePosition LOCAL pos@0, time=1715787665, latI=0, lonI=0, alt=0
INFO  | 15:41:40 1902 [GPS] Setting GPS power=1
INFO  | 15:56:40 2802 [GPS] updatePosition LOCAL pos@0, time=1715788600, latI=0, lonI=0, alt=0
INFO  | 15:57:10 2832 [GPS] Setting GPS power=1
INFO  | 16:12:10 3732 [GPS] updatePosition LOCAL pos@0, time=1715789530, latI=0, lonI=0, alt=0
INFO  | 16:12:40 3762 [GPS] Setting GPS power=1
INFO  | 16:27:40 4662 [GPS] updatePosition LOCAL pos@0, time=1715790460, latI=0, lonI=0, alt=0
INFO  | 16:28:10 4692 [GPS] Setting GPS power=1
INFO  | 16:43:10 5592 [GPS] updatePosition LOCAL pos@0, time=1715791390, latI=0, lonI=0, alt=0
INFO  | 16:43:45 5627 [GPS] Setting GPS power=1
INFO  | 16:58:45 6527 [GPS] updatePosition LOCAL pos@0, time=1715792325, latI=0, lonI=0, alt=0
INFO  | 16:59:20 6562 [GPS] Setting GPS power=1
```

jp-bennett, 2024-05-16:
We've gotten several reports of this. The strange thing is that not every install has this problem.

Talie5in, 2024-05-16:
@jp-bennett Yeah, I saw, and also experiencing it, so thought id do the right thing and open an issue so it can be tracked :)

GPSFan, 2024-05-16:
How is the Pi wired up to the L76K GPS module?

GPSFan, 2024-05-16:
Could you post some of the data you see coming into /dev/ttyS0.

Talie5in, 2024-05-16:
[image]
R13 Appears to be a unpopulated on the board, just like in the product image.

```
cat /dev/ttyS0

$GNRMC,135958.000,A,3505.55665,S,13852.35605,E,0.00,75.47,160524,,,A,V*2F

$GNGGA,135959.000,3505.55666,S,13852.35605,E,1,30,0.5,350.3,M,-0.9,M,,*47

$GNRMC,135959.000,A,3505.55666,S,13852.35605,E,0.00,75.47,160524,,,A,V*2D

$GNGGA,140000.000,3505.55665,S,13852.35604,E,1,30,0.5,350.3,M,-0.9,M,,*42

$GNRMC,140000.000,A,3505.55665,S,13852.35604,E,0.00,75.47,160524,,,A,V*28

$GNGGA,140001.000,3505.55666,S,13852.35604,E,1,30,0.5,350.3,M,-0.9,M,,*40

$GNRMC,140001.000,A,3505.55666,S,13852.35604,E,0.00,75.47,160524,,,A,V*2A

$GNGGA,140002.000,3505.55666,S,13852.35603,E,1,30,0.5,350.3,M,-0.9,M,,*44

$GNRMC,140002.000,A,3505.55666,S,13852.35603,E,0.00,75.47,160524,,,A,V*2E

$GNGGA,140003.000,3505.55666,S,13852.35602,E,1,30,0.5,350.4,M,-0.9,M,,*43

$GNRMC,140003.000,A,3505.55666,S,13852.35602,E,0.00,75.47,160524,,,A,V*2E

$GNGGA,140004.000,3505.55666,S,13852.35602,E,1,30,0.5,350.4,M,-0.9,M,,*44

$GNRMC,140004.000,A,3505.55666,S,13852.35602,E,0.00,75.47,160524,,,A,V*29

$GNGGA,140005.000,3505.55666,S,13852.35601,E,1,30,0.5,350.4,M,-0.9,M,,*46

$GNRMC,140005.000,A,3505.55666,S,13852.35601,E,0.00,75.47,160524,,,A,V*2B

$GNGGA,140006.000,3505.55666,S,13852.35601,E,1,30,0.5,350.4,M,-0.9,M,,*45

$GNRMC,140006.000,A,3505.55666,S,13852.35601,E,0.00,75.47,160524,,,A,V*28

$GNGGA,140007.000,3505.55667,S,13852.35600,E,1,30,0.5,350.4,M,-0.9,M,,*44

$GNRMC,140007.000,A,3505.55667,S,13852.35600,E,0.00,75.47,160524,,,A,V*29

$GNGGA,140008.000,3505.55667,S,13852.35600,E,1,30,0.5,350.4,M,-0.9,M,,*4B
```

Talie5in, 2024-05-16:
@GPSFan UGH, Standby Switch - yup... that is... ON 
But you would think i would get 0 GPS Locks while in that state?

Talie5in, 2024-05-16:
I can restart the app and get a valid GPS time etc each start, which would tell me it's getting a lock and not in Standby?
INFO  | 14:09:50 2 [GPS] Position reply: time=1715868590, latI=-350912512, lonI=1388740608
[image]
[image]

GPSFan, 2024-05-16:
Thanks for the update, lots to discuss here.
With the Standby switch in the ON position and R13 UNPOPULATED, the code in GPS.cpp that tries to put the module into standby or power off (for some boards) will not do anything.
Looks like the switch markings are reversed.
It looks like WaveShare didn't really read the datasheet for the L76K, the schematic symbol corresponds to the L76B, and the function of pin 18 is very different:
L76K switches between GPS/GLONASS and GPS/BeiDou
L76B Forces a wakeup from standby.
If that NMEA stream is getting into Meshtastic, it should be giving you a fix.

Talie5in, 2024-05-16:
@GPSFan 

Sounds.. fun

In the interim i'm attempting this to see if it keeps... updating :)

[image]

GPSFan, 2024-05-16:
If you can recompile, try turning on GPS debugging and Extra Verbose (you may have to edit the variant.h)

Talie5in, 2024-05-16:
If you point me where too for GPS debugging - im very happy to, didnt spot anything for it in gps.cpp ?

GPSFan, 2024-05-16:
Add these ti the variant.h file.

#define GPS_DEBUG
#define GPS_EXTRAVERBOSE

Talie5in, 2024-05-16:
Done, i'll let it run for a bit and i can supply the tee'd logfile

GPSFan, 2024-05-16:
ok, I think I may have an answer, when receiving more than just GPS, the receiver sends GNGGA/GNRMC instead (GPS only) of GPGGA/GPRMC. I'm pretty sure TinyGPS++ doesn't recognize the GNGGA/GNRMC sentences. I thought Meshtastic had made a mod to fix that, and I'm still looking through the code for it.

caveman99, 2024-05-16:
TinyGPS+++ takes care of the *GGA/*RMC handling. There was indeed a recent fix for that, but it's not in meshtastic but in our tinygps fork.

GPSFan, 2024-05-16:
Yeah, brain fart on my part ;>((

caveman99, 2024-05-16:
> Yeah, brain fart on my part ;>((

No sweat, just pointing out the direction to save you some trouble looking :-)

GPSFan, 2024-05-16:
The line of D30CB0D0D30CB0D is puzzling, I'm recompiling my local code with debug & extraverbose to see my corresponding line.

Talie5in, 2024-05-16:
You get that from "Probing" to what i pasted above L76K GNSS init succeeded.

```
DEBUG | ??:??:?? 0 [GPS] Probing for GPS at 9600 
DEBUG | ??:??:?? 0 [GPS] D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D [... cut ...] D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0
DEBUG | ??:??:?? 4 [GPS] 
DEBUG | ??:??:?? 4 [GPS] D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D3L76K GNSS init succeeded, using L76K GNSS Module0CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB0D0
DEBUG | ??:??:?? 5 [GPS] 
```

Talie5in, 2024-05-16:
One Debug Logfile with GPS_DEBUG and GPS_EXTRAVERBOSE enabled.
[meshtasticd_2024-05-17.log](https://github.com/meshtastic/firmware/files/15337622/meshtasticd_2024-05-17.log)

GPSFan, 2024-05-16:
Reading through the log, I can't see where the Pi got an initial position, the positions reported don't seem to be coming from the GPS.. and all of them seem to be the same.
The D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB stuff should be the hex of the NMEA sentences. If I were to guess I'd say there was a serial interface problem with Meshtastic.
I don't have a Pi (recent) an L76K or a waveshare hat, so debugging this is not going to be easy. 
I'll see what I can do about setting something up duplicate your setup.

GPSFan, 2024-05-16:
this is what I'm seeing during normal operation:
DEBUG | 18:43:22 2062 [Power] Battery: usbPower=0, isCharging=0, batMv=0, batPct=0
DEBUG | 18:43:23 2062 [GPS] WANT GPS=1
INFO  | 18:43:23 2062 [GPS] Setting GPS power=1
DEBUG | 18:43:23 2062 [GPS] AGE: LOC=10034 FIX=0 DATE=10037 TIME=10034
DEBUG | 18:43:23 2062 [GPS] AGE: LOC=10236 FIX=0 DATE=10239 TIME=10236
DEBUG | 18:43:23 2063 [GPS] AGE: LOC=10437 FIX=0 DATE=10440 TIME=10437
DEBUG | 18:43:23 2063 [GPS] AGE: LOC=10638 FIX=0 DATE=10641 TIME=10638
DEBUG | 18:43:24 2063 [GPS] AGE: LOC=10838 FIX=0 DATE=10841 TIME=10838
DEBUG | 18:43:24 2063 [GPS] $GNRMC,184325.00,A,3853.88749,N,10734.44632,W,0.781,,160524,,,A,V*04
$GNGGA,184325.00,3853.88749,N,10734.44632,W,1,12,0.72,1896.5,M,-20.6,M,,*44
AGE: LOC=4 FIX=0 DATE=7 TIME=4
DEBUG | 18:43:24 2063 [GPS] WANT GPS=0
DEBUG | 18:43:24 2063 [GPS] GPS Lock took 1, average 0
DEBUG | 18:43:24 2063 [GPS] publishing pos@664653cd:2, hasVal=1, Sats=12, GPSlock=1
DEBUG | 18:43:24 2063 [GPS] GPSStatus.match() new pos@664653cd to old pos@664653c2
DEBUG | 18:43:24 2063 [GPS] New GPS pos@664653cd:3 lat=38.898125, lon=-107.574105, alt=1896, pdop=1.01, track=102.14, speed=0.01, sats=12
DEBUG | 18:43:24 2063 [GPS] onGPSChanged() pos@664653cd, time=1715885004, lat=388981248, lon=-1075741053, alt=1896
INFO  | 18:43:24 2063 [GPS] updatePosition LOCAL pos@664653cd, time=1715885004, latI=388981248, lonI=-1075741053, alt=1896

madeofstown, 2024-05-16:
Just wanted to say I've been dealing with this same issue and I thought it was a hardware problem. Would be great if it wasn't. If you need any logs I dumped an hours worth of just the GPS stuff in discord in the Linux-native channel a few days ago.

GPSFan, 2024-05-16:
Thanks, I'm trying to get some sort of setup that I can run meshtasticd on. Might be able to borrow a Pi from a friend. I can then patch in a non L76K GPS to see what's going on.

madeofstown, 2024-05-16:
> Thanks, I'm trying to get some sort of setup that I can run meshtasticd on. Might be able to borrow a Pi from a friend. I can then patch in a non L76K GPS to see what's going on.

I've also got a neoM8n GNSS Module on order for testing. Will let you know (if I get mine first) how it does.

GPSFan, 2024-05-16:
I haven't done much with linux native as my plate is pretty full of DayJob (tm) and RealLife (tm) stuff, but it's on the list.

GPSFan, 2024-05-17:
@jp-bennett Linux native is a good candidate for tracking down this type of problem, I've got meshtasticd installed on my Rock64 debian Bookworm, but have yet to figure out how to start the daemon.
The docs for linux native could use a bit of help.

GPSFan, 2024-05-17:
Meshtasticd runs and doesn't crash. config is an issue more than likely.

jp-bennett, 2024-05-17:
> The docs for linux native could use a bit of help.

Make notes about what's missing, and we'll get it added. In this case, probably need a quick word about using systemctl to manage services.

Talie5in, 2024-05-17:
> > The docs for linux native could use a bit of help.
> 
> Make notes about what's missing, and we'll get it added. In this case, probably need a quick word about using systemctl to manage services.

https://github.com/meshtastic/meshtastic/issues/1259
Issue opened :)

madeofstown, 2024-05-18:
OK, testing with a different GPS module ( Ublox Neo-M8N from [here](https://www.aliexpress.us/item/3256805918008867.html?spm=a2g0o.order_list.order_list_main.46.f6191802xc5p3e&gatewayAdapt=glo2usa)). I seem to be getting a similar if not the same issue. It seems that the GPS logging is a bit more verbose than for the waveshare module so maybe this log will help.
```
DEBUG | ??:??:?? 13 [GPS] Probing for GPS at 57600
INFO  | ??:??:?? 15 [GPS] Found a UBlox Module using baudrate 57600
INFO  | ??:??:?? 17 [GPS] GNSS configured for GPS+SBAS+GLONASS+Galileo. Pause for 0.75s before sending next command.
WARN  | ??:??:?? 19 [GPS] Unable to disable text info messages.
WARN  | ??:??:?? 19 [GPS] Unable to enable interference resistance.
WARN  | ??:??:?? 20 [GPS] Unable to configure NAVX5 settings.
WARN  | ??:??:?? 20 [GPS] Got NAK for class 06 message 11
WARN  | ??:??:?? 20 [GPS] Unable to enable powersaving mode for GPS.
INFO  | ??:??:?? 21 [GPS] GNSS module configuration saved!
DEBUG | ??:??:?? 21 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | ??:??:?? 21 [GPS] No GPS lock
DEBUG | ??:??:?? 21 [GPS] onGPSChanged() pos@0, time=0, lat=0, lon=0, alt=0
INFO  | ??:??:?? 21 [GPS] updatePosition LOCAL pos@0, time=0, latI=0, lonI=0, alt=0
DEBUG | ??:??:?? 21 [GPS] Setting local position: latitude=0, longitude=0, time=0, timestamp=0
DEBUG | ??:??:?? 21 [GPS] Node status update: 0 online, 25 total
DEBUG | ??:??:?? 21 [GPS] showing standard frames
DEBUG | ??:??:?? 21 [GPS] Showing 0 module frames
DEBUG | ??:??:?? 21 [GPS] Total frame count: 203
DEBUG | ??:??:?? 21 [GPS] Added modules.  numframes: 0
DEBUG | ??:??:?? 21 [GPS] Finished building frames. numframes: 7
DEBUG | ??:??:?? 21 [GPS] NMEA GPS time 2024-05-18 15:35:08
DEBUG | ??:??:?? 21 [GPS] Upgrading time to quality GPS
DEBUG | 15:35:08 21 [GPS] Read RTC time as 1716046508
DEBUG | 15:35:08 21 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 15:35:08 21 [GPS] onGPSChanged() pos@0, time=1716046508, lat=0, lon=0, alt=0
INFO  | 15:35:08 21 [GPS] updatePosition LOCAL pos@0, time=1716046508, latI=0, lonI=0, alt=0
DEBUG | 15:35:08 21 [GPS] Setting local position: latitude=0, longitude=0, time=1716046508, timestamp=0
DEBUG | 15:35:08 21 [GPS] Node status update: 1 online, 25 total
DEBUG | 15:35:08 22 [GPS] hasValidLocation RISING EDGE
DEBUG | 15:35:08 22 [GPS] WANT GPS=0
DEBUG | 15:35:08 22 [GPS] GPS Lock took 21, average 0
INFO  | 15:35:08 22 [GPS] Setting GPS power=0
DEBUG | 15:35:08 22 [GPS] Sleep Time: 120000
DEBUG | 15:35:08 22 [GPS] publishing pos@6648caac:2, hasVal=1, Sats=12, GPSlock=1
DEBUG | 15:35:08 22 [GPS] New GPS pos@6648caac:3 lat=XXXXX, lon=XXXX, alt=30, pdop=1.04, track=0.00, speed=0.00, sats=12
DEBUG | 15:35:08 22 [GPS] onGPSChanged() pos@6648caac, time=1716046508, lat=XXXXX, lon=XXXXX, alt=30
INFO  | 15:35:08 22 [GPS] updatePosition LOCAL pos@6648caac, time=1716046508, latI=XXXXX, lonI=XXXXX, alt=30
DEBUG | 15:35:08 22 [GPS] Setting local position: latitude=XXXXX, longitude=XXXXX, time=1716046508, timestamp=1716046508
DEBUG | 15:35:08 22 [GPS] Node status update: 1 online, 25 total
INFO  | 15:35:08 22 [GPS] Sending pos@6648caac:6 to mesh (wantReplies=1)
DEBUG | 15:35:08 22 [GPS] Sending location with precision 16
INFO  | 15:35:08 22 [GPS] Providing time to mesh 1716046508
INFO  | 15:35:08 22 [GPS] Position reply: time=1716046508, latI=XXXXX, lonI=XXXXX
DEBUG | 15:35:08 22 [GPS] Initial packet id 1804289383, numPacketId 4294967295
DEBUG | 15:35:08 22 [GPS] Update DB node 0x104cf5c, rx_time=1716046508
DEBUG | 15:35:08 22 [GPS] handleReceived(LOCAL) (id=0x6b8b4569 fr=0x5c to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1716046508 priority=10)
DEBUG | 15:35:08 22 [GPS] No modules interested in portnum=3, src=LOCAL
DEBUG | 15:35:08 22 [GPS] localSend to channel 0
DEBUG | 15:35:08 22 [GPS] Add packet record (id=0x6b8b4569 fr=0x5c to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1716046508 priority=10)
DEBUG | 15:35:08 22 [GPS] Expanding short PSK #1
DEBUG | 15:35:08 22 [GPS] Using AES128 key!
DEBUG | 15:35:08 22 [GPS] Installing AES128 key!
DEBUG | 15:35:08 22 [GPS] enqueuing for send (id=0x6b8b4569 fr=0x5c to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1716046508 hopStart=3 priority=10)
DEBUG | 15:35:08 22 [GPS] Set random delay before transmitting.
DEBUG | 15:35:08 22 [GPS] Using channel 0 (hash 0x8)
DEBUG | 15:35:08 22 [GPS] Expanding short PSK #1
DEBUG | 15:35:08 22 [GPS] Using AES128 key!
DEBUG | 15:35:08 22 [GPS] Installing AES128 key!
DEBUG | 15:35:08 22 [GPS] decoded message (id=0x6b8b4569 fr=0x5c to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1716046508 hopStart=3 priority=10)
DEBUG | 15:35:08 22 [GPS] Sent smart pos@6648caac:6 to mesh (distanceTraveled=12783963.000000m, minDistanceThreshold=100m, timeElapsed=22024ms, minTimeInterval=30000ms)
DEBUG | 15:37:08 142 [GPS] WANT GPS=1
INFO  | 15:37:08 142 [GPS] Setting GPS power=1
DEBUG | 15:39:08 262 [GPS] hasValidLocation FALLING EDGE (last read: 0)
DEBUG | 15:39:08 262 [GPS] WANT GPS=0
DEBUG | 15:39:08 262 [GPS] GPS Lock took 120, average 120
DEBUG | 15:39:08 262 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=1
DEBUG | 15:39:08 262 [GPS] No GPS lock
DEBUG | 15:39:08 262 [GPS] onGPSChanged() pos@0, time=1716046748, lat=0, lon=0, alt=0
INFO  | 15:39:08 262 [GPS] updatePosition LOCAL pos@0, time=1716046748, latI=0, lonI=0, alt=0
DEBUG | 15:39:08 262 [GPS] Setting local position: latitude=0, longitude=0, time=1716046748, timestamp=0
DEBUG | 15:39:08 262 [GPS] Node status update: 1 online, 25 total
```

GPSFan, 2024-05-18:
Very strange, I have a Pi3+ and waveshare hat coming on Wed. So I'll be able to see exactly what is going on, I will however need some help to configure the Pi & hat to duplicate your setup..
How did you compile the firmware? on the Pi itself, or using Platformio on a desktop/laptop?
I've built the firmware on my Linux laptop for esp32 and NRF52 successfully, and am no stranger to Linux on ARM and x86 platforms.

GPSFan, 2024-05-18:
Noticed that Meshtastic found your u-blox at 57600baud, that is unusual unless you changed it, and saved the config to flash.

madeofstown, 2024-05-18:
I just ran the latest 2.3.9 alpha release .deb
Yeah, the baud rate is inconsistent over UART for some reason. Most of time it will use 9600 but Ive also seen it use 4800, 57600, and 115200. Now I'm testing with the USB interface on the module and have configured meshtasticd to use `/dev/ttyACM0` and it seems to be holding a lock...

madeofstown, 2024-05-18:
But ive built the releases before using `bin/build-native.sh`

GPSFan, 2024-05-18:
Ok, I'll start with that, I got it so it runs on my Rock64, but without a waveshare hat, no LORA and no GPS. I also have an issue with the SPI config on my Rock64, so I went for broke and got a Pi & waveshare hat on order.

madeofstown, 2024-05-18:
Here's the bootup log for the GPS while using the USB interface.
```
DEBUG | ??:??:?? 0 [GPS] Probing for GPS at 9600
INFO  | ??:??:?? 1 [GPS] Found a UBlox Module using baudrate 9600
DEBUG | ??:??:?? 2 [GPS] Module Info :
DEBUG | ??:??:?? 2 [GPS] Soft version: 2.01 (75350)
DEBUG | ??:??:?? 2 [GPS] Hard version: 00080000
DEBUG | ??:??:?? 2 [GPS] Extensions:5
DEBUG | ??:??:?? 2 [GPS]   2.01 (75331)
DEBUG | ??:??:?? 2 [GPS]   PROTVER 15.00
DEBUG | ??:??:?? 2 [GPS]   FIS 0xEF4015 (79189)
DEBUG | ??:??:?? 2 [GPS]
DEBUG | ??:??:?? 2 [GPS]
DEBUG | ??:??:?? 2 [GPS] Protocol Version:15.00
DEBUG | ??:??:?? 2 [GPS] ProtVer=15
WARN  | ??:??:?? 2 [GPS] Got NAK for class 06 message 3E
INFO  | ??:??:?? 2 [GPS] Unable to reconfigure GNSS - defaults maintained. Is this module GPS-only?
WARN  | ??:??:?? 2 [GPS] Got NAK for class 06 message 11
WARN  | ??:??:?? 2 [GPS] Unable to enable powersaving mode for GPS.
INFO  | ??:??:?? 2 [GPS] GNSS module configuration saved!
```

GPSFan, 2024-05-18:
What type of module was it sold as? Neo-6, Neo-7 or M8?

madeofstown, 2024-05-18:
M8N
[image]

madeofstown, 2024-05-18:
So far I've been unable to download the reference materials for this module because they are behind a baidu link...

GPSFan, 2024-05-18:
That definitely falls into the counterfeit or fake category, the little 8 pin chip to the left is supposed to be an eeprom, the M8 series has internal flash and did away with that leftover of the Neo-6's.
It is not a real u-blox module. It most likely uses a salvaged 8030 chip inside but lacks the internal Flash that would old the upgraded 3.01 firmware.
Datasheets/ integration guides etc are all freely available on the u-blox web site:
https://www.u-blox.com/en/product/neo-m8-series

madeofstown, 2024-05-18:
Well, counterfeit or not, it seems to be maintaining and updating GPS lock. I'm going to leave it for a few hours to see for sure. 
But I'll also be sure to leave a negative review for the AliExpress vendor.

GPSFan, 2024-05-18:
The internal 8030 chip is genuine, it's the surrounding parts that are substandard and not equivalent to the original. Not made by u-blox, made in some CN sweatshop by child labor, being paid in dog poop and being sold as real with a copied label.
Sorry for the rant, but no one should be buying those modules, there are "better" ones on AliExpress.

Talie5in, 2024-05-19:
> Very strange, I have a Pi3+ and waveshare hat coming on Wed. So I'll be able to see exactly what is going on, I will however need some help to configure the Pi & hat to duplicate your setup.. How [...]

So, running standard install from the .deb from the assets page -> https://github.com/meshtastic/firmware/releases/download/v2.3.9.f06c56a/meshtasticd_2.3.9.f06c56a_arm64.deb
But i've also compiled it myself on the pi using remote-ssh extension in VSCode and PlatformIO

Talie5in, 2024-05-19:
URGHHHH

Looks like serial login gets enabled by default on a fresh install..... once you have enable a serial port in /boot/firmware/config.txt

[image]

raspi-config -> Interface Options -> Serial Port -> 
Login Shell accessible over Serial Port -> No
Would You like the serial port hardware to be enabled -> Yes (although this is already enabled in the /boot/firmware/config.txt

Or
```
sudo systemctl disable serial-getty@ttyS0.service
sudo systemctl stop serial-getty@ttyS0.service
```

Found it with

[image]

Example of disabling it!

[image]

@GPSFan @madeofstown

GPSFan, 2024-05-19:
That is what I had surmised, but without the exact hardware and OS I couldn't be sure.
"The D30CB0D0D30CB0D0D30CB0D0D30CB0D0D30CB stuff should be the NMEA sentences. If I were to guess I'd say there was a serial interface problem with Meshtastic."
If the login on uart1 wasn't disabled, Meshtasticd would not be able to get ownership of /dev/ttyS0, there should have been some sort of error message when Meshtasticd tried to grab ttyS0. That probably needs to go into the code.

madeofstown, 2024-05-19:
I can't believe I didn't think of this... I've been watching the RDX light blink while the Pi boots up without even thinking about it!
I think this issue should stay open until the documentation addresses the problem.

GPSFan, 2024-05-19:
Yeah, I think this is one of those face slapping moments. Docs need to be updated for sure. I guess it is a Pi thing to bring the console serial port out the GPIO header...

madeofstown, 2024-05-20:
> I guess it is a Pi thing to bring the console serial port out the GPIO header...

Yeah, OrangePi does it too but unfortunately there is no easy way to disable it. (need to recompile Armbian and u-boot, still not entirely clear)

Talie5in, 2024-05-21:
> (need to recompile Armbian and u-boot, still not entirely clear).

Just u-boot requires a recompile from what I am reading - but that's a huge ooft.

Talie5in, 2024-06-02:
Closing as PR has been merged.

https://github.com/meshtastic/meshtastic/pull/1267

closed 2024-06-02 by Talie5in