meshtastic/firmware #3921: [Bug]: Unable to build firmware for lora_isp4520

danielkucera, 2024-05-16:
### Category

Other

### Hardware

Other

### Firmware Version

master

### Description

I am unable to build firmware for lora_isp4520. See build log output.

### Relevant log output

```Shell
Executing task: platformio run --environment lora_isp4520 

Processing lora_isp4520 (board: lora_isp4520; platform: platformio/nordicnrf52@^10.4.0; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/lora_isp4520.html
PLATFORM: Nordic nRF52 (10.4.0) > lora ISP4520
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES: 
 - framework-arduinoadafruitnrf52 @ 1.10600.0 (1.6.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - tool-adafruit-nrfutil @ 1.503.0 (5.3) 
 - tool-sreccat @ 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
[nanopb] Installing Protocol Buffers dependencies
Requirement already satisfied: protobuf>=3.19.1 in /home/danman/.platformio/penv/lib/python3.10/site-packages (5.26.1)
[nanopb] No generation needed.
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- RadioLib @ 6.5.0
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.3.0+sha.ee628ee
|-- OneButton @ 2.5.0
|-- arduino-fsm @ 2.2.0+sha.7db3702
|-- Thread @ 0.0.0+sha.1ae8778
|-- Nanopb @ 0.4.8
|-- ErriezCRC32 @ 1.0.1
|-- NonBlockingRTTTL @ 1.3.0
|-- SparkFun ATECCX08a Arduino Library @ 1.3.1+sha.5cf62b3
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- Adafruit Bluefruit nRF52 Libraries
|-- Adafruit Unified Sensor @ 1.1.3
|-- Adafruit nRFCrypto
|-- Adafruit TinyUSB Library
|-- Adafruit Internal File System on Bluefruit nRF52 @ 0.11.0
Building in debug mode
Using meshtastic platformio-custom.py, firmware version 2.3.10.8e35e19f
Compiling .pio/build/lora_isp4520/lib411/Adafruit_TinyUSB_Arduino/arduino/Adafruit_TinyUSB_API.cpp.o
Compiling .pio/build/lora_isp4520/lib411/Adafruit_TinyUSB_Arduino/arduino/Adafruit_USBD_CDC.cpp.o
Compiling .pio/build/lora_isp4520/lib411/Adafruit_TinyUSB_Arduino/arduino/Adafruit_USBD_Device.cpp.o
Compiling .pio/build/lora_isp4520/lib411/Adafruit_TinyUSB_Arduino/arduino/Adafruit_USBH_Host.cpp.o
In file included from /home/danman/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/Arduino.h:65:0,
                 from /home/danman/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Adafruit_TinyUSB_Arduino/src/arduino/Adafruit_USBD_CDC.cpp:29:
/home/danman/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Adafruit_TinyUSB_Arduino/src/arduino/Adafruit_USBD_CDC.h:97:26: error: conflicting declaration 'Adafruit_USBD_CDC Serial'
 extern Adafruit_USBD_CDC Serial;
                          ^~~~~~
In file included from /home/danman/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/Arduino.h:60:0,
[... 57 lines cut ...]
                 from /home/danman/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Adafruit_TinyUSB_Arduino/src/arduino/Adafruit_USBH_Host.cpp:37:
/home/danman/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Adafruit_TinyUSB_Arduino/src/arduino/Adafruit_USBD_CDC.h:97:26: error: conflicting declaration 'Adafruit_USBD_CDC Serial'
 extern Adafruit_USBD_CDC Serial;
                          ^~~~~~
In file included from /home/danman/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/Arduino.h:60:0,
                 from /home/danman/.platformio/packages/framework-arduinoadafruitnrf52/libraries/SPI/SPI.h:24,
                 from /home/danman/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Adafruit_TinyUSB_Arduino/src/arduino/Adafruit_USBH_Host.h:30,
                 from /home/danman/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Adafruit_TinyUSB_Arduino/src/arduino/Adafruit_USBH_Host.cpp:37:
/home/danman/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/Uart.h:95:39: note: previous declaration as 'Uart Serial'
   #define SERIAL_PORT_HARDWARE        Serial
                                       ^
/home/danman/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/Uart.h:106:13: note: in expansion of macro 'SERIAL_PORT_HARDWARE'
 extern Uart SERIAL_PORT_HARDWARE;
             ^~~~~~~~~~~~~~~~~~~~
*** [.pio/build/lora_isp4520/lib411/Adafruit_TinyUSB_Arduino/arduino/Adafruit_USBH_Host.cpp.o] Error 1
============================================================= [FAILED] Took 12.55 seconds =============================================================

Environment    Status    Duration
-------------  --------  ------------
lora_isp4520   FAILED    00:00:12.551
======================================================== 1 failed, 0 succeeded in 00:00:12.551 ========================================================

 *  The terminal process "platformio 'run', '--environment', 'lora_isp4520'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.
```

fifieldt, 2024-09-16:
Confirmed with latest master (2.5.0)

thebentern, 2024-09-16:
@fifieldt I think we need to remove these variants since it's misleading folks. We haven't supported any NRF5832 based board in quite some time. It's relatively resource constrained compared to the NRF52840
[image]

closed 2024-09-16 by thebentern