meshtastic/firmware #3929: [Bug]: Fixed position gets reset upon nodedb reset

prokrypt, 2024-05-18:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.3.x

### Description

From https://github.com/meshtastic/Meshtastic-Android/issues/1044

caveman99, 2024-05-18:
This is not just intended behaviour but also intentional behaviour and will not be changed. If this is not documented, documentation needs to be added.

caveman99, 2024-05-18:
ref: https://github.com/meshtastic/firmware/pull/3451

closed 2024-05-18 by caveman99

nicheath, 2024-06-13:
Is there an explanation why this is desired to be this way? I came here to report it because it was not what I was expecting at all. Discovered nodes (dynamic data) vs my node's configured position (configuration static data) seem like two different things.