meshtastic/firmware #3936: [Feature Request]:  Android client share feature.

clavo2, 2024-05-20:
### Platform

ESP32

### Description

Would it be possible to add a android 'share' feature to the app so it shows up as an app that accepts shared data?  I realize droid can share more than text (pictures, etc) between apps, but curious if just text based shares could be sent to the android app to save from cutting text from one app, changing app, and pasting.  

Pardon my ignorance, but this may allow workflow apps like Tasker to automatically route things to the Meshtastic app thus automating certain things (e.g. incoming SMS texts could be pushed to the Meshtastic app and automatically be send to a predefined channel.  

I can see this as being useful in many ways.  Like maybe in the future pushing emails (might require a separate app catered to this).   e.g. email to: mesht@aol.com - filter sees "MT" in the subject and Tasker or app native pushes that to the Meshtastic app and out it goes. 

Just an idea that may be easy, or really hard to do. 

I know this can be done with CLI, but we all live on phones and not everyone knows how to cat herd Termux for this kind of stuff. 

Thanks for the consideration.

todd-herbert, 2024-05-21:
I like the sound of it! Maybe the feature request would be more at home over at https://github.com/meshtastic/Meshtastic-Android/issues ?

clavo2, 2024-05-26:
moved. thx.