meshtastic/firmware #3938: [Bug]: BME680 fails with BSEC Library version 2.5.0.2

brloomis, 2024-05-20:
### Category

Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.3.10.34aec709

### Description

I have a Heltec Wireless Tracker v1.1 that I attached a BME680 to.
I am building my own firmware from master, and the sensor is discovered OK, but does not work with the latest code.
I also have a Heltec Wireless Stick Lite v3 with a BME680 that exhibits the same behavior, it worked OK on 2.3.9.f06c56a but after flashing 2.3.10.34aec709 it failed in the same way.

Downgrading Bosch-BSEC2-Library in platformio.ini from v1.7.2502 to v1.5.2400 causes it to start working again, reverting this commit:
https://github.com/meshtastic/firmware/commit/fce281f54cb89652a62a41ce51f7d6985208a60a

### Relevant log output

```Shell
With Bosch-BSEC2-Library#v1.7.2502:

DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 1 I2C device found at address 0x77
DEBUG | ??:??:?? 1 Wire.available() = 1
INFO  | ??:??:?? 1 BME-680 sensor found at address 0x77
INFO  | ??:??:?? 2 1 I2C devices found
DEBUG | ??:??:?? 2 acc_info = 0
DEBUG | ??:??:?? 2 found i2c sensor meshtastic_TelemetrySensorType_BME680
INFO  | ??:??:?? 2 Meshtastic hwvendor=48, swver=2.3.10.34aec709
(...)
INFO  | ??:??:?? 13 [EnvironmentTelemetryModule] Environment Telemetry: Initializing
ERROR | ??:??:?? 13 [EnvironmentTelemetryModule] setConfig BSEC2 code: -34
INFO  | ??:??:?? 13 [EnvironmentTelemetryModule] No BME680 state found (File: /prefs/bsec.dat).
WARN  | ??:??:?? 13 [EnvironmentTelemetryModule] updateSubscription BSEC2 code: 14
INFO  | ??:??:?? 13 [EnvironmentTelemetryModule] Init sensor: BME680 with the BSEC Library version 2.5.0.2 
DEBUG | ??:??:?? 13 [EnvironmentTelemetryModule] BME680Sensor::runOnce: bme680.status 14
WARN  | ??:??:?? 13 [EnvironmentTelemetryModule] Could not connect to detected BME680 sensor.
 Removing from nodeTelemetrySensorsMap.

With v1.5.2400 and no other code changes:

DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 1 I2C device found at address 0x77
DEBUG | ??:??:?? 1 Wire.available() = 1
INFO  | ??:??:?? 1 BME-680 sensor found at address 0x77
INFO  | ??:??:?? 2 1 I2C devices found
DEBUG | ??:??:?? 2 acc_info = 0
DEBUG | ??:??:?? 2 found i2c sensor meshtastic_TelemetrySensorType_BME680
INFO  | ??:??:?? 2 Meshtastic hwvendor=48, swver=2.3.10.34aec709
(...)
INFO  | ??:??:?? 13 [EnvironmentTelemetryModule] Environment Telemetry: Initializing
INFO  | ??:??:?? 13 [EnvironmentTelemetryModule] No BME680 state found (File: /prefs/bsec.dat).
INFO  | ??:??:?? 13 [EnvironmentTelemetryModule] Init sensor: BME680 with the BSEC Library version 2.4.0.0 
INFO  | ??:??:?? 13 [EnvironmentTelemetryModule] Opened BME680 sensor on i2c bus
```

Donnie650, 2024-05-20:
This is an issue on the rak4631 with environment sensor also.

HarukiToreda, 2024-05-21:
I've tested the same issue on Heltect V3, Wireless Stick Lite, Wireless tracker

```
DEBUG | ??:??:?? 8 [GPS] GPS powerdown using GPS_EN_ACTIVE
DEBUG | ??:??:?? 8 [cardKB] Using I2C Bus 1 (the second one)
INFO | ??:??:?? 8 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 8 [PowerFSM] Enter state: ON
INFO | ??:??:?? 8 [PowerFSM] Initialise the NimBLE bluetooth module
INFO | ??:??:?? 10 [Screen] Done with boot screen...
DEBUG | ??:??:?? 10 [Screen] showing standard frames
DEBUG | ??:??:?? 10 [Screen] Module wants a UI Frame
DEBUG | ??:??:?? 10 [Screen] Showing 1 module frames
DEBUG | ??:??:?? 10 [Screen] Total frame count: 104
DEBUG | ??:??:?? 10 [Screen] Added modules. numframes: 1
DEBUG | ??:??:?? 10 [Screen] Finished building frames. numframes: 7
INFO | ??:??:?? 14 [EnvironmentTelemetryModule] Environment Telemetry: Initializing
ERROR | ??:??:?? 14 [EnvironmentTelemetryModule] setConfig BSEC2 code: -34
INFO | ??:??:?? 14 [EnvironmentTelemetryModule] BME680 state read from /prefs/bsec.dat.
WARN | ??:??:?? 14 [EnvironmentTelemetryModule] updateSubscription BSEC2 code: 14
INFO | ??:??:?? 14 [EnvironmentTelemetryModule] Init sensor: BME680 with the BSEC Library version 2.5.0.2
DEBUG | ??:??:?? 14 [EnvironmentTelemetryModule] BME680Sensor::runOnce: bme680.status 14
WARN | ??:??:?? 14 [EnvironmentTelemetryModule] Could not connect to detected BME680 sensor.
Removing from nodeTelemetrySensorsMap.
DEBUG | ??:??:?? 24 [Power] Battery: usbPower=0, isCharging=0, batMv=4191, batPct=100
INFO | ??:??:?? 34 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)`
```

Talie5in, 2024-05-22:
Only just noticed there were two bug reports open...
Commented on the other one..

https://github.com/meshtastic/firmware/issues/3956#issuecomment-2124562899

Talie5in, 2024-06-02:
This still persists on 2.3.11

Making the same changes as I did for 2.3.10 and rolling bsec2 back to 1.6.2400 and building and dropping onto a rak4631 with a rak1906 returns sensors back to normal behaviour.

caveman99, 2024-06-02:
This is resolved in 2.3.11. Tested on Rak4631 and t-beam

closed 2024-06-02 by caveman99

brloomis, 2024-06-05:
Seems to be working, although I see this message every so often in my logs on both test devices, maybe 26 times over an 8-hour period:
```
WARN  | 13:06:51 28142 [EnvironmentTelemetryModule] runTrigger BSEC2 code: 100
```