meshtastic/firmware #3956: [Bug]: BME680 data output no longer works

memo-567, 2024-05-22:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.3.10.d19607b Alpha

### Description

After updating to FW 2.3.10, the BME680 data output no longer works.
After downgrading to FW 2.3.9, everything works perfectly again.
Tested twice.

Talie5in, 2024-05-22:
I've tested rolling back https://github.com/meshtastic/firmware/pull/3904 as a test case against RAK4631 (only)

Test #1:
 changing the BSEC2 package back to 1.5.2400, building and flashing... sensors come back (no IAQ displaying)
Test #2:
 changing the BSEC2 package back to 1.6.2400, building and flashing... sensors come back (no IAQ displaying)
 Test #3-7
 changing BSEC iteratively backwards until #898bc7d where Sensor Data returns (last good working commit against meshtastic/firmware, also changing lib path where it changes to bsec2 appropriately)

Looks like 

in platformio.ini
```
lib_deps =
  https://github.com/boschsensortec/Bosch-BSEC2-Library#v1.6.2400
```
in variant.h under rak4631
```
build_flags = ${nrf52840_base.build_flags} -Ivariants/rak4631 -D RAK_4631
  -L "${platformio.libdeps_dir}/${this.__env__}/**BSEC2 Software Library**/src/cortex-m4/fpv4-sp-d16-hard"
```

memo-567, 2024-05-23:
A small side effect of the update and back:
Before the update to 2.3.10, the temperature displayed was 0.8 °C higher than that of the reference thermometer. This was fine since it was still within the BME680's one degree manufacturing tolerance.
After going back and forth, there is a difference of 2.8 °C!
A reset to factory settings does not lead to any changes, as it is not a real reset. Data will still be preserved somewhere on the device. Maybe also (wrong) data stored by the BME980?

memo-567, 2024-05-23:
I have shut down the node with the sensor for now. The difference was getting bigger and bigger.
At the end the temperature difference to the reference thermometer was almost 6 °C.

caveman99, 2024-06-03:
resolved with 2.3.11

closed 2024-06-03 by caveman99

Talie5in, 2024-06-03:
> Works again, but with restrictions only. The air pressure is displayed incorrectly (10.1 hPa) and the line is overwritten on the left side by the new added (V and mA) values. ![photo_2024-06-02_14-54-48](https://private-user-images.githubusercontent.com/58190287/336015590-c20db337-703c-493a-bc1d-356cc8db6636.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0MjQxODgsIm5iZiI6MTcxNzQyMzg4OCwicGF0aCI6Ii81ODE5MDI4Ny8zMzYwMTU1OTAtYzIwZGIzMzctNzAzYy00OTNhLWJjMWQtMzU2Y2M4ZGI2NjM2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjAzVDE0MTEyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZmZjdkMWNhOTgyYmYzOWM2MzJhZjgxODY0OGM0MmU4MmU2NjhlNjgyZTU3ZjE0ZWFlODk5MWE5MDg4NjE5ZGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hVKJ-xi2vAc1vD7Zz4bpYAMAnewr5NgLGZwKPK0IR1g)

Thats a client side issue - the Android app needs updating to suit from everything I've seen discussed.

memo-567, 2024-06-03:
Possibly the voltage and current values, but the incorrect air pressure can probably be traced back to the Bosch library.