meshtastic/firmware #3961: [Bug]:

gadiscz, 2024-05-23:
### Category

Hardware Compatibility

### Hardware

DIY

### Firmware Version

2.3.6.7

### Description

[image]
Hi, I'm trying to upload fw pres VS Code to ESP32, but after successfully uploading, I get an ESP32 error and reboot.

It works through the web, I just don't get VS Code.
Other projects work for me in VS Code.
What could be wrong?

roha-github, 2024-05-23:
I don't have problem with ESP32 and that firmware. Tested build and upload via VSCode on old Heltec V2 and newer Heltec V3.

gadiscz, 2024-05-23:
[image]

I downloaded from git fw to the VS Code folder.
I choose "default_envs = meshtastic-diy-v1" or "default_envs = nano-g1-explorer" both have ESP32 and without any change I compile and upload to ESP32. (ESP+OLED+RA-01SH) via the web it works, I just have to change the code because of the RA-01SH)

I tried uploading through esptool and also the same error.

todd-herbert, 2024-05-23:
Just checking: have you seen this page before? https://meshtastic.org/docs/development/firmware/build/
Sorry if I'm repeating info you've already seen. Just want to make sure!

roha-github, 2024-05-23:
If you use DIY, how do you specify the ESP32 variant?

https://github.com/meshtastic/firmware/blob/2f9dc813d3c64b738777bfdba1efc9a4697f7ac2/bin/device-install.bat#L34

gadiscz, 2024-05-23:
Yes, the problem may be that it sets me as a bad ESP32, but I don't know where to set a type? 
And what exactly to set?

todd-herbert, 2024-05-23:
> Yes, the problem may be that it sets me as a bad ESP32, but I don't know where to set a type?
And what exactly to set?

If you follow the [official instructions](https://meshtastic.org/docs/development/firmware/build/), it tells you how to set a type, or how to make a diy type.

gadiscz, 2024-05-24:
yes, I made new hw in Variants and set it in platformio.ini.   but how do i define a board?
[image]
[image]

gadiscz, 2024-05-24:
Ok, I'll try to make a new board. But on the website it was enough to use DIY v1. I also found out that there is no esp32dev.json file created for ESP32 dev in the boards, so I will try to create it and set the parameters in it.

todd-herbert, 2024-05-24:
>But on the website it was enough to use DIY v1

Oh, I didn't read this part close enough.
If you use "DIY V1" with https://flasher.meshtastic.org, then it works? Is that correct?

Can we try this fresh, from the very start with no changes? First want to check that it's possible to use DIY V1 with meshtastic

Instead of setting default_env, maybe it's easier to select like this:
[image]

Then you can click this button to build the firmware and upload it to your device:
[image]

Is this the same technique you have already tried?
I think it's good to test if this technique works first, without any code changes for RA-01SH.

gadiscz, 2024-05-24:
I tried again on the website and the device started up. Then I tried the same bin file with esptool and it also worked.
I then downloaded firmware-master again and selected "meshtastic-diy-v1" again and the same ESP32 error still reboots with the same error. I tried uploading the same generated file "firmware.factory.bin" via esptool and also the same error.

[image]
[image]

gadiscz, 2024-05-24:
[image]

I tried to open another ESP32 project and upload it normally and it works through VS Code.

todd-herbert, 2024-05-24:
Is it okay to see fullscreen screenshots instead? Especially when you are using VS Code, there might be some other clue available there.

I see that when you use the web flasher, you select 2.3.6. I think right now you are probably building the newest version, instead of 2.3.6? Maybe it's good to test if it's possible to build 2.3.6 with VS Code. This might show whether the difference is because of different firmware versions, or whether it's different when you use VS Code.

You can select v2.3.6 in VS Code like this:
[image]

todd-herbert, 2024-05-24:
Interesting! Does 2.3.10 work through VS Code? If yes, then we know the issue is quite new, and maybe know where to look.

todd-herbert, 2024-05-24:
A suggestion just now from @caveman99: can you please try change the VS Code version from 2.3.6 to "master", then use "Erase Flash", and then upload?

[image]

Gnu-Bricoleur, 2024-05-25:
Hi, 

I was working on adding keypad support for the Chatter https://github.com/meshtastic/firmware/issues/2896
I was building and flashing with VSCode, everything was working properly, I tried to pull the latest master (from a12b992 to 2233507) in order to submit a merge request and now, when flashing the Chatter (it's ESP32 based), it goes into a bootloop.

I reverted all my changes and the current master doesn't works for me on the Chatter
I tried erasing the flash and flashing again

I'm suspecting something happened with this commit https://github.com/meshtastic/firmware/commit/0c9da9aec7ef6b01ab209f22cafb5538cdf58b6e
And I feel it's related to this issue too

thebentern, 2024-05-25:
Yes there is a regression on the original ESP32 that is not present on the S3 based devices which are working fine on the new framework. We will hunt down a resolution before a new release, but as of now, master is not working for these devices.

caveman99, 2024-06-07:
fixed in 2.3.11 firmware release

closed 2024-06-07 by caveman99