meshtastic/firmware #3962: [Feature Request]: Regulatory gain setting for BLE

KyleMaas, 2024-05-23:
### Platform

NRF52, ESP32

### Description

Similar to #3836, but for the BLE radio on devices which support it.

It would be nice if you could set the gain level of the Bluetooth antenna to stay within regulatory limits.  The purpose for this would be so you could mount a router/repeater node in inaccessible areas and use a Yagi/Uda antenna for the BLE side pointing at a more accessible area.

thebentern, 2024-05-23:
I'm not aware of the BLE radios used in Meshtastic devices being even capable of exceeding any ISM region's 2.4GHz max EIRP. Can you point us to some documentation of one that you're concerned about?

KyleMaas, 2024-05-23:
Not with a stock antenna, but if you use a high-gain antenna and the limits are in EIRP, the default power level for a radio assuming a dipole is going to be drastically too high without a TX gain reduction at the radio.

KyleMaas, 2024-05-23:
Example antenna for what I'm talking about:

https://www.amazon.com/Tupavco-TP512-Wi-Fi-Antenna-2-4GHz/dp/B008Z4DNFC/

KyleMaas, 2024-05-23:
Here's one rated at 18dBi:

https://www.amazon.com/TECHTOO-Directional-Connector-Wireless-Extension/dp/B00OCJYPCY/

And another for 23.5dBi:

https://www.amazon.com/Premiertek-Directional-High-Gain-Parabolic-ANT-GRID-24DBI/dp/B005M8KU3W/

KyleMaas, 2024-05-23:
This stems from this discussion:

https://github.com/meshtastic/firmware/discussions/3932#discussioncomment-9499785

Which would be great, except that in some places where you could install a router/repeater node, they are so far from the ground that they would be outside of normal Bluetooth range when running on a regular dipole antenna.  A directional antenna would be great and would drop the RX noise floor so you could talk to it over a longer range, but you'd need to drop the transmit power by a corresponding amount to stay within regulatory limits.

thebentern, 2024-05-23:
Even the NRF52 at max only makes it to 8db. 99% of folks aren't using high gain antennas with BLE, so I'm not sure there is a compelling reason to parameterize this in the mainline firmware. This seems like a good case for a compile time macro perhaps.