meshtastic/firmware #3967: [Feature Request]: Proposal of new role

mdziczkowski, 2024-05-24:
### Platform

ESP32

### Description

I would like to propose a new role for Meshtastic:

**Router/Repeater**

This would allow Meshtastic to work in same time in both modes

### Proposal idea explanation

When implemented and installed (for example on a radio tower) then it would allow to extend the Meshtastic range.

For example:

Besides recieving and transmiting the Meshtastic signal locally, It would allow to send recive the signal thru other media, with would additionally increase the communication range

thebentern, 2024-05-24:
I'm afraid this is not possible at the hardware level. Transceivers cannot RX while simultaneously TXing

mdziczkowski, 2024-05-25:
I didn't meant "at same time" 😉

I meant it more in the way that it could work normally as (for example) repeater (to avoid unessecary signal delays in between the devices) for the internal network, but would also allow to allow to send to (and recieve from) external data (like router).

Maybe someone in the community would have some free time to experiment and find a nice solution how to implement it 😜

As there goes a saying:

"Nothing is impossible... it's just difficult to achive" 😉