meshtastic/firmware #3973: Request for new definition to support Ebyte E22 modules[Board]:

NancyDeville, 2024-05-25:
### Discussed in https://github.com/meshtastic/firmware/discussions/3948

<div type='discussions-op-text'>

<sup>Originally posted by **NancyDeville** May 21, 2024</sup>
### SOC

ESP32

### Lora IC

Ebyte E22-900M30S / Ebyte E22-900M33S

### Product Link

Aliexpress

### Description

I've been trying to achieve full transmit power with these Ebyte modules for around a year and so far none of my builds have performed to give full power output, they all top out at 22db representing the standard SX126x,
I have tried everything at this point in terms of wiring options of TX_EN/DIO2 also tried all sorts of regulators and capacitors in power supply, people on discord have sent variant.h files trying different things and nothing has worked,
I'm building firmware DIY hydra, V1 /V1.1,
After hours of online research I think the issue is with the settings in radio lib that in particular the register codes, txco settings, regulator mode, PA_ CONFIG and PA_Ramptime, 
Reading other projects online we also need to change max power settings in TX_PARMS from 22 to 30 for the E22-900M30S, note that also the E22-900M33S has different requirements and only needs 9db drive, and seems to have some different things in setup, I think again registers are different, 
Myself I'm more into hardware and I'm not really knowledgeable when it comes to writing code so I'm hoping someone can have a look at this and fill in the blanks, from what I can see it looks like we need a setup specifically for the ebyte modules that include PA and LNA, treating them totally as a standard SX1262x is not running them to their full potential as legal limits and power restrictions are written into the meshtastic radiolib</div>

NancyDeville, 2024-05-25:
Ok so ebyte got back to me, they don't know meshtastic and can't offer any support,
I've done a hell of alot of reading on this issue and the more I read up on the ebyte E22 full power output issue the more it seems to now be a meshtastic issue,
Other projects that have had this issue have overcome the problem now buy changing things in the code,

I have managed to find the bits in the code I was looking for regarding power limits and regions and max tx power and pa ramp time etc, and I've spent about 40 hours now trying different things with no luck,

Me being into hardware and not doing much code I really wanted the team here to have another look into this issue and maybe write a profile specifically for the E23 modules, I say modules plural as we now have the E22-900M33S which has different setup requirements than the E22-900M30S ( 33 only needs 9db drive) ( this need to be at radio_begin otherwise we are potentially going to damage modules during channel changes as I've noted this also updates output power as part of the ch settings)

I have noted EBYTE_E22 in the code but I don't know just how much work has been done to fully support it let alone the issue I mentioned with the latest module so that would need it's own definition

I don't feel moving this from and issue as I reported it to a discussion helps, there has been endless discussion and there is obviously an issue here

Anyway I'm hoping we can resolve this issue and get on with using these modules to their full potential

NancyDeville, 2024-06-03:
Ok so today I built the nrf pro micro build with the ebyte E22 and guess what, same thing again only this time even the Rx LNA seems to not be engaged, pulls 170mA max, 
So that's three builds that don't work properly using the ebyte E22, 
It's really not good enough to just wire up a module and get them receiving and sending messages and saying the build is tested and working then adding them to the list of tested working builds,
I think meshtastic needs to review this situation as there are a lot of builds all affected by this same issue, frustratingly it seems like it's something really minor, I personally don't think using sx126x is working for the ebyte and would like to see a specific E22 definition in variant.h,

I can't stress enough how much time and money I've put into this and done a year of intensive testing on the basis that these build were tested and working and I'm not alone there are many frustrated meshtastic users all following this for updates most have largely given up but I would like to see it resolved or actually pull the builds from the list or clearly state the current limitations of the E22

NancyDeville, 2024-09-13:
Hi, sadly I've had to pack down my workbench ATM as forced to move out where i live under extreme duress going through a bad time at the moment but I'm glad to see some progress and believe me I'm itching to get back to my happy place messing with electronics but could be a while, maybe get nestpebble to try it out, i know hes also got the modules and keen to get the 33s working I'm still trying to keep up to date on the feed but unable to do anything practical right now

fifieldt, 2024-09-16:
Hi @NancyDeville , just happened upon your last message and just wanted to send happy thoughts your way :) The community is here for you.

jesselichtenberg, 2024-09-29:
> Use this variant https://github.com/meshtastic/firmware/blob/master/variants/EBYTE_ESP32-S3/variant.h, all details are in the comments, choose the RF switching mode you prefer, simply connect the module to your chosen ESP32S3 pin in your variant file, and feed the 30S/33S module a stable 5 V. You can adapt the board.json and pins_arduino.h file for the standard ESP32.
> 
> I recommend outputting **maximum 8 dBm** from the SX1262 for the E22-900M**33**S, this should get you 33 dBm output, the datasheet has this information. The 33S will need more current, make sure your 5 V power supply can handle it. Use common GND. The datasheet says anything above 9 dBm will damage the board.
> 
> On the E22-900M**30**S, you can safely output 22 dBm from the SX1262 to get 30 dBm module output, check [the correlation for the E22-900M**30**S **only**](https://github.com/S5NC/EBYTE_ESP32-S3/blob/main/E22-900M30S%20power%20output%20testing.txt).

Just getting into this and using the 33S as well as an ESP32 S3 with the larger 16MB flash and 8MB PSRAM. My plan was to use your variant, but I believe I need to specify a different board.json given the flash and PSRAM differences. I'd also planned to limit the radio to 8dbi in the variant.h file and then use the app to restrict it farther to remain in compliance with local regulations. Is it easier to just specify another ESP32S3 board (Pico for instance) which supports PSRAM?

Thank you!