meshtastic/firmware #3976: [Bug]: Triple-click behaviour for GPS doesn't work as expected

Nestpebble, 2024-05-26:
### Category

Other

### Hardware

DIY

### Firmware Version

2.3.10

### Description

Following on from testing for [issue 3742](https://github.com/meshtastic/firmware/issues/3742), it appears that triple-click to disable the GPS doesn't always behave correctly.
I'm using a Pro-micro DIY board with a transistor to switch the GPS ground as required, but the results have been replicated on other hardware.
The issue is: when triple-click is used to disable the GPS whilst the GPS is in sleep mode, the pin_gps_en pin doesn't go low (i.e. off), and the GPS isn't told to sleep permanently. After the sleep period has elapsed (i.e. 120 seconds), the GPS wakes up and tries to communicate with the MCU, but is ignored. It consumes full power doing this, and does so for an indefinite period of time, effectively negating any power savings from the sleep mode.
When the GPS is triple-click disabled while it is in search mode, it is powered off via the gps_en pin correctly, and the sleep command is issued correctly.
Some work has been done to find the issue, discussed on the discord in #"GPS EN Pin discussion" and summarised [in these messages](https://discord.com/channels/867578229534359593/1241937144382230588/1242803343621885962).

todd-herbert, 2024-05-26:
Just to add: [this branch](https://github.com/todd-herbert/meshtastic-firmware/tree/blind-gps-fix-attempt) *might* fix the issue. I gave it a quick check with the device I had on-hand, and it seemed to work, but needs some more testing to confirm.

todd-herbert, 2024-06-07:
Hopefully solved by #4041

Nestpebble, 2024-06-09:
Solved by #4041

closed 2024-06-09 by Nestpebble