meshtastic/firmware #3981: [Bug]: rak4631 shutting down from keypress that never happened

robertfoss, 2024-05-27:
### Category

Other

### Hardware

Rak4631

### Firmware Version

firmware-rak4631-2.3.10.d19607b

### Description

Hardware: RAK19007+RAK4631+RAK12500(gps)+RAK13800(ethernet)+RAK19018(poe)

After running for ~30 minutes the device shuts itself off. I've tested two firmware versions , `firmware-rak4631-2.3.6.7a3570a` and `firmware-rak4631-2.3.10.d19607b`.

The log mentions a long press, but no button is being pressed.

### Relevant log output

```Shell
INFO  | 21:37:57 902 [Button] Shutdown from long press
INFO  | 21:38:00 905 [Button] Shutting down
INFO  | 21:38:00 905 [Button] Entering deep sleep forever
INFO  | 21:38:00 905 [Button] Disable NRF52 bluetooth
DEBUG | 21:38:00 905 [Button] SX126x entering sleep mode (FIXME, don't keep config)
DEBUG | 21:38:00 905 [Button] Completed sending (id=0x48df96cb fr=0xaf to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1716845876 hopStart=3 priority=64)
INFO  | 21:38:00 905 [Button] GPS deep sleep!
INFO  | 21:38:00 905 [Button] Saving /prefs/db.proto
INFO  | 21:38:03 908 [Button] Saving /prefs/config.proto
INFO  | 21:38:05 910 [Button] Saving /prefs/module.proto
INFO  | 21:38:07 912 [Button] Saving /prefs/channels.proto
INFO  | 21:38:09 914 [Button] Setting GPS power=0
DEBUG | 21:38:09 914 [Button] Sleep Time: 0
```

todd-herbert, 2024-05-27:
Does it always happen at the same time? (Roughly 900 seconds after boot)

todd-herbert, 2024-05-27:
Wonder if the default button asignment is conflicting with one of your modules?
I've just spotted [this discussion](https://meshtastic.discourse.group/t/rak-4631-shutting-down-without-button-push/12186/7), where the problem was solved by moving the module, or reassigning the button gpio

ianmcorvidae, 2024-05-27:
If the RAK12500 isn't in slot A, then yes, the user button should be reassigned, as the default pin for that (9) is among the pins that the gps might hit, in other slots. This is a pretty common source of these sorts of phantom keypresses

robertfoss, 2024-05-27:
The RAK12500 was in Slot D, and not Slot A. So this is a user error on my part. Thanks for the very quick attention!

closed 2024-05-27 by robertfoss