meshtastic/firmware #3982: [Feature Request]: Setting to configure a periodic reboot

clwgh, 2024-05-28:
### Platform

NRF52, ESP32, RP2040

### Description

It would be useful to have an option to configure a periodic reboot in the firmware. This would accept a number of seconds, perhaps no fewer than 300 (5 minutes) to ensure there is always time to change the value or disable it.

For example, something like the below to enable a automatic reboot every 6 hours.

`meshtastic --set power.periodic_reboot true --set power.reboot_every_secs 21600`

The rationale for this is that in the event of a bug which manifests over time, due to the nature of the node processing packets (such as https://github.com/meshtastic/firmware/issues/3931), a reboot can temporarily clear the problem and provide a workaround. This makes the difference between a usable node and an unusable node.

This can, of course, be performed manually, and for a connected node by scheduling a `--reboot`. For an unconnected node the above would provide a means to do the same.

garthvh, 2024-05-28:
If there is a bug we need to find the bug, rebooting is not a solution to whatever the issue is.

closed 2024-05-28 by garthvh

clwgh, 2024-05-28:
And while the bug continues to not be found, rebooting is a workaround that makes the node usable. I'll do it electronically instead if you're unable to help.

garthvh, 2024-05-28:
We have been pretty consistent about not adding this feature over several years.  You can reboot a node with the admin channel, and can automate that a number of ways yourself as you have noted. 

Duplicate of https://github.com/meshtastic/firmware/issues/1193

clwgh, 2024-05-28:
Sure, was just an idea to avoid the need for even more connected hardware and power supplies. Sadly, a periodic reboot is the only way I can use this node while it continues to crash after 12-20 hours of uptime. On reflection it makes sense to keep edge case functionality out of core firmware so I'm in agreement. I did search for any previous but didn't find that. Thanks.

b8b8, 2024-05-28:
what hardware are you running when it crashes?

clwgh, 2024-05-28:
Pico W's with Ebyte and Waveshare modules.

NobodyYouKnow, 2026-03-19:
I'll cast my vote for the automatic reboot capability. Pulling an offline remote node down off a pole to reboot it is problematic. This will likely work even when the device has become disabled by an as yet unknown bug. Saying "lets fix the bugs instead" does not equate to "there are no bugs to be found". Once all the bugs are found and Meshtastic no longer hangs I will happily reconsider.