meshtastic/firmware #3983: [Feature Request]: Report altitude from pressure

Adrelien, 2024-05-28:
### Platform

ESP32

### Description

I have noticed that altitude is not accurate at all from GPS modules like neo 6m. I think the bme280 pressure sensor is far more reliable. 
https://pilotinstitute.com/pressure-altitude-explained/

todd-herbert, 2024-05-28:
Does this vary depending on the weather? I know aircraft often need to get weather information from the local airport to find to get their local pressure and calibrate the altimeter.

Adrelien, 2024-05-28:
Well the BME280 does have a temperature sensor as well.

todd-herbert, 2024-05-28:
Is it that simple? I'm all for it if it's possible. It probably doesn't have to be super accurate either to beat some of the GPS altitudes I've seen.

Jorropo, 2024-05-28:
Planes uses barometers as differential measurements more than absolute ones.
For example when flying at FL310 you will be 1000ft above someone flying at FL300, even tho in reality you might be flying at 25000ft and 24000ft, it's more convenient to use a wrong but standard sea level pressure as separation to avoid mid air collisions is what matters.
It also help to normalize flying performance as lower pressures at identical altitude very directly correlate to identical pressure at higher altitude.

In situations where flying into terrain is likely\* they will use a local reference (usually an airport which have an automated weather station), the airport knows it's local temperature, local barometric pressure, and it's altitude, it then use t

\*this is given to you by controllers and automated robots as TA and TL, which are **T**ransition **A**ltitude and **T**ransition **L**evel respectively, theses are altitude and level where bellow you use local corrected pressure and above you use standard 1013.25hpa.
The interesting thing to note is that there are two numbers not one, usually 1000ft apart, this create a 1000ft buffer so planes on different references don't fly into each other because they are on different references.

---

# All of this to say:

> Can you get altitude from pressure ?

Yes however it requires knowing the pressure and altitude at some nearby site.

You could imagine a `FIXED_POSITION` node reporting pressure and altitude, and some nearby node comparing the barometric value to give an altitude.

> Can you get altitude without a reference ?

No.
But you can assume the sea level pressure is 1013.25hPa.
You can imagine embedding a database of average sea level pressure based on the location, time in the day and time of year.

> Will assuming the reference be accurate ?

If you want to get an idea, go to https://metar-taf.com/ then look for hPa measurements, then as a rule of thumb for every hPa above or bellow 1013.25 you would incorrectly report the node 30ft above or bellow where it really is.
I havn't tried using a database to correct for local tendencies, I guess it would be better as the sun and yearly weather patterns are significant contributors. This approach becomes worst overtime as climate change make the climate less predictable and does not match historical data as well.

> Can a temperature sensor help ?

Yes but not that much, the temperature sensor help correct assumptions in how pressure scales as you climb or descent it doesn't let you recover any reference.

closed 2024-05-28 by garthvh

emiliogon, 2024-11-20:
This has been marked as "not planned", but I think it should be reconsidered.  It might not produce precise altitudes, but the request is for something _better than GPS._

There's a standard formula, calibration can be accomplished a number of means (manual entry, region table, fancy app-linked GPS lookups, etc.), and even approximate numbers have a good chance of being far superior and faster to acquire than a GPS lock.  Proper application of temperature and humidity sensors requires calibration, and this is no different.  Some logic could then derive the most accurate number from GPS and sensor values.

The goal here is to produce higher-quality trackers, which is one of the major appeals of Meshtastic.  Why not pursue that, even just as an option?

garthvh, 2024-11-20:
> This has been marked as "not planned", but I think it should be reconsidered. It might not produce precise altitudes, but the request is for something _better than GPS._
>
> There's a standard formula, [...]

Are you willing to write the code or sponsor the feature with money?

emiliogon, 2024-11-20:
Possibly; I'm set up with the dev environment and getting oriented in this sizable code base.

But, is that the only bar for features?  My feature evaluation is always "what value does this add to the project versus the level of effort required?"